JMRI/JMRI #6886: Printed "BASIC" info is wrong

chrisridd, 2019-05-01:
I have a loco using long addresses, which displays correctly in the summary but when I choose to print the roster entry, prints that it is using short addresses in the "BASIC" section.

Attaching the printed roster entry (using the Mac's print to PDF)

[110 165-8.pdf](https://github.com/JMRI/JMRI/files/3134647/110.165-8.pdf)

bobjacobsen, 2019-05-01:
Very interesting!  Could you also attach a copy of the roster file?  It should be called  110_165_8.xml and be in the “roster” directory inside your JMRI preferences folder. The easiest way to find that folder is to open the Locations... item in the JMRI Help menu, then clck Open Roster Location to open the folder.

Or it is (probably) located in the ~/Library/Preferences/JMRI/(current profile)/roster folder.

You have to add a “.txt” to the filename to be able to attach it here.

Thank you, and sorry for the trouble.

Bob

> On May 1, 2019, at 6:40 AM, Chris Ridd <notifications@github.com> wrote:
> 
> I have a loco using long addresses, which displays correctly in the summary but when I choose to print the roster entry, prints that it is using short addresses in the "BASIC" section.
> 
> Attaching the printed roster entry (using the Mac's print to PDF)
> 
> 110 165-8.pdf

chrisridd, 2019-05-01:
Thanks Bob, here's my roster XML file. I don't think I've changed much from the defaults yet, apart from the address, which certainly looks "long".

[110_165_8.txt](https://github.com/JMRI/JMRI/files/3134737/110_165_8.txt)

bobjacobsen, 2019-05-01:
Thank you!  Definitely a bug, though I’m not quite sure where yet.  Will try to get it fixed this weekend.

chrisridd, 2019-05-01:
No problem! It isn't too critical, I'm just kicking the tyres on my new DCC system to see how things work.

PR #6998 (2019-05-25): Roster Print needs to load locomotive (CV) file
The roster print operation must make sure it has the full contents of the roster entry's variables from the locomotive file before printing.

Fixes #6886
closed 2019-05-25 by bobjacobsen