JMRI/JMRI #6890: Consistently failing tests in Jenkins build jobs

bobjacobsen, 2019-05-01:
There are currently some consistently failing tests in the Jenkins Builds job, which runs `ant logheadlesstest`

First group: 
 - [ ]  jmri.implementation.DefaultConditionalTest.testTriggers  
 - [ ] jmri.implementation.DefaultConditionalTest.testActionCurrentState
 - [ ] jmri.implementation.DefaultConditionalTest.testAction

All three of those seem to be failing because the tests are run at logging level DEBUG to get more coverage, and some debug statements are throwing an exception:

```
java.lang.UnsupportedOperationException: Not supported yet.
	at jmri.implementation.DefaultConditionalTest$TestConditionalAction.getTypeString(DefaultConditionalTest.java:896)
	at jmri.implementation.DefaultConditional.takeActionIfNeeded(DefaultConditional.java:1248)
	at jmri.implementation.DefaultConditional.calculate(DefaultConditional.java:304)
	at jmri.implementation.DefaultConditionalTest.testActionCurrentState(DefaultConditionalTest.java:566)
```

The other error is:
 - [ ] jmri.server.json.time.JsonTimeSocketServiceTest.testOnMessage

The traceback for this is:
```
junit.framework.AssertionFailedError: Time is current expected:<2019-05-01T22:16:3[8.03]1+0000> but was:<2019-05-01T22:16:3[7.97]1+0000>
	at jmri.server.json.time.JsonTimeSocketServiceTest.testOnMessage(JsonTimeSocketServiceTest.java:66)
```

pabender, 2019-05-01:
That JSon time test is failing because it is comparing time using a moving clock.  It either 

 1. Shouldn’t be comparing for equality.
 2. Should be working with a stopped clock

rhwood, 2019-05-01:
That time one is verifying that a Date value is copied to two other values while firing a PropertyChangeEvent. That’s it.  It *was* using a moving value, but is now testing that two listeners get a copy of the listened to value. How that copy is producing inconsistent values is a mystery to me.

bobjacobsen, 2019-05-02:
PR #6897 fixed the 1st group of three.

bobjacobsen, 2019-05-02:
I don't understand the analysis above about what's going wrong in the JSON code.  Here's the code:
```
        ObjectNode data = connection.getObjectMapper().createObjectNode();
        data.put(JSON.RATE, rate); // integer
        data.put(JSON.STATE, JSON.ON); // start the fast clock -- to test that listeners set in onMessage work
        service.onMessage(JsonTimeServiceFactory.TIME, data, JSON.POST, Locale.ENGLISH);
        message = connection.getMessage();
        current = manager.getTime();
        Assert.assertNotNull("Message is not null", message);
        Assert.assertEquals("Rate is fast", rate, message.path(JSON.DATA).path(JSON.RATE).asDouble(), 0.0);
        Assert.assertEquals("Timebase is on", JSON.ON, message.path(JSON.DATA).path(JSON.STATE).asInt());
        Assert.assertEquals("Time is current", formatter.format(current),
                message.path(JSON.DATA).path(JsonTimeServiceFactory.TIME).asText());
```
(`manager` is set to the TimebaseManager a few lines above)  The assert is comparing `current` to something taken from `message`.  `current` was loaded from a specific call to `manager`, which wasn't the same call as where-ever the `message` got it's time.  So the clock could clearly have run between them.

rhwood, 2019-05-02:
I made a change in that test, which this is not showing, so maybe I made it in a branch that not's a PR yet. I'll go look and see.

rhwood, 2019-05-02:
I had a change to this test where there were two listeners and I compared the two listeners to ensure the time change was 1) being captured, and 2) being encoded correctly, but I can't find it in any branch (I may have inadvertently lost it when I recently cleaned up some branches). I'll have it recreated by Saturday if you wait on that.

bobjacobsen, 2019-05-02:
Thank you! Sorry for the extra effort.  

Git doesn't really delete branches directly, so you might still be able to locate those via e.g. `git log` or gitx/gitk.

bobjacobsen, 2019-05-02:
(sorry, that should have been `git reflog`, bitten by autocorrect)

bobjacobsen, 2019-05-03:
Another approach: https://stackoverflow.com/a/16813291/8408461

rhwood, 2019-05-03:
#6898 should have the correct JsonTimeSocketServiceTest tests now.

bobjacobsen, 2019-05-05:
@rhwood Thank you!

closed 2019-05-05 by bobjacobsen