JMRI/JMRI #6894: LayoutEditor - WebServer - Block State Indication - 4.15.6

sidlo64, 2019-05-02:
In Layout Editor is created layout with Entry-Exit and Full Interlocking.
Set route by T13 and T11.
Block B10, B9, B7, B6 and B5 are set as alternate.
Then change state sensors S10, S9, S7, S6 and S5 to state Active and back Inactive.
Blocks step by step change state.

Open web browser with panel BugLayoutWeb. Repeat steps above. Block does not change state in web browser.

Affected release current 4.15.6.
Production release 4.14 is OK.

[layout-web.xml.txt](https://github.com/JMRI/JMRI/files/3139173/layout-web.xml.txt)

[1-SetAllUnknownSensorsToInactive.py.txt](https://github.com/JMRI/JMRI/files/3139175/1-SetAllUnknownSensorsToInactive.py.txt)

[image]

mstevetodd, 2019-05-02:
Hi Petr, 
The json websocket service was recently changed to no longer accept username as the "name" when retrieving layoutBlocks. Not sure why that was done. Usernames are still accepted for other types, such as "turnout".
For example, in your panel you have a layoutBlock with system name "ILB1" and username "B1". This websocket request works at 4.14:
`{"type":"layoutBlock","data":{"name":"B1"}}`
but returns a 404 error at 4.15.6ish:
`{"type":"layoutBlock","data":{"name":"ILB1"}}`
works in both versions.
As a workaround, you can insure your system and user names are identical.

rhwood, 2019-05-02:
@mstevetodd can you point to that change? I recently changed the WebSocket server to accept username or systemname for every NamedBean type; even though I also made the web panel code include the systemname instead of the username in the XML sent to a client. The recommendation to use systemname instead of username stems from potential encoding issues between different browsers for some common characters in usernames that are not in system names.

mstevetodd, 2019-05-02:
@rhwood, I provided an example of what was working and is no longer working. I did not search for the actual change. The xml panel contains references (such as tracksegments) that are only by username. If systemnames are to be used, that's great, but that change needs to be made throughout the existing code.

rhwood, 2019-05-03:
@mstevetodd The changes were all verified against existing test cases, which *all* use system names for the name property assuming the test didn't just verify that a service instance could be instantiated without error. I can only write code changes against what I know, and what documentation and code comments indicates should be true (that the name property in JSON will be a system name) and since the JSON services have *always* set the `name` property to the system name of a NamedBean, there was no apparent reason to expect a `name` property to not be a system name. If there is code that is not behaving as documented and does not have a test to verify that its behaving correctly, expect it to slip through changes unless you can vouch for the continual testing of it in JMRI/master and don't chid us for not thoroughly testing everything unless you are actively providing test cases for us to test against--none of us use *all* of JMRI, and we can't test those portions of what we don't know (I, for example, make extensive use of the JSON services, but do not use all elements in the panel.js client, so I can't test how it expects the JSON services to behave with that particular client).

rhwood, 2019-05-03:
@sidlo64 Can you provide a complete profile with the panel and the connections it requires?

mstevetodd, 2019-05-03:
@rhwood, no offense or chiding was intended.

I agree that the json "name" property was not previously documented. Given past comments from others regarding using username throughout JMRI, I would have assumed that username was actually the preferred path. 
I see you've now stated that system name is required, with this addition: https://github.com/JMRI/JMRI/commit/a2fbddb#diff-752442da7e9bbf1667e90dede0bd125e

I ask you to reconsider this requirement. It is a great convenience to be able to request by username.

I don't expect you to test or support panel.js, I've been happily doing so for years. I would appreciate if a change is made that you think might impact panel.js, you'd send me an email or otherwise alert me of the change.

pabender, 2019-05-03:
> I don't expect you to test or support panel.js, I've been happily doing so for years. I would appreciate if a change is made that you think might impact panel.js, you'd send me an email or otherwise alert me of the change.

It is highly preferred that we do this via automated testing.

We can certainly write tests that load panels and interact with them through a web browser.  We do some web interactions with our cucumber tests already.

mstevetodd, 2019-05-03:
@pabender If you're willing to set up an example of that test, I'd suggest using the OneOfEach panel already shipped with JMRI. I can then clone your test for other functionality I normally look for manually.

pabender, 2019-05-03:
@mstevetodd ok, let me work on that.  It might take me a couple of days (end of semester, busy time of year at work).

sidlo64, 2019-05-03:
Profile 
[BugMultiSensor.jmri.zip](https://github.com/JMRI/JMRI/files/3142816/BugMultiSensor.jmri.zip)

rhwood, 2019-05-04:
The JSON system uses systemNames and not userNames [because userNames are optional and may not be unique](http://jmri.org/JavaDoc/doc/jmri/NamedBean.html); because encoding whitespace reliably in a URL is difficult, and because encoding "non-safe" characters in a URL is insanely difficult (there are multiple schemes for encoding non-safe characters; different clients may make different choices about how to encode; some RFCs even discuss the fact that whitespace *may* be encoded differently in a URL depending where it falls in the URL). systemNames are less likely to suffer from those issues because they generally are much more well defined.

PR #6900 (2019-05-05): fix: handle LayoutBlock usernames in JSON
Fixes #6894

Note that both 7997671 and 49daf95 are independent fixes that work on their own.
closed 2019-05-05 by mstevetodd