JMRI/JMRI #6903: Ecos communication breakdown

rekolok, 2019-05-05:
Hi,

I started using JMRI just a few weeks ago and all worked well until a few days ago.
My Ecos is connected to the LAN port and the Mac via WiFi.
All connections are fine, I do have access to the Ecos with the Mac directly but no more via DecoderPro.
Whenever I open DecoderPro, it says Ecos is online but the program does not recognize my Ecos (button = unknown status) and the status window remains blank.
Even when the Ecos is turned off I see the that it is apparently online in the lower text bar in green.

I have tried re-installing JMRI on the computer, of course after de-installing and deleting all program files, but it does not work.
I cannot start from scratch, after installing the software again I get back the old window where it got stück. I do not even get the start up windows.
Should I also remove my profile file before re-loading JMRI?

Resetting the Ecos does not solve the problem either.

Can anyone help to solve my problems?

Rgds,
Alain

bobjacobsen, 2019-05-05:
Have you checked to make sure that JMRI has the ecos connection properly configured in the JMRI connection preferences?

Also, check for relevant messages in the console (available from the Help menu)

rekolok, 2019-05-06:
The configuration fits and there is no message in Ecos.
As I said, all worked fine in the beginning and it stopped without obvious reason.
When I set up JMRI for the first time, it would not start without the Ecos being online. Now it starts whatever status of the Ecos and it says that the Ecos is online even when this is turned off.

I tried setting up JMRI from scratch but it always returns to the non-working state from before.

dheap, 2019-05-06:
Check Preferences->Defaults to make sure that nothing is set to "Internal".

rekolok, 2019-05-06:
Everything is set to Ecos, except "command station".
Problem is, I cannot change the settings. They stay what they are.

rekolok, 2019-05-10:
#[image]
[image]
[image]

Made a few screen shots.
After deleting everything that was related with JMRI, emptied waste bin and reinstalled.
Guess what, the set up that I had before popped up again, no start up screen appeared and again all is stuck.

Why is it that I cannot start over from scratch, downloading and setting everything up again?

Now I give up!

pabender, 2019-05-10:
If you want to start over, you also need to delete your profile, or use a new one.

When your in the preferences, click the item from the menu that says “Config Profiles”.  Once you do that, you will find a button that lets you create a new one.  Do that, give the profile a name, and restart and you will be able to configure from scratch.

rekolok, 2019-05-11:
Ok, that worked.
Still I cannot make the connection with the Ecos.
The IP is correct since it is no problem connecting my Mac to Ecos.
Any other idea?

rekolok, 2019-05-11:
This is the log from the last attempt:

2019-05-11 19:18:31,932 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-05-11 19:18:33,453 util.Log4JUtil                        INFO  - This log is appended to file: /Users/Macky/Library/Preferences/JMRI/log/messages.log [main]
2019-05-11 19:18:33,454 util.Log4JUtil                        INFO  - This log is stored in file: /Users/Macky/Library/Preferences/JMRI/log/session.log [main]
2019-05-11 19:18:33,461 apps.AppsBase                         INFO  - DecoderPro version 4.14+Rd060e0b starts under Java 1.8.0_161 on Mac OS X x86_64 v10.14.4 at Sat May 11 19:18:33 CEST 2019 [main]
2019-05-11 19:18:33,791 gui3.Apps3                            INFO  - Starting with profile Thale.3e79b4cc [main]
2019-05-11 19:18:33,911 node.NodeIdentity                     INFO  - Using jmri-opoQEbHzGhOiaaxpJ2QLhD-3e79b4cc as the JMRI Node identity [AWT-EventQueue-0]
2019-05-11 19:18:34,641 util.FileUtilSupport                  INFO  - File path program: is /Applications/JMRI/ [main]
2019-05-11 19:18:34,641 util.FileUtilSupport                  INFO  - File path preference: is /Users/Macky/Library/Preferences/JMRI/Thale.jmri/ [main]
2019-05-11 19:18:34,642 util.FileUtilSupport                  INFO  - File path profile: is /Users/Macky/Library/Preferences/JMRI/Thale.jmri/ [main]
2019-05-11 19:18:34,643 util.FileUtilSupport                  INFO  - File path settings: is /Users/Macky/Library/Preferences/JMRI/ [main]
2019-05-11 19:18:34,643 util.FileUtilSupport                  INFO  - File path home: is /Users/Macky/ [main]
2019-05-11 19:18:34,644 util.FileUtilSupport                  INFO  - File path scripts: is /Applications/JMRI/jython/ [main]
2019-05-11 19:18:36,196 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: request(1, view) consecutive timeouts = 0 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:38,203 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: get(1, status) consecutive timeouts = 1 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:40,209 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: request(11, view) consecutive timeouts = 2 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:42,215 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: queryObjects(11, addrext) consecutive timeouts = 3 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:44,222 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: queryObjects(26, ports) consecutive timeouts = 4 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:46,228 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: queryObjects(10, addr, name, protocol) consecutive timeouts = 5 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]
2019-05-11 19:18:48,237 jmrix.AbstractMRTrafficController     WARN  - Timeout on reply to message: set(1, stop) consecutive timeouts = 6 in ecos.EcosTrafficController [ecos.EcosTrafficController Transmit thread]

rekolok, 2019-05-30:
Nobody has a solution?

danielb987, 2020-08-19:
I'm closing this issue since there has been no comment on the problem for over a year.

closed 2020-08-19 by danielb987