JMRI/JMRI #6906: Three tests failing consistently when Jenkins runs AllTest

bobjacobsen, 2019-05-05:
The [Jenkins AllTest job](http://builds.jmri.org/jenkins/job/Development/job/AllTest/) is consistently showing three similar failures:

- [jmri.jmrit.display.controlPanelEditor.EditPortalFrameTest.testCTor](http://builds.jmri.org/jenkins/job/Development/job/AllTest/lastCompletedBuild/testReport/jmri.jmrit.display.controlPanelEditor/EditPortalFrameTest/testCTor/) is logging:
```
java.awt.IllegalComponentStateException: component must be showing on the screen to determine its location
	at java.awt.Component.getLocationOnScreen_NoTreeLock(Component.java:2062)
	at java.awt.Component.getLocationOnScreen(Component.java:2036)
	at jmri.util.PlaceWindow.getScreen(PlaceWindow.java:77)
	at jmri.util.PlaceWindow.nextTo(PlaceWindow.java:128)
	at jmri.jmrit.display.controlPanelEditor.EditPortalFrame.<init>(EditPortalFrame.java:118)
	at jmri.jmrit.display.controlPanelEditor.EditPortalFrameTest.testCTor(EditPortalFrameTest.java:24)
```

- [jmri.jmrit.display.controlPanelEditor.EditPortalDirectionTest.testCTor](http://builds.jmri.org/jenkins/job/Development/job/AllTest/lastCompletedBuild/testReport/jmri.jmrit.display.controlPanelEditor/EditPortalDirectionTest/testCTor/) is logging:
```
java.awt.IllegalComponentStateException: component must be showing on the screen to determine its location
	at java.awt.Component.getLocationOnScreen_NoTreeLock(Component.java:2062)
	at java.awt.Component.getLocationOnScreen(Component.java:2036)
	at jmri.util.PlaceWindow.getScreen(PlaceWindow.java:77)
	at jmri.util.PlaceWindow.nextTo(PlaceWindow.java:128)
	at jmri.jmrit.display.controlPanelEditor.EditPortalDirection.<init>(EditPortalDirection.java:70)
	at jmri.jmrit.display.controlPanelEditor.EditPortalDirectionTest.testCTor(EditPortalDirectionTest.java:24)
```

- [jmri.jmrit.display.controlPanelEditor.EditCircuitPathsTest.testCTor ](http://builds.jmri.org/jenkins/job/Development/job/AllTest/lastCompletedBuild/testReport/jmri.jmrit.display.controlPanelEditor/EditCircuitPathsTest/testCTor/) is logging:
```
java.awt.IllegalComponentStateException: component must be showing on the screen to determine its location
	at java.awt.Component.getLocationOnScreen_NoTreeLock(Component.java:2062)
	at java.awt.Component.getLocationOnScreen(Component.java:2036)
	at jmri.util.PlaceWindow.getScreen(PlaceWindow.java:77)
	at jmri.util.PlaceWindow.nextTo(PlaceWindow.java:128)
	at jmri.jmrit.display.controlPanelEditor.EditCircuitPaths.<init>(EditCircuitPaths.java:88)
	at jmri.jmrit.display.controlPanelEditor.EditCircuitPathsTest.testCTor(EditCircuitPathsTest.java:24)```

It's not clear to me from reading the code why this happens with Jenkins and not during Travis and AppVeyor CI.  The main environmental difference is that Jenkins is running the logalltest target, which runs with DEBUG level logging.  But I'd do see how that's relevant.

rhwood, 2019-05-05:
What OS is Jenkins running?

pabender, 2019-05-05:
@rhwood I believe it is a Linux System.

pabender, 2019-05-05:
Notice that all three of these are occurring within their respective constructors.

The line in the first one of those reads:
setLocation(jmri.util.PlaceWindow. nextTo(_parent._editor, null, this));

I think the error is saying that _parent._editor is not currently visible on the screen.

It would probably make sense to catch this error when determining the position in that nextTo method.

pabender, 2019-05-05:
After looking at this further, this appears on Jenkins because of debug mode:

https://github.com/JMRI/JMRI/blob/62c61262667c97b994f72c14a5806a7c90e978a2/java/src/jmri/util/PlaceWindow.java#L75-L85

line 77 above is causing the issue.

pabender, 2019-05-05:
#6907 should take care of this.

commit 5843f377 (2019-05-05): Merge pull request #6907 from JMRI/pabender-patch-1

Fix #6906
commit c13cbffd (2019-05-05): Fix #6906
closed 2019-05-05 by bobjacobsen