JMRI/JMRI #6912: Intermittent Swing crash running Warrant tests

bobjacobsen, 2019-05-06:
The log in the vicinity of the crash (some redundant java lines elided):
```
    [junit] INFO  - Throttle at 0.19372946 needs 6369.262mm to ramp to Stop but only 1100.0mm available. May Overrun. Warrant EastToWestLink [main] jmri.jmrit.logix.Warrant.?()
    [junit] INFO  - Throttle at 0.25100264 needs 7441.349mm to ramp to Stop but only 100.0mm available. May Overrun. Warrant EastToWestLink [main] jmri.jmrit.logix.Warrant.?()
    [junit] ERROR - paint failed in thread AWT-EventQueue-0 34:  [main] jmri.jmrit.display.Editor.?()
    [junit] java.lang.ArrayIndexOutOfBoundsException: No such child: 31
    [junit] 	at java.awt.Container.getComponent(Container.java:336)
    [junit] 	at javax.swing.JComponent.rectangleIsObscured(JComponent.java:4390)
    [junit] 	at javax.swing.JComponent.paint(JComponent.java:1054)
    [junit] 	at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
    [junit] 	at jmri.jmrit.display.Editor$TargetPane.paint(Editor.java:658)
    [junit] 	at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
    [junit] 	at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:290)
    [junit] 	at javax.swing.RepaintManager.paint(RepaintManager.java:1272)
    [junit] 	at javax.swing.JComponent._paintImmediately(JComponent.java:5158)
...
    [junit] 	at javax.swing.RepaintManager.access$1200(RepaintManager.java:64)
    [junit] 	at javax.swing.RepaintManager$ProcessingRunnable.run(RepaintManager.java:1732)
    [junit] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
...
    [junit] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
    [junit] INFO  - Throttle at 0.091487125 needs 3627.754mm to ramp to Stop but only 2100.0mm available. May Overrun. Warrant WestToEastLink [main] jmri.jmrit.logix.Warrant.?()
    [junit] INFO  - Throttle at 0.12749943 needs 5271.1187mm to ramp to Stop but only 1100.0mm available. May Overrun. Warrant WestToEastLink [main] jmri.jmrit.logix.Warrant.?()
```

This can happen multiple times in a single AllTest run. This particular one was during `jmri.jmrit.logix.LinkedWarrantTest.testBackAndForth`, but it happens in other warrant tests occasionally too.

The fault happens in `jmri.jmrit.display.Editor$TargetPane.paint(Editor.java:658)`, which is a `super.paint(g)` call while properly running on the Swing/AWT thread to do the repaint:

```
            // It is rather unpleasant that the following needs to be done in a try-catch, but exceptions have been observed
            try {
               super.paint(g);
               paintTargetPanel(g);
            } catch (Exception e) {
                log.error("paint failed in thread "+
                    Thread.currentThread().getName()+" "+Thread.currentThread().getId()+": ", e);
            }
```

(Love that comment!) 

The fault is (probably) that some other part of the code is accessing, and probably removing, Swing components within the `Editor` frame while _not_ on the Swing/AWT thread.  From the stack trace, it looks like a component directly within the `JLayoutPanel _targetPanel` variable, i.e. one of the icons displayed in the `Editor` window.

I don't know the Warrant code well enough to find that reference, however.  Where does it remove (or perhaps replace?) items on an Editor window?

pabender, 2019-05-06:
Sent from my iPad

> On May 6, 2019, at 11:06 AM, Bob Jacobsen <notifications@github.com> wrote:
> I don't know the Warrant code well enough to find that reference, however. Where does it remove (or perhaps replace?) items on an Editor window?
> 

I know at least some of the warrant tests are moving marker icons around the panel.  This could be the culprit.  I don’t know this code well enough either, though I could probably track down the issue...

pabender, 2019-05-17:
Looking through the Editor code, there are methods `addToTarget()` and `removeFromTarget()` that both call remove on the target panel. (addToTarget does a remove then an add).

If you trace where these are called, both methods can be called via public methods.

Since the tests in question appear to be using LocoIcons, there are at least three methods we need to look at further: `addTrainIcon()`,`addLocoIcon()`, and `putLocoIcon()`.

I've run out of time to look at this tonight, and I won't have much time to look further until sometime on Monday.

Paul

pabender, 2019-05-28:
#7018 ( now merged ) hopefully hides this problem during tests.  This does not resolve the race condition leading to the error.

closed 2020-01-08 by bobjacobsen