JMRI/JMRI #6919: CMRI Network Monitor Help Missing

KenC57, 2019-05-08:
When you open the CMRI Network Monitor, the window is not sized right. Only the top half shows, only error metrics. The Data Metrics and buttons do not show.

Resizing and closing the window does not save the size, so it has to be resized every time.

If you hit the Help->Window Help, you get the following stack dump and error.

> javax.help.BadIDException: Not valid ID: package.jmri.jmrix.cmri.serial.serialmon.CMRInetMetricsFrame
	at javax.help.Map$ID.create(Map.java:126)
	at javax.help.CSH.displayHelp(CSH.java:743)
	at javax.help.CSH.access$200(CSH.java:80)
	at javax.help.CSH$DisplayHelpFromSource.actionPerformed(CSH.java:1668)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.AbstractButton.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(Unknown Source)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

icklesteve, 2019-05-08:
Other CMRI support pages are missing as well, see #6164

jerryg2003, 2019-10-15:
First item in this Issue actually refers to the C/MRI Network Metrics window (not the Network Monitor) and duplicates what was reported in Issue #6433 (second item),  Still an issue in PanelPro 4.17.3

jerryg2003, 2021-02-26:
The original issue from above is not resolved as of 4.21.4:

When you open the CMRI Network Metrics {incorrectly stated above as Network Monitor], the window is not sized right. Only the top half shows, only error metrics. The Data Metrics and buttons do not show.

Resizing and closing the window does not save the size, so it has to be resized every time."

The issue of clicking on help throwing an exception is resolved.

jerryg2003, 2021-03-15:
Remaining part of issue is redocumented in #9542.

closed 2021-03-15 by jerryg2003