JMRI/JMRI #6921: LokProgrammer CV List Import fails in in low memory situations

dheap, 2019-05-09:
This arose from a proposed change to `-Xmx` in `build.xml` (since reverted) as part of #6917.

A quick summary is that [LokProgImporter.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/symbolicprog/LokProgImporter.java)  fails with an `OutOfMemoryError: GC overhead limit exceeded` when invoked from a Loksound 5 definition in a low-memory situation such as `ant` targets or a Raspberry Pi.

For initial comments, see #6917. Later ones have been transferred to this issue.

dheap, 2019-05-09:
Present indications are that a Raspberry Pi user should be able to edit/read/write a LokSound 5 with the new definitions but that the rarely-used "Import LokProgrammer CV list file" will fail.

We can look at the cause of the heap blowout with that command later.

dheap, 2019-05-09:
> @pabender it seems to take a long time to open the LokSound 4 definitions too, even on my primary development machine. Making that more efficient would improve the user experience.

Agreed, but:

- A V4 definition/roster entry has 1016 CVs and 4410 Variables.
- A V5 definition/roster entry has 2035 CVs and 9895 Variables.
- There are also some CVs/Variables in both I'm not yet reading ~60ish from the import log.

I dont think any other definition in JMRI is in this CVs/Variables league...

dheap, 2019-05-09:
When LokProgImporter reads a V5 file it invokes CvValue.setValue(value) 2,035 times.

1. This fires up to 2,035 PropertyChangeEvents.
2. This causes up to 9,895 VariableValues to receive a PropertyChangeEvent and update the variable's value.
3. This causes up to 9,895 VariableValues to fire a PropertyChangeEvent.
4. These up to 9,895 PropertyChangeEvents may cause up to ~9,895 pane field updates and subsequent screen updates.

This is what I believe overloads the heap space and the Garbage Collector.

If we can throttle (1) without freezing the remainder, I think  the problem will go away.

Code change suggestions for [LokProgImporter.java](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrit/symbolicprog/LokProgImporter.java) gladly welcomed...

dheap, 2019-05-09:
@rhwood I welcome your comment and suggestions as well.

rhwood, 2019-05-09:
Two things in `CvTableModel`:
- Try replacing Vector with ArrayList (see https://stackoverflow.com/a/1386288/176160)
- When doing a bulk set of changes (a.k.a. reading the CV file, not reading the values from the decoder (that takes long enough each CV can be updated as its read)), don't call `fireTableDataChanged()` for each change, but call it only once at the end of the bulk set of changes, and then iterate over the CvValues and fire any pending change events. (`fireTableDataChanged()` causes the entire table to be dumped and redrawn). You may also want to look at firing only the specific per-row and per-cell change notifications that allow only that cell or row to be redrawn instead of requiring the entire table be redrawn; note however, that these change notifications all need to be able to immediately read the model, so you still want to not fire them while performing a bulk change.

In `CvValue`:
- Provide a way to hold a pending change event instead of immediately firing it so that bulk changes in the holding CvTableModel can be done without firing changes.

In `LokProgImporter`, I don't see anything that can be done to improve memory usage, since I am not familiar with an less expensive ways of String handling.

Just looking at this code, I honestly think the bottlenecks are in the JTable <-> CvTableModel and anything <-> CvValue interactions during changes, not in the LokProgImporter itself (its just that using the LokProgImporter is revealing them).

dheap, 2019-05-09:
@rhwood Thanks for those insights. It sounds like deep waters!

bobjacobsen, 2019-05-09:
Could somebody please attach a test file for importing?  I can look at this with some heap tools late this evening, but I don't know how to recreate it.

dheap, 2019-05-09:
@KenC57 wrote:
> The importer is only needed for what? I take it that is the thing that would make all the sound slots and like meaningful instead of just 'slot 1'?

No, the importer is low-usage. It only imports CV values that can be exported from LokProgrammer via its Tools->Export CV List File.

It doesn't have anything to do with Sound Slot names. Th's the subject of some other work I started years ago in the V4 era but haven't progressed far.

Uses are:

1. A user who has received a decoder that has been incorrectly programmed by the Vendor/OEM, with failure to write-through the project-specific CVs to the rewritable "Factory Defaults". This is an almost non-existent case these days. Earlier versions of LokProgrammer had the "Write CV values Factory to Defaults" (paraphrased as I don't have LokProgrammer open) off by default. This was fixed years ago and Matthew Herman requested I remove the warning I had in DecoderPro's Factory Reset item. The user can request a CV List file from another user/OEM/ESU. With V5, almost certainly a non-event.
2. An advanced user who also has and uses a LokProgrammer and desires to quickly transfer CV settings from a publicly-available or own custom project. A handful of users.
3. A user who is developing and testing ESU definitions. One user; @dheap 

> I doubt there is a way to defer the updates (change events) until done. Or is there a way it could trigger the GC every 500 elements or something like that?

@rhwood  has made suggestions [here](https://github.com/JMRI/JMRI/issues/6921#issuecomment-490846972). None of this impacts on code in #6917.

> Can you document what is the smallest amount of memory required during the current form for using this? I would say if the release note could warn "only use importer if you have XXX memory or larger", that would do for now. With so many looking forward to this feature, I'd say it is worth having if we know the risk.

On my Mac Pro I just found -Xmx768m failed but -Xmx1024m had no problems.

dheap, 2019-05-09:
@pabender wrote:
> @dheap can we just download one of the projects from http://projects.esu.eu/ To use as a test file?  ( working from my IPad, can’t try it right now ).

You also need Windows (I use VMWare Fusion) and the LokProgrammer software (free download) to extract the CV list file.

pabender, 2019-05-09:
@dheap as Bob and I have both requested, please provide a data file.

We need to see what part of this process is causing the memory to be out of control.

I will ask a question I have asked already:
can we just download one of the projects from

http://projects.esu.eu/

To use as a test file?

pabender, 2019-05-09:
> @pabender wrote:
> 
> > @dheap can we just download one of the projects from http://projects.esu.eu/ To use as a test file?  ( working from my IPad, can’t try it right now ).
> 
> You also need Windows (I use VMWare Fusion) and the LokProgrammer software (free download) to extract the CV list file.

Could you please provide that CV list file as an example?

I don’t have a windows machine available.

dheap, 2019-05-09:
> @bobjacobsen Could somebody please attach a test file for importing? I can look at this with some heap tools late this evening, but I don't know how to recreate it.

Use with a new "lokSound 5" roster entry.
[S0601-LS5H0M4-Diesel-ALCO-12-244-V1-R1.txt](https://github.com/JMRI/JMRI/files/3163718/S0601-LS5H0M4-Diesel-ALCO-12-244-V1-R1.txt)

dheap, 2019-05-09:
@bobjacobsen wrote:
> I recommend _actually_ _measuring_ memory usage as a first step. Our intuition of usage tends to be off-target for GC systems; estimates based on execution tend to be bad too. 
> 
> It’s not that hard to get a first cut:
> - get a clean start
> - attach jvisualvm and select Monitor
> - do the memory intensive thing
> - take a heap dump
> - ask it for the top retained memory items, then take a break while it works
> - when finally done, look to see what’s retaining memory

I've been using jvisualvm but couldn't see what was retaining it. I wasn't aware how to do your last two steps.

KenC57, 2019-05-09:
Dave,
Sounds like you have another winner. I'd mention your test results as a clue
about memory. But I'd say you got it ok.

pabender, 2019-05-10:
Just after loading a version 5 ESU LokSound definition, I get the following as the top 10 memory consumers (second line is the retained size):

```
jmri.jmrit.symbolicprog.VariableTableModel#1
31,803,599
jmri.jmrit.symbolicprog.tabbedframe.PaneServiceProgFrame#1
16,169,739
java.util.Vector#4564
15,837,210
java.lang.Object[]#310743
15,837,174
java.util.Vector#4560
15,836,692
java.lang.Object[]#302741
15,836,656
org.jdom2.ContentList#19528
14,168,577
org.jdom2.Content[]#12019
14,168,529
org.jdom2.Element#26400
12,098,096
org.jdom2.ContentList#26403
12,098,024
```
Total memory used at the point of the heap dump was 460,002,879 bytes.

During the read of the CV list @dheap provided, memory spiked multiple times over 600,000,000 bytes.
It crashed on me during a spike that jvisualvm shows consumed 650,000,000 bytes (and this was on my main development machine, running from ant).

I did manage to take a heap dump before the load, when total memory used was 636,731,610 bytes.
jvisualvm is working on top 10 consumers now.  I will post when it finishes.

dheap, 2019-05-10:
@pabender Thanks for looking into that.

I'm currently doing some tests of everyday operations (identify, read, write) with V5 definitions and severely restricted Xms values. The results are surprising me. More when completed...

pabender, 2019-05-10:
not sure what happened, but jvisualvm never  did give me the top 10 list for the second heap dump.  Will try again later.

dheap, 2019-05-10:
Set "-Xmx512m" (is that reasonable for a typical Raspberry Pi with 1GB RAM?) and did tests (on my old Core II Duo iMac with NCE Power Pro connection) of the typical operations 99% of users do:

- Read type from decoder.
- Open and save new roster entry.
- Read Full Sheet on the CVs pane.
- Save and close.
- Open existing Roster Entry
- Write Full Sheet on the CVs pane.

No problems or speed differences (relative to "-Xmx2048m") were encountered.

The two things that were not possible:
1. Have two V5 roster entries open at the same time.
2. Import LokProgrammer CV List file.

I'm sure most Raspberry Pi users could live without (1). I'd estimate than less than 1% of users will ever have a need/desire to use (2) and almost none would use it on a regular basis.

But let's see if we can eventually get (2) working with "-Xmx512m".

dheap, 2019-05-10:
Note #6927 for speed improvement when reading/writing.

bobjacobsen, 2019-05-12:
Here's what I'm seeing for memory usage from the decoder definition, and during the load:
[image]

Unfortunately, the memory is so tight at the end that the heap dump is getting inconsistent results.  More to follow on details...

dheap, 2019-05-12:
That's much the same as I've been seeing.

bobjacobsen, 2019-05-12:
Two big things so far:  `_writeButton` and `_readButtons` in `variableModel` in the `PaneServiceProgFrame` are (1) holding 9893 items and (2) some of those are retaining a lot of storage, so they each end up being responsible for about 15MB.  That 9893 items sounds about right from the discussion above, and 2*15MB actually isn't that big on the scale of the retained total. I'll keep looking.

bobjacobsen, 2019-05-12:
The 2nd bit thing is 9755 JComboBoxes, which together retain 172MB.  That's  lot. These are also rooted in `variableModel` via `rowVector`.  Much of that storage comes from each one having it's own editor and renderer; perhaps we can save some space by using common ones.

dheap, 2019-05-13:
I believe I've found the problem!

A few debugging statutes to see when it bottlenecked and I found it.

Expect a PR today...

dheap, 2019-05-13:
Tests show it imports a LokSound 5 CV list in about 15 seconds on my old Core 2 Duo iMac, even when limited to "-Xmx512m".

pabender, 2019-05-13:
@dheap, I am glad you found something that helped.  That is a tremendous improvement.

@bobjacobsen long term reducing that memory load is what it takes for this to scale.  When this was designed, I am sure nobody thought we would see the day when a single decoder had almost 10,000 configurable items in it.  Now, I must admit it wouldn’t surprise me if 10 years from now, there is some decoder that has at least an order of magnitude more.

dheap, 2019-05-13:
#6940 on my Mac Pro with "-Xmx640m" imports:
- A LokSound 5 CV list in about 6 seconds , down from 140 seconds and needing "-Xmx1024m".
- A LokSound V4.0 CV list in 3 seconds as against 15 seconds with V4.15.5

dheap, 2019-05-13:
Before:
[image]

After:
[image]

dheap, 2019-05-13:
Further test results at #6921.

rhwood, 2019-05-13:
@dheap Good job finding that optimization.

> The two things that were not possible:
>
> 1. Have two V5 roster entries open at the same time.
> 2. Import LokProgrammer CV List file.
>
> I'm sure most Raspberry Pi users could live without (1). I'd [...]

But I do want to comment on this: I am certain there are Raspberry Pi users who would like to be able share a configuration profile between a Raspberry Pi and another computer and have things "just work" even if the profile contains multiple ESU decoder equipped locomotives (I'm not suggesting they want to open the programming GUI on the Raspberry PI, but they do want to be able to query the roster, and run the engines). I don't know if running a WiThrottle with an ESU equipped locomotive "opens" a RosterEntry in the sense you are using, but I know WiThrottle does open and load the RosterEntry.

I'd personally like the -Xmx512m target to be even smaller, so that a Raspberry Pi Zero could be reliably used as an embedded JMRI instance in an N-Trak module, but that's probably looking for other optimizations elsewhere (since I'll only be wanting a roster in that case if I need it for managing LocoNet or OpenLCB devices in the module).

dheap, 2019-05-13:
@rhwood wrote:
> But I do want to comment on this: I am certain there are Raspberry Pi users who would like to be able share a configuration profile between a Raspberry Pi and another computer and have things "just [...]

My understanding has always been that WiThrottle simply used the information cached in `roster.xml`, which is opened at startup - it has Function Labels etc.

Just did a WiThrottle test, picked a LokSound 5 loco from my roster on the iPad app, Set and drove it. No signs of any loading delay and the screenshot below shows no signs of memory loading.
 
> I'd personally like the -Xmx512m target to be even smaller, so that a Raspberry Pi Zero could be reliably used as an embedded JMRI instance in an N-Trak module, but that's probably looking for other [...]

Yes, that'll have to be elsewhere in the code. See [earlier comment](https://github.com/JMRI/JMRI/issues/6921#issuecomment-491658483) by @pabender.

[image]

dheap, 2019-05-13:
@rhwood In case you were wondering about that rising usage, see below - obviously the GC kicks regularly:
[image]

dheap, 2019-05-13:
Closed via #6940

closed 2019-05-13 by dheap