JMRI/JMRI #6926: Create new NamedBean hardware mismatch

icklesteve, 2019-05-10:
First noticed with creating a sensor range, reproduced with Turnouts.

Mismatch between the hardware type selected and the validation performed.

Eg. with NCE Sim + MERG CBUS Sim ( can create both types with integers ),
create a range of sensors 1-10 in one type.
When creating a range of sensors from 1-10 in the other type you may get 10 dialogue boxes saying that the address already exists, then only create 1 of the new type.

Is also causing range creation to fail silently intermittently.

Behaviour changes, seemingly dependent on what hardware tab has been selected in the Table, and if the hardware type has been re-selected ( or not re-selected ) in the hardware dropdown in the Add New Sensor / Turnout window.

silverailscolo, 2019-05-14:
Looking into this, but not seeing this happen in the PP tabbed tables with NCE Sim and Merg Sim. If a hardware number is in use, the Add pane just adds the next 10 silently. I might have the warnings turned off as a user setting.
Can @icklesteve describe the exact order of tabs/combos selected?

silverailscolo, 2019-05-14:
Confirmed that when adding a Merg range of sensors (leaving the All tab of the table frame untouched, selecting MERG from the Add sequence dialog) stops after adding 1 item (named ``MS+1``) with this dialog:
[image]
System name mentioned in dialog is referring to ``M1`` instead.

silverailscolo, 2019-05-17:
Working on a fix, only CbusLightManagerTest remaning to do.

icklesteve, 2019-05-17:
Thanks for looking into this @silverailscolo much appreciated :-)
Validation on the non-leading + format has been awkward since it was added.

When I was getting the mis-match there seemed to be some sort of trigger which made it was repeatable for a while, then it de-triggered itself and returned to normal.

Will re-attempt to re-create the mis-match and log every click!
Steve

silverailscolo, 2019-05-22:
@icklesteve can you check if this issue is fixed in current head?

icklesteve, 2019-05-23:
> @icklesteve can you check if this issue is fixed in current head?

Thanks, will do,
Steve

icklesteve, 2019-05-24:
Hi @silverailscolo, am not able to reproduce in current head :-)

Many thanks for looking into this,
Steve.

closed 2019-05-24 by icklesteve