JMRI/JMRI #6936: Text on Sensor icon invisible on web page

Bluegoldfish1947, 2019-05-12:
I am creating a route setting control panel and have a collection of buttons each with text on them. 
This text appears on the 'normal' console screen 
[image]
and on the panel preview displayed on the web site
[image]
however when I select the panel to use on the web site the button text vanishes
[image]
The text has been entered using the overlay button text dialog
[image]

Is this a known problem or am I missing something about adding text to sensor images?

Thanks,

Chris Morgan

rhwood, 2019-05-12:
Please zip your profile containing this panel and add it to this bug. That will allow us to use your setup while working on the fix.

Bluegoldfish1947, 2019-05-12:
Hi,

Thanks for the prompt reply. I assume by profile you mean the .xml file. Here it is...
[Slattocks3.zip](https://github.com/JMRI/JMRI/files/3169982/Slattocks3.zip)

rhwood, 2019-05-12:
@Bluegoldfish1947 That will do, but I was hoping for a complete Config Profile (in PanelPro, go to Preferences->Config Profiles; select the profile to share, and click "Export" to create a zipped profile).

rhwood, 2019-05-12:
@Bluegoldfish1947 I am getting errors loading your panel XML file related to missing resources. Please provide the complete profile.

Bluegoldfish1947, 2019-05-12:
I'm getting an error when I try to perform the export...
[image]

After checking in the Slattocks folder I can confirm that the ProfileConfig.xml file is missing.

This is a ZIP of the Slattocks.jmri folder if it is of any help...
[Slattocks.jmri.zip](https://github.com/JMRI/JMRI/files/3170160/Slattocks.jmri.zip)

FYI I am running JMRI PanelPro 4.14+Rd060e0b with Java version 1.8.0_211(en_GB) under windows 10. 

The Slattocks folder originates from a Raspberry Pi running the same version of JMRI. The same missing text problem occurs when running on the Raspberry PI.

rhwood, 2019-05-13:
The overlay text is available to the web browser, but its Z index places it under, not over, the button. @mstevetodd Can you take a look at that?

mstevetodd, 2019-05-14:
I will look into the color and centering issues pointed out by @rhwood in #6945 

@Bluegoldfish1947 can you zip up and attach the resources/icons/buttons and resources/icon/signals folders from your user files folder? That should get me past the loading errors so I can create your panel and see the issue.

Bluegoldfish1947, 2019-05-14:
@mstevetodd Here is the complete resources folder for you...
[resources.zip](https://github.com/JMRI/JMRI/files/3178851/resources.zip)
I hadsent the entire profile folder earlier, see a posting above.
Let me know if you need anything further.

mstevetodd, 2019-05-15:
@Bluegoldfish1947 Thanks. Sorry I missed the earlier file. I've made some changes to better handle the font, color, font size, and centering. (Note that the centering only works if you specify a fixed width.)
I'll submit my changes shortly. I expect I've missed getting them into v4.16, but they should be in the following test version.

Bluegoldfish1947, 2019-05-20:
Thanks for all your help and hard work om this. I look forward to a future version containing the fixes. I'll report back here the effect of any updates.

Bluegoldfish1947, 2019-08-10:
I have at last installed version 4.16. 
I'm pleased to see that the text on buttons does now appear on the web display **_but_** it is placed in the top left corner and is not centred horizontally and vertically as in the main display.

mstevetodd, 2019-08-12:
@Bluegoldfish1947 did you specify a fixed width for the text?

Bluegoldfish1947, 2019-08-20:
Hi,

Sorry for the delay in getting back to you, I've been away.

The text settings I used were...

Font: Dialog
Size: 12
Style: bold
Colour: black
Justification: Centre
Orientation: Horizontal

I hope this is the information you need.

Regards,

Chris

mstevetodd, 2020-07-02:
Chris, for the centering to work on the client-side panel, you must specify a fixed size. See screenshot.
[image]

mstevetodd, 2021-03-30:
IIRC, the issue with the z-index has been addressed.

PR #6945 (2019-05-14): fix: Place overlay text above, not below, button.
Fixes #6936 

Note that this only fixes the immediate issue, but that other cosmetic changes (centering and color) are needed to match the button in the panel.
commit b3ad522b (2019-05-14): fix: Place overlay text above, not below, button.

Fixes #6936 

Note that this only fixes the immediate issue, but that other cosmetic changes (centering and color) are needed to match the button in the panel.