JMRI/JMRI #6985: Digitrax CV11 decoder file reversed

jimexler, 2019-05-24:
Recently I found what appeared to be a contradiction in a JMRI Digitrax file for the sound decoder. The CV11 (sound time out) was reversed in Decoder Pro.

I searched online and found a discussion on the JMRI site but with no conclusion so I went to work.

Attached is my revision of the file in question for review. The changes are around line 270 in the area for CV11. I changed the selections from Yes No to On Off and added multiple timing for the On. I also modified some of the nomenclature. I am not sure if there are other files that relate to this as I don't have a roadmap of the file layout. I put it in the Decoders folder where directed for modified decoder files and it reads fine.

Can you review this or provide it to one who is familiar with Digitrax?

I find that Digitrax did not implement the NMRA standard for CV11 as it was a recommendation and instead used it for their sound time out setting which matches their manual.
[Digitrax_0SFX_Series6.zip](https://github.com/JMRI/JMRI/files/3214989/Digitrax_0SFX_Series6.zip)

bobjacobsen, 2019-05-31:
Diff of the attached file:
```
% diff xml/decoders/Digitrax_0SFX_Series6.xml ~/Downloads/Digitrax_0SFX_Series6.xml 
20c20,21
<   <version author="Alain Le Marchand" version="4" lastUpdated="20170218"/>
---
>   <version author="Alain Le Marchand" version="4" lastUpdated="20170218"/>  
>   <version author="James Exler" version="5" lastUpdated="20190521"/>  
38a40,41
>   <!--  Version 5 (James Exler jimXexler@hotmail.com)
>     Revised choices and selections at CV11 (Sound Time Out) as they were reversed and confusing. Changed selection tabs to On/Off and time out to 06, 30 or 60 sec -->
265c268
<       <variable CV="11" tooltip="Sound stays on after loco is de-selected" item="Sound Option 7" default="6">
---
>       <variable CV="11" tooltip="Sound times out after loco is de-selected or idle" item="Sound Option 7" default="60">
267,269c270,272
<           <enumChoice choice="Yes" value="0">
<             <choice>Yes</choice>
<             <choice xml:lang="it">Si</choice>
---
>           <enumChoice choice="Off" value="0">
>             <choice>Off</choice>
>             <choice xml:lang="it">spento</choice>
271,272c274,281
<           <enumChoice choice="No" value="6">
<             <choice>No</choice>
---
>           <enumChoice choice="On:06" value="06">
>             <choice>On:06</choice>
>           </enumChoice>
>           <enumChoice choice="On:30" value="30">
>             <choice>On:30</choice>
>           </enumChoice>  
>           <enumChoice choice="On:60" value="60">
>             <choice>On:60</choice>
275c284
<         <label>Sound Time Out</label>
---
>         <label>Sound Time Out (seconds)</label>
```

This is in PR #7040

MARCFOUR, 2019-06-03:
There was no contradiction in the original definition. The CV11 was added to definition in 2007. The intent of CV11 in Digitrax sound decoders is as per the Digitrax Tech Ref V2 manual dated Jan 2014.  There were two "recommended " values:  '0' or '6', for CV11.

Value of '6' is "To shutdown sound when decoder is released" . Sound stops 6 seconds after release.
Value of '0' is"  Do not shutdown sound when decoder is released". 

The tooltip was clear and proper value defined.

'6' is the default value after a reset to this decoder, not '60' as he states. I certainly do not want 60 as a default.

Personal preferences should not dictate definition changes.

rhwood, 2019-06-03:
While I agree that the default value should be 6, not 60, to match the documented default, I do not agree that the tooltip was correct. The referenced manual states on page 77 that the decoder sound stops after not receiving a packet for the specified time, and that some systems merely do not send packets to idle locomotives, so the experience of a Digitrax decoder on a non-Digitrax command station may be that allowing the engine to idle long enough is the trigger for sound stopping.

jimexler, 2019-06-03:
Marc and all,

I agree the default should be 6 and not 60. I accidentally left that in the file from my testing and forgot to put it back to the default. Defaults should be sacred and used as the basis whenever possible.

The contradiction to me is:

The title of the CV definition is "Sound Time Out". The tooltip states "sound stays on...". In my opinion, that is reversing the logic and therefore I changed the tooltip to "Sound times out..."; staying parallel with the definition.

The Yes/No seemed weak to me and I felt that On/Off better defined the action of the CV; the "time out" is either on or off.

I read in one of the discussions (can't remember where) that the number represented the number of seconds of delay before the sound timed out so I experimented with my decoder and that seemed to be true. Based on that, I included the default 06 time out and added two other choices of 30 and 60. I also added (seconds) to the label so the number would be relative.

I am new to this group and appreciate the user comments. Our goal should be to have the best product available for the community.

Jim Exler, Nampa, ID

On 6/3/2019 3:12 AM, MARCFOUR wrote:

There was no contradiction in the original definition. The CV11 was added to definition in 2007. The intent of CV11 in Digitrax sound decoders is as per the Digitrax Tech Ref V2 manual dated Jan 2014. There were two "recommended " values: '0' or '6', for CV11.

Value of '6' is "To shutdown sound when decoder is released" . Sound stops 6 seconds after release.
Value of '0' is" Do not shutdown sound when decoder is released".

The tooltip was clear and proper value defined.

'6' is the default value after a reset to this decoder, not '60' as he states. I certainly do not want 60 as a default.

Personal preferences should not dictate definition changes.

—

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-green-avg-v1.png]<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> Virus-free. www.avg.com<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient>

bobjacobsen, 2019-06-03:
> I agree the default should be 6 and not 60. I accidentally left that in the file from my testing and forgot to put it back to the default. Defaults should be sacred and used as the basis whenever possible. 

I’ll make that change for JMRI 4.15.8

MARCFOUR, 2019-06-03:
Randal, the solution to that problem ( if one wants the sound to continue ) is to set CV11= 0 which was the YES of the original definition file and the logic of the tooltip. That or correct the command station in question so that it sends packets while the engine/decoder is Idle.  That was all fine when there were no sound decoders around, not the case anymore. 

Taken from any of the Digitrax Sound decoder instruction sheets and the Digitrax Tech Ref, exact wording. 

CV11 = Sound Time Out, 
06 = Sound ends when loco address is de-selected, 
00 = Sound stays on after loco is de-selected.

So the tooltip was based on a YES which equals a numeric value of 0 and follows the description of the Digitrax documentation.

My point was the definition should follow the available documentation and not be "interpreted".

dheap, 2019-06-03:
> @MARCFOUR That or correct the command station in question so that it sends packets while the engine/decoder is Idle. That was all fine when there were no sound decoders around, not the case anymore.

@MARCFOUR There is a very good reason some command stations stop sending packets after the loco has been at speed 0 for some time. It is to remove unused locos from the track queue (the loop of locos that need regular speed packets to keep moving). These command stations don't have a concept of slots to be assigned/released/dispatched.

The NCE systems are such a case. For a Power Pro the track queue has a size limit of 250 locos. By default a loco is removed from the track queue after 10 speed zero packets are sent. This can be shortened/lengthened/disabled but there is a warning that if you disable it, have a large layout and leave the command station powered up you may eventually end up unable to run new locos. Despite the fact that there are only 63 possible physical throttles available, you can connect an unlimited number of JMRI throttles (and there is no way of dispatching those either).

As far as the sound-timeout, many of us regard this as non-prototypical (a prime mover is started/shut down by a delilberate action on the part of the driver/engineer) and at least one decoder manufacturer (ESU) deliberately does not provide an automatic shutdown feature.

I'm not trying to get into brand wars, just pointing out that the DCC world doesn't all think the Digitrax Way and that DCC manufacturers don't have to "correct" their systems to behave the Digitrax Way.

rhwood, 2019-06-03:
@MARCFOUR The manual does not state that only 0 and 6 are possible values; merely that 6 is the default. The language in http://www.digitrax.com/tsd/KB721/sound-from-sound-fx-decoder-shuts-down-on-non-digi/ is that 6 is the default timeout value when idle, not the only non-zero timeout value.

Randall Wood

closed 2021-04-14 by stale[bot]