JMRI/JMRI #6992: Two Logix at same time Error Dialogue

icklesteve, 2019-05-25:
Clean startup, No tables loaded, no other PanelPro windows loaded.
Tables > Logix > Add
System Name: leave auto checked
UserName : Needs to be valid
Click on Create

Edit Logix Window opens, click Delete

[image]

But we've only been working with 1 Logix and did not attempt to edit it.
The Logix does not get deleted from the table as per the Edit Logix Window Delete button, 
though can be deleted via other methods.

You can add a few conditions and the same popup is obtained.

dsand47, 2019-05-26:
I have confirmed the issue and will have a fix soon.

icklesteve, 2019-05-29:
Many thanks @dsand47 , confirmed resolved

closed 2019-05-29 by icklesteve