JMRI/JMRI #7015: CI unstable

silverailscolo, 2019-05-28:
Since build 75 Jenkings CI AllTest shows regulare failures. Stack trace points to ``apps.DecoderPro.DecoderProTest``. When run that locally I get some migration dialogs that could be the cause of timeouts:
[image]

[image]

[image]

Console: 
```
INFO  - Starting with profile Prevent_Init_Loop.3f01644a [Time-limited test] apps.Apps.<init>()
WARN  - passing to xmit: unexpected exception:  [LnPowerManager LnTrackStatusUpdateThread] jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage()
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:117)
	at jmri.jmrix.loconet.LnPowerManager$LnTrackStatusUpdateThread.run(LnPowerManager.java:209)
WARN  - passing to xmit: unexpected exception:  [LnSensorUpdateThread] jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage()
java.lang.NullPointerException
	at jmri.jmrix.loconet.LnPacketizer.sendLocoNetMessage(LnPacketizer.java:117)
	at jmri.jmrix.loconet.LnSensorManager$LnSensorUpdateThread.run(LnSensorManager.java:294)
WARN  - testLaunchInitLoop(apps.DecoderPro.DecoderProTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
INFO  - DecoderPro version 4.15.7ish+me+20190528T1303Z+R5dc2b1f19d starts under Java 1.8.0_66 on Mac OS X x86_64 v10.13.6 at Tue May 28 15:07:58 CEST 2019 [Time-limited test] apps.Apps.setStartupInfo()
INFO  - Starting with profile DecoderPro.3f7227b7 [Time-limited test] apps.Apps.<init>()
INFO  - No saved preferences, will open preferences window.  Searched for /private/var/folders/5t/ztqyl52x3q52m0fp3hdr30hc0000gn/T/junit1239694361218807195/junit3128878949369945488/DecoderPro.jmri/ProfileConfig.xml [Time-limited test] apps.Apps.<init>()
WARN  - testLaunchInitLoop(apps.DecoderPro.DecoderProTest) : run  2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - testLaunchInitLoop(apps.DecoderPro.DecoderProTest) : giving up after 2 failures [main] util.junit.rules.RetryRule.evaluate()
INFO  - DecoderPro version 4.15.7ish+me+20190528T1303Z+R5dc2b1f19d starts under Java 1.8.0_66 on Mac OS X x86_64 v10.13.6 at Tue May 28 15:08:24 CEST 2019 [Time-limited test] apps.Apps.setStartupInfo()
INFO  - Starting with profile Sprog_Simulator.3d40b3b8 [Time-limited test] apps.Apps.<init>()
WARN  - testLaunchSprog(apps.DecoderPro.DecoderProTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
INFO  - Starting with profile Sprog_Simulator.3d40b3b8 [Time-limited test] apps.Apps.<init>()
INFO  - Starting with profile tmcc.3f0b414a [Time-limited test] apps.Apps.<init>()
INFO  - Starting with profile LocoNet_Simulator.3eac0cdc [Time-limited test] apps.Apps.<init>()
INFO  - Starting with profile EasyDcc_Simulator.3f033c2b [Time-limited test] apps.Apps.<init>()
------------- ---------------- ---------------
Testcase: testLaunchInitLoop(apps.DecoderPro.DecoderProTest):	Caused an ERROR
test timed out after 20 seconds
org.junit.runners.model.TestTimedOutException: test timed out after 20 seconds
	at java.lang.Thread.sleep(Native Method)
	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:418)
	at apps.LaunchJmriAppBase.runOne(LaunchJmriAppBase.java:75)
	at apps.DecoderPro.DecoderProTest.testLaunchInitLoop(DecoderProTest.java:44)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.lang.Thread.run(Thread.java:745)

Test apps.DecoderPro.DecoderProTest FAILED
```

bobjacobsen, 2019-05-31:
This should be better now, as a couple updates have been merged.

silverailscolo, 2019-05-31:
Thanks for the efforts, closing for now.

closed 2019-05-31 by silverailscolo