JMRI/JMRI #7016: JMRI Web Server Error 500: internal car handing error

samusi01, 2019-05-28:
Per the request of Randall Wood I am creating this issue. I am running 4.15.6 on one laptop and viewing various functions on a tablet attached to the layout. This evening, I attempted to load up the cars table to inventory and received the following message:

Error 500:Internal car handling error. See JMRI logs for information. 

I receive the same message with the locations and engines tables (I don't have anything presently set up in the other tables). Roster seems to be ok, and the Operations pages seem to be functioning normally. I am receiving this error on the remote tablet when using both Microsoft Edge browser and Chrome. I have attached a copy of my Operations Pro backup.
[2019_05_28_00 (Web Server Error).zip](https://github.com/JMRI/JMRI/files/3228101/2019_05_28_00.Web.Server.Error.zip)

rhwood, 2019-05-29:
@samusi01 I am able to recreate the error in 4.15.6, but am not able to create it in the "nightly" builds. Can you test that the JMRI installed from http://jmri.tagadab.com/jenkins/job/Development/job/Packages/lastSuccessfulBuild/artifact/dist/release/ works for you?

samusi01, 2019-05-29:
@rhwood Partly successful. Using the build JMRI.4.15.7ish+jenkins+20190528T2037Z+R2a0fb41, the tables that I do use (again, that's limited at the moment to cars, engines, and locations) function normally. On the operations page, the conductor view works and functions normally but the manifest page doesn't. It sits on "getting trains..." The roster page, which I don't normally use, seems to work very slowly at best on both Chrome and Edge, with delays, pages such as the web throttle not loading or loading slowly if access for the second time, etc.

rhwood, 2019-05-30:
Unfortunately, this seems to be something that needs to be documented. The HTML manifests for JMRI older than 4.15.6 cannot be used with current JMRI and vice versa. I'll make sure the release notes explicitly note this and the following workaround:

If you do not want to terminate and rebuild each train, you will need to delete existing manifests and then individually preview each train to generate a new manifest for each train (I apologize for this). To delete the manifests:
1. Open the "Help->Locations..." menu item (on most JMRI windows).
2. Click "Open User Files Location"; this will open a window in your default file system explorer application showing the JMRI application data.
3. Enter the "operations" directory.
4. Delete the directories "manifests", "csvManifests", "jsonManifests".
5. In the Operations Trains window, individually build or preview each train to rebuild the manifest for that train.

@DanielBoudreau Do you have differing advice for forcing all manifests to be rebuilt?

PR #7042 (2019-06-01): doc: Note about operations web access
Fixes #7016 by adding a note (that should be a warning) about the fact that web access for operations manifests will only work after trains are built in 4.15.7/4.16 once 4.15.7/4.16 is installed.
closed 2019-06-01 by bobjacobsen