JMRI/JMRI #7017: Time to disable LinkedWarrantTest?

bobjacobsen, 2019-05-28:
I've restarted 17 cases of Travis failures in `LinkedWarrantTest` tests over the past weekend.  Is it time to disable this test file?

I've included a summary of the symptoms below to centralize them.

```
[ERROR]   LinkedWarrantTest.testLinkedMidScript:351 "Tinker finished first leg" did not occur in time
```

```
5868    [junit] Testcase: testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest):	FAILED
5869    [junit] "Evers starts to move at 8th command" did not occur in time
5870    [junit] junit.framework.AssertionFailedError: "Evers starts to move at 8th command" did not occur in time
5871    [junit] 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:426)
5872    [junit] 	at jmri.jmrit.logix.LinkedWarrantTest.testLinkedMidScript(LinkedWarrantTest.java:357)
5873    [junit] 	at jmri.util.junit.rules.RetryRule$1.evaluate(RetryRule.java:50)
```

```
[ERROR]   LinkedWarrantTest.testLinkedMidScript:351 "Tinker finished first leg" did not occur in time
```

```
[ERROR]   LinkedWarrantTest.testLinkedMidScript:383 "Chance starts to move at 8th command" did not occur in time
```

```
[ERROR] jmri.jmrit.logix.LinkedWarrantTest.testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest)
[ERROR] Run 1: LinkedWarrantTest.testBackAndForth:273 "WestToEastLink finished third leg" did not occur in time
[ERROR] Run 2: LinkedWarrantTest.tearDown:421 Unexpected ERROR or higher messages emitted: null

...

[INFO] Running jmri.jmrit.logix.LinkedWarrantTest
ERROR - paint failed in thread AWT-EventQueue-1 9722: [AWT-EventQueue-1] jmri.jmrit.display.Editor.paint()
java.lang.ArrayIndexOutOfBoundsException: No such child: 33
at java.awt.Container.getComponent(Container.java:334)
at javax.swing.JComponent.rectangleIsObscured(JComponent.java:4390)
at javax.swing.JComponent.paint(JComponent.java:1054)
at javax.swing.JLayeredPane.paint(JLayeredPane.java:586)
at jmri.jmrit.display.Editor$TargetPane.paint(Editor.java:658)
at javax.swing.JComponent.paintToOffscreen(JComponent.java:5210)
at javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:290)
at javax.swing.RepaintManager.paint(RepaintManager.java:1272)

WARN - testLoopedWarrant(jmri.jmrit.logix.LinkedWarrantTest) : run 1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
WARN - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run 1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
WARN - No room for Train goBack ramp to speed "Stop" in block "East", warrant= EastToWestLink [Thread-9043] jmri.jmrit.logix.Warrant.setMovement()
WARN - testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) : run 1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - paint failed in thread AWT-EventQueue-1 9722: [AWT-EventQueue-1] jmri.jmrit.display.Editor.paint()
java.lang.ArrayIndexOutOfBoundsException: No such child: 31
at java.awt.Container.getComponent(Container.java:334)
at javax.swing.JComponent.rectangleIsObscured(JComponent.java:4390)
at javax.swing.JComponent.paint(JComponent.java:1054)

WARN - No room for Train goBack ramp to speed "Stop" in block "East", warrant= EastToWestLink [Thread-9050] jmri.jmrit.logix.Warrant.setMovement()
WARN - testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) : run 2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) : giving up after 2 failures [main] util.junit.rules.RetryRule.evaluate()
[ERROR] Tests run: 5, Failures: 2, Errors: 0, Skipped: 0, Time elapsed: 329.833 s <<< FAILURE! - in jmri.jmrit.logix.LinkedWarrantTest
[ERROR] testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) Time elapsed: 75.548 s <<< FAILURE!
java.lang.AssertionError: "WestToEastLink finished third leg" did not occur in time
at org.junit.Assert.fail(Assert.java:88)
at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:426)
at jmri.jmrit.logix.LinkedWarrantTest.testBackAndForth(LinkedWarrantTest.java:273)
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

[ERROR] testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) Time elapsed: 75.55 s <<< FAILURE!
java.lang.AssertionError: Unexpected ERROR or higher messages emitted: null
at org.junit.Assert.fail(Assert.java:88)
at org.junit.Assert.assertTrue(Assert.java:41)
at org.junit.Assert.assertFalse(Assert.java:64)
at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:334)
at jmri.jmrit.logix.LinkedWarrantTest.tearDown(LinkedWarrantTest.java:421)
at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
```

```
[ERROR]   LinkedWarrantTest.testBackAndForth:273 "WestToEastLink finished third leg" did not occur in time
```

```
[ERROR]   LinkedWarrantTest.tearDown:421 Unexpected ERROR or higher messages emitted: null
```

```
[ERROR]   LinkedWarrantTest.testLinkedMidScript:378 "Evers finished second leg" did not occur in time

[INFO] Running jmri.jmrit.logix.WarrantManagerTest
[INFO] Tests run: 1, Failures: 0, Errors: 0, Skipped: 0, Time elapsed: 0.001 s - in jmri.jmrit.logix.WarrantManagerTest
[INFO] Running jmri.jmrit.logix.LinkedWarrantTest
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : giving up after 2 failures [main] util.junit.rules.RetryRule.evaluate()
WARN  - No room for Train goBack ramp to speed "Stop" in block "East", warrant= EastToWestLink [Thread-9053] jmri.jmrit.logix.Warrant.setMovement()
WARN  - testBackAndForth(jmri.jmrit.logix.LinkedWarrantTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
[ERROR] Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 246.957 s <<< FAILURE! - in jmri.jmrit.logix.LinkedWarrantTest
[ERROR] testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest)  Time elapsed: 74.304 s  <<< FAILURE!
java.lang.AssertionError: "Evers finished second leg" did not occur in time
	at org.junit.Assert.fail(Assert.java:88)
	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:426)
	at jmri.jmrit.logix.LinkedWarrantTest.testLinkedMidScript(LinkedWarrantTest.java:378)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
```

```
[ERROR] Failures: 
[ERROR]   LinkedWarrantTest.testLinkedMidScript:357 "Evers starts to move at 8th command" did not occur in time

[INFO] Running jmri.jmrit.logix.LinkedWarrantTest
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : giving up after 2 failures [main] util.junit.rules.RetryRule.evaluate()
[ERROR] Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 191.706 s <<< FAILURE! - in jmri.jmrit.logix.LinkedWarrantTest
[ERROR] testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest)  Time elapsed: 55.342 s  <<< FAILURE!
java.lang.AssertionError: "Evers starts to move at 8th command" did not occur in time
	at org.junit.Assert.fail(Assert.java:88)
	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:426)
	at jmri.jmrit.logix.LinkedWarrantTest.testLinkedMidScript(LinkedWarrantTest.java:357)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
```

```
[INFO] Running jmri.jmrit.logix.LinkedWarrantTest
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  1 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
WARN  - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : run  2 failed, RetryRule repeats [main] util.junit.rules.RetryRule.evaluate()
ERROR - testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest) : giving up after 2 failures [main] util.junit.rules.RetryRule.evaluate()
[ERROR] Tests run: 4, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 158.351 s <<< FAILURE! - in jmri.jmrit.logix.LinkedWarrantTest
[ERROR] testLinkedMidScript(jmri.jmrit.logix.LinkedWarrantTest)  Time elapsed: 52.255 s  <<< FAILURE!
java.lang.AssertionError: "Tinker finished first leg" did not occur in time
	at org.junit.Assert.fail(Assert.java:88)
	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:426)
	at jmri.jmrit.logix.LinkedWarrantTest.testLinkedMidScript(LinkedWarrantTest.java:351)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
```

pabender, 2019-05-28:
Perhaps.  We really haven’t gotten past the issue spotted in #6912....

But I just had an idea.  I think the issue is the interaction between the warrant and the panel.  So what if the panel isn’t visible during the test run.  That would prevent the issues with repainting the panel, because the panel will not need repainting.

pabender, 2019-05-28:
#7018 hides the panels early in the tests.  Hopefully that helps.

geowar1, 2019-11-05:
I'm still having to restart the tests often because of the LinkedWarrantTest failure. Can this be fixed or disabled?

bobjacobsen, 2020-01-08:
This has been done through the jmri.skipTestsRequiringSeparateRunning mechanism.

closed 2020-01-08 by bobjacobsen