JMRI/JMRI #7034: Rename Light Control

icklesteve, 2019-05-30:
I'm doing some updates to JMRI Light Controls ( automated Sensor / FastClock followers ), and have spotted that there are 2 JMRI _things_ called "Light Control".

There's the Light automation control beantable stuff, eg.
[image]

[image]

and there's the JMRI Light Control Tool, jmrit.simplelightctrl eg.

[image]

I think we need to change the name of one of them to avoid confusion.
Personally, I'd change the Light Control Tool name, though needs to match the Turnout Control Tool?

silverailscolo, 2019-05-30:
We might rename the Light Control Tool to "Simple Light/Turnout Control", "Manual Light/Turnout Control", "Operate Light/Turnout" or include the "Tool" part in the name and menus so you would get "Light/Turnout Control Tool". I would agree to change the Tool items, and go for "Simple..."

bobjacobsen, 2019-05-31:
The original Turnout Control Tool was called “tool” because it opened a little window with closed/thrown buttons. Before there were panels or tables, you could have a bunch of those on your screen to run your layout.

Now, people generally use panels for that. The Turnout Control Tool is more of a one-off for ad-hoc use, testing, etc.  

I think it would be fine to rename both light and turnout control tools for 4.7.1.

bobjacobsen, 2019-05-31:
But formally, the things in those screen shots are not _controls_ (the button or knob you manipulate to change state) but rather _controllers_ (who act themselves to change state).

GnTim, 2019-06-03:
The initial problem is only partly solved. See: [openlcb] 4.15.7 Lights bug
Looks to me like the xml is not being stored correctly or the xml is not being read correctly OR the interface between the table and the clock read is broke. I hope a developer can take this on and resolve the issue.
Thanks.

icklesteve, 2019-06-04:
Hi @GnTim This is a thread discussing some window names, was there supposed to be a link after you wrote See: [openlcb] 4.15.7 Lights bug?

I'm not able to re-create the OpenLCB Light Control issue in #7037 if that's what you're referring to?

If there is still a problem with OpenLCB Lights, please create a new issue with a description of how to recreate the problem, or provide more information on the JMRI Users Group.

Note there is an improvement for handling multiple fast-clock times planned for release 4.17.1 on #7052 , with discussion for this on #5316 

Steve.

icklesteve, 2019-06-09:
Hi @silverailscolo ,

I think **Simple** would work, also matches **Simple** Signal Logic :-)

silverailscolo, 2021-02-23:
The automation items have been renamed to ...controllers since. Closing this.

closed 2021-02-23 by silverailscolo