JMRI/JMRI #7043: Create Test Release 4.15.8

bobjacobsen, 2019-06-02:
This is hopefully the last test release of the 4.16 cycle.  It will include _only_ specific PRs. If you have something you'd like included, attach a note here and create your PR based on the `v4.15.7` tag or earlier.

silverailscolo, 2019-06-02:
PR #7045 intends to improve CI, so it might be useful from 4.15.8.

bobjacobsen, 2019-06-10:
The process for building this is being developed in a JMRI-developers thread:  https://jmri-developers.groups.io/g/jmri/message/1346 and following posts.

bobjacobsen, 2019-06-12:
The release-4.15.8 branch has been created. 

Please set the 4.17.1 milestone on new PRs from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://builds.jmri.org/jenkins/job/TestReleases/job/4.15.8/)

bobjacobsen, 2019-06-15:
JMRI 4.15.8 has been released. Files are available in the GitHub release section.

closed 2019-06-23 by bobjacobsen