JMRI/JMRI #7047: Windows (un)installer could use update

bobjacobsen, 2019-06-02:
From a jmri-developers post by @devel-bobm:

> An Windows install of 4.15.7 over the top of 4.15.2, followed by an
uninstall, leaves behind a variety of files, including some help files, a
couple of lib/*.jar files, some of the files from the web directory, and
three xml/decoders/*/*.xml files.  
>
> I know that those lib/*.jar files need to be specifically added to the
InstallJMRI.nsis file.  Perhaps some of those other files need to be added,
too.

rhwood, 2019-06-02:
See also discussion at #359

devel-bobm, 2019-06-03:
It appears that the JMRI Windows "installer/uninstaller" is used to install over the top of an existing install, it ill use the "uninstaller" operations from the build-being-installed, not the "uninstaller" operations from the build-being-installed-over.

In my example (quoted in the initial @bobjacobsen description), the older install, 4.15.2, had several files which are _not_ present in 4.15.7.  A cursory glance suggests that the files which were _left-behind_ by the "install 4.15.7 over 4.15.2 then uninstall 4.15.7" process may be those files which were present in 4.15.2 but not present in 4.15.7.

I wonder if it may make more sense for the JMRI installer to execute the "existing uninstaller" (from the existing installer) instead of runnig the "current uninstaller" process before installing the "current installer" process.  I _assume_ that this is something which NSIS is capable of.  @mattharris - does this possible solution make sense to you?

rhwood, 2019-06-03:
I think it is better to simply remove the entire JMRI installation directory before install, and as part of the uninstallation. This will radically simplify installation/uninstallation logic, allow developers to worry about features without also worrying about how a single platforms installer handles those features. The presumption that we should be supporting JMRI installations that have been modified to not be what we ship simply leads to significant support cycles that we’d rather not be spending.

bobjacobsen, 2020-01-08:
This was updated for 4.19.1, which (tentatively) closes this.

closed 2020-01-08 by bobjacobsen