JMRI/JMRI #7050: Possible RFID issues in 4.15.7

bobjacobsen, 2019-06-03:
From a [jmriusers post](https://groups.io/g/jmriusers/message/160494):

> I've been experimenting with the latest test release 4.15.7 and noted that RFID reporting on a layout panel now shows the Tag ID (e.g. ID0418AC8103) rather than the 'English' User Name (e.g. Tanker).
>
> The RFIDs display fine with 4.14.
>
> I'm running on Ubuntu 19.04 with openjdk version "11.0.3" 2019-04-16
The system uses a MERG Reporter over CBUS.
>
> Looking into it a bit further I note that in 4.14 when I edit the ID Tags table to add a User Name both the User Name and System Name change in tandem to reflect the new 'user' name. The Tag ID column shows the ID  (e.g. 0418AC8103 - without the 'ID').
> 
> In 4.15.7 the ID Tags table shows the System name as 'ID0418AC8103' and User name as 'Tanker'.
> 
> A possible explanation (?) is that the layout panel has been showing System name all along (rather than User name), but with the new version the system name is showing correctly in the table. I've included an extract from the ID Tags xml below just showing one tag.
> 
> I'm not sure sure if its directly connected, but when I open the ID Tags table in 4.15.7 there are no tags listed. However, as soon as an RFID tag is read/reported the complete list pops up in the table dialog!
> 
> This does not happen in 4.14 where the ID Tags list is shown as expected when first accessed (even before any tags read in this session).
> 
> I tried to look back a few versions to see when the change in behaviour occurred. 4.15.6 behaves in the same way. However, earlier versions started to give me other exceptions on loading, so I wasn't sure how useful following that tack would be. 
> 
> The only thing I found possibly relating to this issue in the release notes was about memory variable naming. I confirmed that all the ones I use on the layout panel do begin with 'IM' as required.
> 
> At the moment this issue prevents me from using the latest test version.
> 
> Regards
> Andy
```
<?xml version="1.0" encoding="UTF-8"?>
<?xml-stylesheet href="/xml/XSLT/idtags.xsl" type="text/xsl"?>
<idtagtable xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="http://jmri.org/xml/schema/idtags.xsd">
  <!--Written by JMRI version 4.14+Rd060e0b on Mon Jun 03 13:15:58 BST 2019-->
  <configuration>
    <storeState>yes</storeState>
    <useFastClock>no</useFastClock>
  </configuration>
  <idtags>
    <idtag>
      <systemName>ID0418AC8103</systemName>
      <userName>Tanker</userName>
      <comment>LF RFID Tag - Yellow fuel tanker</comment>
      <whereLastSeen>MR3</whereLastSeen>
      <whenLastSeen>3 Jun 2019, 13:06:04</whenLastSeen>
    </idtag>
```

icklesteve, 2019-06-03:
Same behaviour in v 4.15.4, ie the table starts up empty, which rules out #6914 causing the issue.

Perhaps a side effect of #6493 @pabender ?

pabender, 2019-06-03:
> Same behaviour in v 4.15.4, ie the table starts up empty, which rules out #6914 causing the issue.
> 
> Perhaps a side effect of #6493 @pabender ?

Possibly, but more likely #6466 or #6483.

icklesteve, 2019-06-03:
Thanks Paul,

Would it be possible to display the IdTag Table contents?
I'm afraid it definitely feels like a bug when I've just been comparing the behaviour of 14.4 vs 15.7ish.
If the other tables lazy loaded entries, the behaviour would be expected, but the empty table is unexpected when the other tables are all loaded.

pabender, 2019-06-03:
@icklesteve,

None of the other tables are loaded by default.  Generally, you either have to create an entry or you have to load a file manually.

The code prior to the 4.15.x updates DID have a bug in it where the system name was updated in the table when the user ID was updated.  Clearly that isn't right.

The reason the file loaded in version 4.14 was that the DefaultIdTag manager was loaded and initialized directly at startup.  Now, the ProxyIdTag manager is loaded instead.  The type specific IDTag managers (currently DefaultIdTagManager, DefaultRailComManager, and TranspondingTagManager ) are loaded as required (DefaultIdTagManager is the default).  This lets us keep track of all the different types of IdTags in a single table (before the changes, the IDTag table and the RailCom table were completely separate.  Eventually, I want to remove the RailCom table).

Now, for some behavior verification testing.  I am using a standalone ID20LA as my RFID reader with head of master for some testing, and here is what I see....  

After a fresh start of PanelPro I load:  

1. a ControlPanelEditor panel with only ReporterIcon
2. The Reporters Table
3. The ID Tags table

After those three items are displaying:

1. The Reporters table contains one entry (for the reporter defined in the panel). it contains no current report and no last report
2. The ID Tags Table is empty
3. The ReporterIcon correctly shows <no report> 

As soon as a tag is read in by the reporter, I see:

1. The IdTags Table is populated from the file.  
2. The ReporterTable entry for the reporter shows (brefly ) a report of the user name associated with the tag and a Last Report of the user name associated with the tag.
3. The ReporterIcon shows (brefly ) a report of the user name associated with the tag and then returns to <no report>.

All of this is the expected behavior.

So I am back to needing to know how exactly the user is displaying his IdTags on the panel.

pabender, 2019-06-03:
I just checked with a Block Contents Label on a LayoutEditor Panel and it also behaves correctly.

The user reported that he is using a Memory.  The Memory has to be loaded by some process and it may be one we didn't supply (i.e. a user written script).  I think the issue is that the user needs to update his script to populate the memory with the value of tag.toReportString() instead of the value of tag.toString().  tag.toString() was modified to conform with other NamedBean objects.

I don't think there is a bug here, but the release note is missing any mention of this set of changes.  I'll work on drafting that (I thought I already had, but it clearly isn't in place).

pabender, 2019-06-03:
Ok, so there is a note in the release note about 4.15.3 about the updates, but it doesn't explicitly mention using the toString vs toReportString.  a warning about that needs to be added.

icklesteve, 2019-06-03:
I'd expect that if the table has data within it, it's displayed when the user views the table.

By displaying an empty table it appears like all of the tags have been lost.
This is misleading

If you want to delete a tag offline you can't, instead you have to create a new tag to bring up the table, then delete both tags.
If you want to manually add a tag, you can't  see if the tag is already in the table.

Would it be possible to grey out / disable the table to say that it just hasn't been loaded yet?
This seems better than displaying an empty table?

bobjacobsen, 2019-06-03:
The real difference here is that people _don't_ manually load the tag file; it's done automatically.  So when should it be loaded?

It's true that not loading the file saves memory, but how often is that a useful thing?  If a particular setup uses tags, then it has to load the tags before using them. 

How about "if the tags are referenced, including when the table is opened, the contents are loaded"?  That could be done when there are references to the (proxy) manager.

pabender, 2019-06-03:
@icklesteve the empty table on start is another issue entirely. It isn't the issue the user is experiencing.

The real issue with the table is that the type specific managers have to be loaded and initialized to populate the tables.  The proxy manager doesn't do that.  The proxy manager has no idea which of the type specific managers it is appropriate to load until we ask for the specific type.  If we are going to load and initialize one tag manager at startup, we need to load and initialize all of them.

icklesteve, 2019-06-03:
No issues with displaying the correct tag username in panels from upgrade v4.14 to latest as far as I can tell, tested using only CBUS network commands and JMRI built-in tools.

pabender, 2019-06-03:
@bobjacobsen yes, we can certainly initialize all the managers if the table is loaded.

This is not an issue that needs to be addressed prior to version 4.16 however.

pabender, 2019-06-03:
Hmmm... so the user is using a memory, but he is populating that memory using a jmri.jmrit.display.layoutEditor.LayoutBlock object.  

The memory is updated when the block contents change, but it is updated with the toString of the object instead of just copying the object to the memory and letting the memory icon decide how to decode the value.... 

more to follow.

pabender, 2019-06-03:
#7053 contains a correction for the display of Reportable objects in LayoutEditor MemoryIcon objects, including correctly updating the memory when updating from block contents.

pabender, 2019-07-23:
Closed via #7060

closed 2019-07-23 by pabender