JMRI/JMRI #7058: ESU LokSound 5 README pane errors.

dheap, 2019-06-04:
Could someone please take a look at this quite simple correction? I doubt I will be in a position to create a PR prior to the 4.15.8 deadline.

Details at:
https://groups.yahoo.com/neo/groups/loksound/conversations/messages/29553

Dave

dheap, 2019-06-04:
My current browser access version won't even let me set a Milestone 🙁

dheap, 2019-06-05:
Copied from [my post to LokSound group](https://groups.yahoo.com/neo/groups/loksound/conversations/messages/29553) (brain fade, thinking I was posting to jmrusers)

Line 218 in "xml/decoders/esu/v5readMePane.xml " currently reads:
<text>The LokSound 5 DCC/MM/SX/M4 decoders use the NMRA standard</text>

Line 218 should read:
<text>The LokSound 5 DCC decoders use the NMRA standard</text>
(This also needs correction in the translation lines immediately below)

I don't currently have practical access to create a PR so I'd appreciate it if one of the other developers could raise a correcting PR so this makes it into V4.15.8.

Note that the existing text at lines 233 ff.; "LokSound 5 DCC/MM/SX/M4" is correct.

bobjacobsen, 2019-06-06:
I'll do this.

bobjacobsen, 2019-06-06:
PR #7065

dheap, 2019-06-06:
@bobjacobsen Thanks very much. We should also apply the same change to lines 219, 220 & 221 (translations) as per my comment at #7065.

bobjacobsen, 2019-06-12:
Closed by #7065

closed 2019-06-12 by bobjacobsen