JMRI/JMRI #7059: ID Tags Table Preferences not maintained

icklesteve, 2019-06-04:
The 2 IDTag Table options,

Save when and where last seen
Use Fast Clock Time

[image]

Now operate on a session-only basis, unchecked by default.
Does not appear to be UI issue as the correct functions are carried out when the boxes have been checked for the session.

When previously checked in 4.14 are no longer checked when loaded in 4.15.8ish

Checking both options, Store Config to File, then restarting JMRI, 
both options are unchecked.

Checking both options, Edit Preferences, Save, then restarting JMRI, 
both options are unchecked.

No difference if IdTag Table has been ( auto ) populated or not.

pabender, 2019-06-04:
The data is actually being stored.  It is stored in the same file the ID tags are stored in.

This may very simply be a case of the data not being available when the table loads.

Also, look at the "internal" tab instead of the "all" tab.  I think you will see the checkboxes are checked there after the data is loaded, or at least if the data is loaded and then the table displayed.

pabender, 2019-06-04:
This is definitely not propagating to/from the individual tabs to the "all" tab.  There isn't any data stored for "all".

pabender, 2019-06-04:
#7060 loads the data at construction time.  It does not address propagation between tabs.

pabender, 2019-07-23:
Closed via #7060

closed 2019-07-23 by pabender