JMRI/JMRI #7064: RePosting 4.15.7 Light Issue

GnTim, 2019-06-05:
Using OpenLCB with RR-CirKits Signal LCC board
Fast Clock at 90:1 (for testing purposes)

I entered the information to operate one light. The light worked per clock time.
I entered two more lights. All three lights worked.
I saved the files using Save Configuration and Panels. (There is no layout panel.)
I later loaded the file and the lights again worked.
I added more lights for a total of 7. These worked.
I saved the files as before.
Two days later I opened the files and Light 1 gets a continuous stream of ON/OFF. I watched this in the OpenLCB Traffic Monitor and did about 30 seconds of logging. There were over 6,000 commands to the light.
Obviously this can be a OpenLCB traffic saturation issue. 
This is very important to me to get resolved, so I am willing to help in any manner except code as I retired from C++ about 10 years ago and never learned xml.
Thanks, GnTim

bobjacobsen, 2019-06-05:
It would probably help if you attached that configuration file to this issue.  You might have to add .txt to the name to be able to do that.

icklesteve, 2019-06-07:
Hi @GnTim 

I'm thinking this is the same issue as #5316 

```
<light minIntensity="0.0" maxIntensity="1.0" transitionTime="0.0">
      <systemName>ML02.01.57.00.01.B5.00.02;02.01.57.00.01.B5.00.03</systemName>
      <userName>Light-1</userName>
      <lightcontrol controlType="2" fastClockOnHour="1" fastClockOnMin="0" fastClockOffHour="1" fastClockOffMin="15" />
      <lightcontrol controlType="2" fastClockOnHour="3" fastClockOnMin="0" fastClockOffHour="3" fastClockOffMin="15" />
</light>
```

Currently with 2 Fast Clock Light Controls there is a clash whereby 1 LightControl requests ON, the other requests OFF.

This should be resolved when #7052 has been merged,
Steve.

icklesteve, 2021-03-12:
Closing issue pending further info.
Improved code for multiple LightControls with crossover / matching times has been in place and unit tested for a while now.

closed 2021-03-12 by icklesteve