JMRI/JMRI #7066: Duplication of labels in NamedBeanBundle.properties

rhwood, 2019-06-06:
`jmri/NamedBeanBundle.properties` has, for a number of Named Bean types, both "BeanNameFoo" and "ItemTypeFoo" strings. It would be a (slight) improvement to replace all uses of "ItemTypeFoo" with "BeanNameFoo" throughout the JMRI codebase.

danielb987, 2019-06-06:
The ItemTypeFoo items are for the enum ItemType in jmri.Conditional at line 182.

It will probably be enough to simple changing the strings in this enum and then remove these items. For example, changing:
`SENSOR(ITEM_TYPE_SENSOR, IsStateVar.IS_STATE_VAR, "ItemTypeSensor"),`
to:
`SENSOR(ITEM_TYPE_SENSOR, IsStateVar.IS_STATE_VAR, "BeanNameSensor"),`
and then remove these items, except `ItemTypeNone` and `ItemTypeOther`.

But I don't have the time to test this at the moment.

silverailscolo, 2021-02-23:
@danielb987 do you recall these duplicates have been migrated to `enums` as you suggested?

danielb987, 2021-02-24:
If I do a text search for all *.java files in the java/src folder, the only match for `ItemTypeSensor` is in jmri.Conditional at line 162.

```
    enum ItemType {
        NONE(TYPE_NONE, IsStateVar.IS_STATE_VAR, "ItemTypeNone"),        // There is no ITEM_TYPE_NONE so use TYPE_NONE instead
        SENSOR(ITEM_TYPE_SENSOR, IsStateVar.IS_STATE_VAR, "ItemTypeSensor"),
        TURNOUT(ITEM_TYPE_TURNOUT, IsStateVar.IS_STATE_VAR, "ItemTypeTurnout"),
        LIGHT(ITEM_TYPE_LIGHT, IsStateVar.IS_STATE_VAR, "ItemTypeLight"),
        SIGNALHEAD(ITEM_TYPE_SIGNALHEAD, IsStateVar.IS_STATE_VAR, "ItemTypeSignalHead"),
        SIGNALMAST(ITEM_TYPE_SIGNALMAST, IsStateVar.IS_STATE_VAR, "ItemTypeSignalMast"),
        MEMORY(ITEM_TYPE_MEMORY, IsStateVar.IS_STATE_VAR, "ItemTypeMemory"),
        CONDITIONAL(ITEM_TYPE_CONDITIONAL, IsStateVar.IS_STATE_VAR, "ItemTypeConditional"),  // used only by ConditionalVariable
        LOGIX(ITEM_TYPE_LOGIX, IsStateVar.IS_STATE_VAR, "ItemTypeLogix"),                    // used only by ConditionalAction
        WARRANT(ITEM_TYPE_WARRANT, IsStateVar.IS_STATE_VAR, "ItemTypeWarrant"),
        CLOCK(ITEM_TYPE_CLOCK, IsStateVar.IS_STATE_VAR, "ItemTypeClock"),
        OBLOCK(ITEM_TYPE_OBLOCK, IsStateVar.IS_STATE_VAR, "ItemTypeOBlock"),
        ENTRYEXIT(ITEM_TYPE_ENTRYEXIT, IsStateVar.IS_STATE_VAR, "ItemTypeEntryExit"),

        AUDIO(ITEM_TYPE_AUDIO, IsStateVar.IS_NOT_STATE_VAR, "ItemTypeAudio"),
        SCRIPT(ITEM_TYPE_SCRIPT, IsStateVar.IS_NOT_STATE_VAR, "ItemTypeScript"),
        OTHER(ITEM_TYPE_OTHER, IsStateVar.IS_NOT_STATE_VAR, "ItemTypeOther");

        ....
```
I guess that the only change you need to do is to change `ItemTypeFoo` to `BeanNameFoo` in these strings. Note the string `ItemTypeNone` on the first line.

closed 2021-06-09 by stale[bot]