JMRI/JMRI #7070: NamedBeanHandle Move / Layout Block with no Sensor Issue

dsand47, 2019-06-08:
After a user name move, a subsequent request with an **empty ("")** user name returns the bean that no longer has the user name.

The attached file contains a very simple test case.  After loading, move the sensor user name from IS101 to IS102.  The panel will be re-drawn when made active and the following error will occur.  Since the **paint** process is interrupted, the rest of the panel will not be drawn and it looks like a disaster.
```
     [java] 2019-06-07 22:22:30,843 layoutEditor.LayoutBlock              INFO  - getOccupancy part 1:  name = '', s = null [AWT-EventQueue-0]
**Move User name**
     [java] 2019-06-07 22:22:56,039 layoutEditor.LayoutBlock              INFO  - getOccupancy part 1:  name = '', s = IS101 [AWT-EventQueue-0]
     [java] 2019-06-07 22:22:56,041 layoutEditor.LayoutBlock              INFO  - getOccupancy: name = '', obj = IS101, sys = 'ILB1', user = 'Blk' [AWT-EventQueue-0]
     [java] 2019-06-07 22:22:56,042 display.Editor                        ERROR - paint failed in thread AWT-EventQueue-0 14:  [AWT-EventQueue-0]
     [java] java.lang.IllegalArgumentException: name cannot be empty in getNamedBeanHandle
     [java] 	at jmri.NamedBeanHandleManager.getNamedBeanHandle(NamedBeanHandleManager.java:60)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutBlock.getOccupancy(LayoutBlock.java:625)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutBlock.getBlockColor(LayoutBlock.java:468)
     [java] 	at jmri.jmrit.display.layoutEditor.LayoutTrack.getColorForTrackBlock(LayoutTrack.java:144)
```
The log.info statements were used to isolate the issue.

Fixing LayoutBlock#getOccupancy() is very easy.  It should never be requesting a sensor with an empty name.

The question is whether the NamedBeanHandle Move process is leaving an unexpected state
 for the source bean.

[NBH Test.xml.txt](https://github.com/JMRI/JMRI/files/3268184/NBH.Test.xml.txt)

pabender, 2019-06-08:
Sent from my iPad

> On Jun 8, 2019, at 12:55 AM, Dave Sand <notifications@github.com> wrote:
> The question is whether the NamedBeanHandle Move process is leaving an unexpected state
> for the source bean.
> 

According to the Javadoc for NamedBean, the user name cannot be empty, so if the move process is leaving an empty string for the user name, that is where the bug lies.

Paul

dsand47, 2019-06-08:
> According to the Javadoc for NamedBean, the user name cannot be empty, so if the move process is leaving an empty string for the user name, that is where the bug lies. Paul

Thanks.  This confirms my recent research that indicates that the NamedBeanHandle process detected an error that was caused elsewhere.

jmri.jmrit.beantable.BeanTableDataModel#moveBean sets the user name for the source bean to **empty("")** instead of **null**.  This error, combined with the inappropriate set sensor request with an empty string in LayoutBlock created the IllegalArgument exception in NamedBeanHandleManager.

closed 2019-11-26 by dsand47