JMRI/JMRI #7085: Block contents from report able objects

pabender, 2019-06-11:
There are two user driven use cases for reportable objects, and we are not handling one of them correctly.

The first use case is the use case of the old LocoNet reporter behavior and it's interaction with scripts such as ReporterFormatter.py.  The old LocoNet reporter produced a string report.  The toReportString method for the new TranspondingTag Objects (and the revised RailCom objects) produces the equivalent of the old LocoNet reporter reports (the toReportString is a tag ID + attributes).  This all works as intended.

The second use case is for equipment tracking on a panel.  In this case, we don't want to include the attributes of the report.  We only want the ID (typically either the UserID or the SystemID) to display.  This is how a plain IDTag works, but the plain IDTags don't have any other attributes.

The proposed solution is to have Memory and BlockContents Icons display the result of getDisplayName for any contents that is an IDTag derivative but to let a ReporterIcon continue to display the result of toReportString for all Reportable objects.  (the Reportable interface does not require an object to be an IdTag...)  the toReportString would continue to be available for scripts, such as ReporterFormatter.py (which actually loads a string in a memory...)

Thoughts before I continue?

icklesteve, 2019-06-11:
Hi Paul, 
Not sure whether this is relevant but here we go anyways . . .

I've been contemplating using Reporter locations for "double docking" on NX / IECC like panels.
 ie when a train arrives at a terminus station with one head-code, another train enters on the same platform, so often 2 TD's ( Train Describers ) are displayed and populated, even though there is only 1 occupancy block.
( There are London stations with 5 TD berths per platform )

I was going to see how easy it would be so if 1x reporter location had 2x ID Tags registered for it, then display both ID Tags ( or request a Memory variable to follow whatever what was in Reporter 1st In, or Reporter 2nd in, and insert that value as the Block content String for departing trains ).

Ultimately it's going to need a Reporter query, perhaps something along the lines of 
Reporter.getLabelFromLeftToRight(boolean leftFirstTrueRightFirstFalse, int n)

Perhaps KISS and just provide a reporter method to get ALL contents in 1 spaced out String?

Obviously very early rough thoughts on this, plan to play around with code at some point for the best way of achieving this.

Steve.

pabender, 2019-06-12:
> I've been contemplating using Reporter locations for "double docking" on NX / IECC like panels.
> ie when a train arrives at a terminus station with one head-code, another train enters on the same [...]

This is actually a different issue than what I am concerned about.

> Ultimately it's going to need a Reporter query, perhaps something along the lines of
> Reporter.getLabelFromLeftToRight(boolean leftFirstTrueRightFirstFalse, int n)

The general form of this is the CollectingReporter interface:
http://jmri.org/JavaDoc/doc/jmri/CollectingReporter.html

There are two merged implementations of this:
http://jmri.org/JavaDoc/doc/jmri/jmrix/internal/TrackReporter.html
http://jmri.org/JavaDoc/doc/jmri/jmrix/roco/z21/Z21CanReporter.html

and PR #7057 has a third implementation.

The Z21CanReporter and the LocoNetReporter (which is the one that isn't merged) reflect the status of the hardware, so we don't let the user's manipulate the underlying collection.

The TrackReporter allows a user to push and pull from each of two ends (designated East and West in the API).  Users get to decide what ends are appropriate to push and pull from at any given time, using scripts.  To model your subway berths, you just push from one end and pull from the other.

PR #7057 also includes a script that can be used to list the contents of the collection associated with the reporter.

I have layout where I use the TrackReporter to represent stub ended tracks in a yard and display the reporting marks of the next available car.   The code and layout for this are available here: 
https://github.com/pabender/Sensys2014

pabender, 2019-07-05:
Closed via #7172

closed 2019-07-05 by pabender