JMRI/JMRI #7089: Test failing on AppVeyor only

bobjacobsen, 2019-06-12:
Since some recent merges, including #7045 but perhaps due to another on around the, every AppVeyor run has failed with
```
7825    [junit] Testcase: testCtorMacOSX(jmri.plaf.macosx.EawtApplicationTest):SKIPPED: got: <false>, expected: is <true>
7826    [junit] Testcase: testLaunchGrapevine(apps.DecoderPro.DecoderProTest):	FAILED
7827    [junit] "first Info line seen in Console after startup" did not occur in time
7828    [junit] junit.framework.AssertionFailedError: "first Info line seen in Console after startup" did not occur in time
7829    [junit] 	at jmri.util.JUnitUtil.waitFor(JUnitUtil.java:427)
7830    [junit] 	at apps.LaunchJmriAppBase.runOne(LaunchJmriAppBase.java:75)
7831    [junit] 	at apps.DecoderPro.DecoderProTest.testLaunchGrapevine(DecoderProTest.java:39)
7832    [junit] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
7833    [junit] 	at java.lang.Thread.run(Thread.java:748)
7834    [junit] 
7835    [junit] 
7836    [junit] Testcase: testCTor(apps.InstallTest.InstallTestTest):SKIPPED: gives error message about an invalid profile
```

Interestingly, the other app start tests are ok, and this error does not seem to occur on Travis CI.  So it’s this one test on AppVeyor only.

pabender, 2019-06-12:
PR #7090 has my work to move the application start tests to cucumber.  Let's see if it has the same issue.

bobjacobsen, 2019-06-13:
An odd thing, perhaps a clue:

[image]

I'm running the current head of master on macOS. The first three times that `apps.DecoderPro.DecoderProTest` pops a console window, it has only the first line in the figure above.  The fourth time, it has the first two lines:  It's showing Sprog_Simulator.  The fifth (and last) time it opens the window, it's showing all three lines.  

So some state is not being properly cleared, and the console is picking it up.

pabender, 2019-06-13:
I just found one thing that could very well be leading to this.

All of these tests (both the original form and the Cucumber ones I am working on) call the shutdown manager, but the shutdown manager includes a static flag that says "i am shut down".  turning on DEBUG logging, you see that the shutdown manager runs for the first test, but it never runs for any subsequent tests because the flag has been flipped by that point.

pabender, 2019-06-13:
Still working on #7090, but I have now merged #7093, which includes the code to reset the shuttingDown flag and triggers that from the tearDown routine for the launch tests.

bobjacobsen, 2020-01-08:
This has been done through the jmri.skipTestsRequiringSeparateRunning mechanism.

closed 2020-01-08 by bobjacobsen