JMRI/JMRI #7103: JmriFaceless startup issues (discovered while working on #7090)

pabender, 2019-06-16:
JmriFaceless has a couple of startup issues I discovered while working on #7090 

First, the application assumes loconet is in use.  It's start routine starts the LocoNetOverTcp Server.

 https://github.com/JMRI/JMRI/blob/d73ea5ce7a91c0a1fd7f50c7f9f684bea89b537e/java/src/apps/JmriFaceless.java#L33-L39

This actually causes an error when you attempt to start the program with a profile that does not configure a LocoNet connection, so that needs to be removed.

Second, main sets the value of the system property java.awt.headless to true.

 https://github.com/JMRI/JMRI/blob/d73ea5ce7a91c0a1fd7f50c7f9f684bea89b537e/java/src/apps/JmriFaceless.java#L41-L50

This may or may not be a problem at runtime, but for testing this is leaked state that I'm not sure we can reliably clean up.  It also doesn't seem to have the desired impact.  Testing on a headed system using the same profiles we use for other application start tests, the windows (other than the main window, because there isn't one) still appear.

I think we should stop setting this system variable.

rhwood, 2019-06-16:
I think that, within a running JVM, if setting the java.awt.headless property to true of anything that drew to the screen already did so does not work, so while testing, unless it’s the very first thing tested, that will fail to change state.

On the other hand, the point of the app is to run headless, so removing that does break the app in a subtle way (in that if there is other logic in jmrit.* that checks for that property before doing something that would crash if headless can’t prevent it).

Randall Wood

pabender, 2019-06-16:
> I think that, within a running JVM, if setting the java.awt.headless property to true of anything that drew to the screen already did so does not work, so while testing, unless it’s the very first thing tested, that will fail to change state.

Which would mean tests become order dependent.

 > On the other hand, the point of the app is to run headless, so removing that does break the app in a subtle way (in that if there is other logic in jmrit.* that checks for that property before doing something that would crash if headless can’t prevent it).

But shouldn’t we be setting this property external to JMRI?  i.e. in the scripts that launch JmriHeadless?

pabender, 2019-06-16:
Oracle indicates that once the property has to be set before the toolkit is created to have an impact.  ( an older article, but see https://www.oracle.com/technetwork/articles/javase/headless-136834.html  ) so this is unlikely to be an issue unless, as @rhwood pointed out, the JmriFaceless app is the first test executed.

pabender, 2019-06-16:
There is at least one of the profiles (TMCC ) that fails in headless mode with a headless exception.

The grapevine profile may also fail, but I am running it once without the preceding TMCC test to see if that persists.

pabender, 2019-06-16:
The grapevine profile also fails in headless mode.  This means the only two profiles we are loading are the LocoNet and EasyDCC simulator profiles.

pabender, 2019-06-16:
@rhwood, the new JmriFaceless app appears to have found a real bug in the webserver preferences.

If we run the tests in headed mode, all of the cucumber apps appear to function correctly.

If we run the tests in headless mode, the web page for preferences gives us a 404 error:

Failed scenarios:
java/acceptancetest/features/web/WebServer.feature:18 # Basic Http requests
java/acceptancetest/features/web/WebServer.feature:33 # Basic Http requests
28 Scenarios (2 failed, 26 passed)
84 Steps (2 failed, 82 passed)
1m31.113s
org.junit.ComparisonFailure: Page Title expected:<[Directory: /prefs/ | My JMRI Railroa]d> but was:<[Error 404 Not Foun]d>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at jmri.web.WebServerAcceptanceSteps.lambda$new$4(WebServerAcceptanceSteps.java:53)
	at ✽.a page with title Directory: /prefs/ | My JMRI Railroad is returned(file:java/acceptancetest/features/web/WebServer.feature:7)
org.junit.ComparisonFailure: Page Title expected:<[Directory: /prefs/ | My JMRI Railroa]d> but was:<[Error 404 Not Foun]d>
	at org.junit.Assert.assertEquals(Assert.java:115)
	at jmri.web.WebServerAcceptanceSteps.lambda$new$4(WebServerAcceptanceSteps.java:53)
	at ✽.a page with title Directory: /prefs/ | My JMRI Railroad is returned(file:java/acceptancetest/features/web/WebServer.feature:7)

(This is for the first of the two scenarios that failed, the second is the same, but it refers to the chrome tests) 

This is being run after the JmriFaceless tests, so the environment variable is set when these tests start.

I haven't  investigated further at this point, but is it possible the preferences aren't loaded if the environment variable is set (they really should be)?

I think we have been getting a false negative on that particular test (i.e something we thought should work in a headless environment may not actually work in a headless environment).

My fear is that we have many other false negative results. and that much less of what we think should work in a headless environment really does.

rhwood, 2019-06-17:
In the web server, `/prefs/` is a directory listing of the current profile's directory (this name was used before I joined the project for this purpose and some WiThrottle clients reject the web server as not JMRI if that directory is not responded to, so its still available in that name).

If the JmriFaceless app is not loading a profile, then that result from the web server is completely  correct, since a profile's home directory cannot be found if there is no profile.

rhwood, 2019-06-17:
We've long known that the JMRI unit tests *are* order dependent and my attempts to find those dependencies by having tests run in a random order on Travis-CI were eventually undone, so making the JmriFaceless tests run first wouldn't be a change in that.

The other possibility is that you only run the cucumber tests for `JmriFaceless.main()` in the headless tests on Travis-CI and run some other non-headless tests on other platforms, knowing that only running tests in the correct environment is reliably possible.

pabender, 2019-06-17:
Sent from my iPhone

> On Jun 17, 2019, at 5:16 AM, Randall Wood <notifications@github.com> wrote:
> 
> In the web server, /prefs/ is a directory listing of the current profile's directory (this name was used before I joined the project for this purpose and some WiThrottle clients reject the web server as not JMRI if that directory is not responded to, so its still available in that name).
> 
> If the JmriFaceless app is not loading a profile, then that result from the web server is completely correct, since a profile's home directory cannot be found if there is no profile.
> 
That app certainly appears to be loading the profile just like all the rest of the apps.  One of the info lines produced by the application indicates the profile being loaded.

The fact is we get a failure if we run the JmriFaceless app in the headless tests.  We don’t get that failure running it with the headed tests.

The cucumber tests are running in the same order on both headed and headless.

I don’t think this is about JmriFaceless itself, but a difference between how it starts when the property is set before the application starts.

rhwood, 2019-06-17:
I can't recreate the failure your test is seeing with java.awt.faceless=true set by the main method or by the launcher when run outside a test except when a profile is not loaded (or loading correctly--and I don't have any of those), so until precise reproduction procedures are provided for reproducing the problem outside the tests, the problem should be considered to be purely in the tests themselves and not in the web server.

pabender, 2019-06-17:
@rhwood,  regardless of whether or not this is purely in the tests, this needs to be corrected.

The basic web tests reset the InstanceManager and the ProfileManager before each test runs.  That should mean we have no profile loaded any time this test runs, yet it works in the headed case and not the headless case.  By your statement above, I would expect this to always fail.

pabender, 2019-06-18:
reexamining that, jmri.util.JUnitUtil.resetProfileManager() installs a Null profile.  

This happens before every single run of the tests in web related cucumber tests.

So there IS a profile loaded when the web tests run.

rhwood, 2019-06-18:
Does it install a NullProfile with or without an existing directory or does it set the profile to null?

pabender, 2019-06-18:
> Does it install a NullProfile with or without an existing directory or does it set the profile to null?

It installs a NullProfile:
https://github.com/JMRI/JMRI/blob/bc8ed19d0599abff749a6ee3f32108d59dbd529a/java/test/jmri/util/JUnitUtil.java#L989-L999

https://github.com/JMRI/JMRI/blob/bc8ed19d0599abff749a6ee3f32108d59dbd529a/java/test/jmri/util/JUnitUtil.java#L1015-L1018

pabender, 2019-06-23:
Resolved via #7105

closed 2019-06-24 by pabender