JMRI/JMRI #7117: Request for Smart Coupler Decoder definition

bobjacobsen, 2019-06-20:
Jack from Litchfield Station would like to see the "Smart Decoupler Decoder" added. I've created this issue for him.

Jack1412LS, 2019-06-20:
Requesting the addition of the Smart Coupler Decoder to DecoderPro If required, samples are available from Litchfield Station to someone willing to add the Decoder to DecoderPro Subarashi Models Japan; Hiromi Kusumato; email: reply@crm.wix.com
[Scan1.PDF](https://github.com/JMRI/JMRI/files/3311428/Scan1.PDF)

dheap, 2019-06-20:
This decoder doesn't seem to support ManufacturerID (CV7) nor CV8, the NMRA-mandated CVs.

Can JMRI support a definition where these are not defined?

devel-bobm, 2019-06-20:
It might be helpful to have scans of any other pages from the Smart Coupler decoder documentation, especially if it shows any "tricks" to programming the decoder.  The wiring diagram might be useful, too.

In the absence of any programming "tricks", I see a number of challenges coming up.

- The PDF file provided by @Jack1412LS seems to imply that the smart coupler decoder is simply installed in parallel with the mobile decoder which runs the locomotive's motor, lights, sounds, etc.  If this is the case, and a separate decoder definition is provided for the Smart Coupler decoder, then it poses a big challenge for JMRI's "Roster": how to maintain consistency between the two Roster entries for the locomotive?  I infer from the PDF that the coupler decoder only implements CVs 1, 9, 17, 18, and 29.  If any of these are changed in one Roster entry, then the other Roster entry is no longer consistent.  This will lead to _bad_ user experiences!

- The Smart Coupler scanned document page implies that CV9 is not typically used for anything useful by the typical mobile decoder.  I don't think that is a true statement.  By my count, there are many JMRI decoder mobile definitions which define `<variable CV="9" ...>` either explicitly in the "top-level" decoder definition, or via an "include" of common parts defined in subdirectories of `xml/decoders`.  This includes decoders referencing content in the `digitrax`, `doehler_haass`, `esu` and `lenz` subdirectories.  These all appear to control "PWM period", "PWM frequency" or "motor control algorithm".  This is some **SIGNIFICANT** overlap on CV9, and, in my opinion, does _not_ equate to the statement from the PDF saying "Even if you change CV value of locomotive, this decoder will not be affected".

- Even if used with a mobile decoder which does not use CV9 for anything useful, what is the behavior of that mobile decoder when a programmer reads CV 9?  Does the combination of decoders give any sort of reliable information when reading CV9?  If not, what does that mean for the "user-experience" when using a JMRI "Roster" entry for the Smart Coupler?

This decoder design, and the small amount of information provided thus far, looks to be a BIG challenge to fit into the way JMRI "normally" does decoder programming!

KenC57, 2019-06-20:
One source of examples for decoders that don't have normal CV values would
be some of RR-CirKits products. They are very successful but use the CV for
its own needs. Granted these are stationary decoders but from JMRI
perspective, I'm not sure it is noticeable. So ask Dick about some of the
tricks he pulled. Granted without CV7 or CV8 it would never identify, but
other comments say that's a very minor issue with how to use it correctly.

devel-bobm, 2019-06-20:
@KenC57, The RR-CirKits .xml files define CV7 and 8.

This allows JMRI to identify them via service-mode programming (over LocoNet) when the device is configured for service-mode programming.  

And, similarly JMRI can identify the RR-CirKits device type when using ops-mode programming, with the correct device address.  (The RR-CirKits devices allow CV reads in ops-mode by providing a reply over LocoNet.)

So, I think that the RR-CirKits device example does not help much for the "Smart Coupler" case.

devel-bobm, 2019-06-20:
It would seem to be a much simpler problem if the "Smart Coupler" decoder had support for the "Decoder Lock" CVs (CV15, 16).  That, unfortunately, does not appear to be the case.

pabender, 2019-06-20:
> This decoder doesn't seem to support ManufacturerID (CV7) nor CV8, the NMRA-mandated CVs.
> 
> Can JMRI support a definition where these are not defined?

Why would we not be able to support a decoder without these CVs defined.

But to answer the question.... we support decoders that only support Registers.  I have an old Lenz decoder in an Arnold N-scale Alco S2 ( released around 1993! ) that I have programmed with JMRI.

Identification of decoders without these values is another story....

pabender, 2019-06-20:
> It might be helpful to have scans of any other pages from the Smart Coupler decoder documentation, especially if it shows any "tricks" to programming the decoder. The wiring diagram might be useful, too.
> 

The wiring diagram is on the website....

https://www.smart-coupler.com/decoder-product

From the smart coupler page ( https://www.smart-coupler.com/smartcoupler-english ) you are not required to use their decoder to open the coupler.  Perhaps they intend for the decoder to be installed in rolling stock.

KenC57, 2019-06-20:
Maybe check the Team Digital entries. I recall at least a couple of times
none of the normal uses of CV's was used (it didn't really apply). None of
these things had to deal with being in the same place as normal motor
decoders (locos).

dheap, 2019-06-21:
@pabender 
> @dheap Why would we not be able to support a decoder without these CVs defined.

Sorry, brain-fade. The NCE Light-It definitions I wrote don't implement CVs 7 nor 8!

But, since they don't support CV read-back (nor program track mode), identification is not possible.

closed 2021-04-14 by stale[bot]