JMRI/JMRI #7126: Create Test Release 4.17.1

bobjacobsen, 2019-06-23:
This will be the first test release of the 4.18 cycle. It will be built from HEAD of master.

bobjacobsen, 2019-06-24:
Currently, this is scheduled to build from HEAD of master (including all non-WIP passing-CI no-change-request PRs) sometime in the afternoon/evening of Tuesday June 25.   The release will be on the GitHub release page. The main JMRI web pages will be updated and the announcement email will be sent later, right after JMRI 4.16 is announced.

bobjacobsen, 2019-06-27:
The release-4.17.2 branch has been created. 

Maintainers, please set the 4.17.3 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://builds.jmri.org/jenkins/job/TestReleases/job/4.17.2/)

rhwood, 2019-06-27:
> The release-4.17.2 branch has been created.

Did you mean _4.17.1_ instead?

> Maintainers, please set the 4.17.3 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Did you mean _4.17.2_ instead?

bobjacobsen, 2019-06-27:
> Did you mean ...

Thanks for catching that!  The automation apparently doesn't handle the 4.15.9 -> 4.17.1 transition right. For clarity, I've edited the comment.

bobjacobsen, 2019-06-27:
[image]

bobjacobsen, 2019-06-27:
[image]

bobjacobsen, 2019-06-27:
Output arising during `ant alltest` on macOS (tracebacks truncated):

```
     [java] WARN  - Timeout on reply to message: 21 24 05 consecutive timeouts = 0 in z21.Z21XNetPacketizer [z21.Z21XNetPacketizer Transmit thread] jmri.jmrix.AbstractMRTrafficController.handleTimeout()
```

A SPROG dialog was displayed for a while right before the following:
```
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame85,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.Sprogv4UpdateFrame[frame87,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame85,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.Sprogv4UpdateFrame[frame87,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame85,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.Sprogv4UpdateFrame[frame87,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.SprogIIUpdateFrame[frame85,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
     [java] WARN  - notify: During dispatch to jmri.jmrix.sprog.update.Sprogv4UpdateFrame[frame87,0,23,240x96,invalid,hidden,layout=java.awt.BorderLayout,title=SPROGII/SPROG 3/Pi-SPROG Firmware Update,resizable,normal,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,0,22,240x74,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
     [java] Exception java.lang.NullPointerException [AWT-EventQueue-1] jmrix.sprog.update.SprogVersionQuery.notifyVersion()
```

```
     [java] Exception in thread "Default Speeds Dialog Close Thread" org.netbeans.jemmy.TimeoutExpiredException: Dialog with title "Defaults..." (WindowWaiter.WaitWindowTimeout)
     [java] 	at org.netbeans.jemmy.Waiter.waitAction(Waiter.java:210)
     [java] 	at org.netbeans.jemmy.WindowWaiter.waitWindow(WindowWaiter.java:264)
[... cut ...]
     [java] java.lang.AssertionError: Open Button not found
     [java] 	at org.junit.Assert.fail(Assert.java:88)
     [java] 	at org.junit.Assert.assertTrue(Assert.java:41)
     [java] 	at org.junit.Assert.assertNotNull(Assert.java:712)
     [java] 	at jmri.util.JUnitUtil.pressButton(JUnitUtil.java:1332)
     [java] 	at jmri.jmrit.catalog.PreviewDialogTest.testPreviewDialog(PreviewDialogTest.java:60)
     [java] 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
...
     [java] 	at apps.tests.AllTest.main(AllTest.java:39)

```

bobjacobsen, 2019-06-28:
JMRI 4.17.1 has been released. Files are available in the GitHub release section.

closed 2019-06-28 by bobjacobsen