JMRI/JMRI #7129: Trouble creating blocks (possible 4.15.9 issue)

bobjacobsen, 2019-06-24:
From a [JMRIusers post](https://groups.io/g/jmriusers/message/161040):

> Just downloaded JMRI 4.15.9 Tried to batch create a few blocks with blank name and get an error message cannot create block "" Check the name is not in use. If I create singly with no name, there is no problem.
> OS Win 10

Later in the same thread:

> Creating sensors works just fine. I can create a batch with no name.

It's not clear if this is a long-standing issue or new regression.

bobjacobsen, 2019-06-24:
Tried to recreate without success:  I opened the Block Table, Clicked the Add… button, selected “Automatically generate System Name” and “Add a sequential range”, selected 5 for number of items, and clicked Create.  That seemed to work.

Asked on JMRIusers for more info, see thread linked above.

icklesteve, 2021-03-12:
Unable to reproduce, no further info. on JMRIUsers thread, closing issue pending further info.

closed 2021-03-12 by icklesteve