JMRI/JMRI #7164: Error messages when there are two LocoNet connections.

dsand47, 2019-07-02:
When there are two LocoNet connections, opening any table results in two error messages.
```
2019-07-02 13:36:41,604 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2019-07-02 13:36:47,853 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.TurnoutTableAction:All share the identity "Bypass Bushby Bit"; saving table state will not be reliable. [Listed Table Generation]
2019-07-02 13:36:47,855 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.TurnoutTableAction:All share the identity "Send ON and OFF"; saving table state will not be reliable. [Listed Table Generation]
2019-07-02 13:36:48,506 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
```
The errors occur regardless of the turnout table content.

To create the errors, create a profile with two LocoNet connections, save, restart and select any Tools >> Tables entry.

The errors occur with both 4.15.9 and 4.17.1.  They occur using LocoNet simulation and LocoBuffer-USB connections.

Pugwash1, 2019-07-03:
I added these properties recently, so did a little digging and found that the code I used a template has the same issue (openlcb). I dug further and I think that it  gets called twice on the same connection and I don't know how to fix it, and if that thoughts wrong I still don't know what to do. Help.
The good news is that you can still create edit save & open  and it appears to work.
 Its unlikely that I will be near a dev machine again before tomorrow morning.(Toronto time)

KenC57, 2020-10-26:
The problem looks to be with the column name when in the 'All' display. It iterates the system specific columns for each connection. You end up with as many of the 'Bypass Bushby Bit' and 'Send ON and OFF' as you have connections using those items. But it may be when building the table for 'All' that the column name needs to be modified to include the connection prefix or name. With the right logic behind the table it might figure out how to have one column and know which connection to pick from the system name but I'm not sure on that. One of the experts of the table would have to speak up on this.

devel-bobm, 2020-11-05:
Same basic problem happens when a configuration profile containing two LCC (CAN bus simulator) connections is used - open the "Turnouts" table and get the following console messages:
```
2020-11-05 14:17:50,897 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.TurnoutTableAction:All share the identity "Authoritative"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,897 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.TurnoutTableAction:All share the identity "Always listen"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,943 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:All share the identity "Authoritative"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,943 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:All share the identity "Always listen"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,959 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:LCC share the identity "Authoritative"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,959 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:LCC share the identity "Always listen"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,975 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:CAN share the identity "Authoritative"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,975 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:CAN share the identity "Always listen"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,990 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:Internal share the identity "Authoritative"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:50,990 swing.JmriJTablePersistenceManager    ERROR - Columns in table jmri.jmrit.beantable.SensorTableAction:Internal share the identity "Always listen"; saving table state will not be reliable. [Listed Table Generation]
2020-11-05 14:17:51,224 beantable.ListedTableFrame            ERROR - Error when adding jmri.jmrit.beantable.LightTableTabAction to display [Listed Table Generation]
java.lang.ArrayIndexOutOfBoundsException: 4 >= 4
	at java.util.Vector.elementAt(Unknown Source)
	at javax.swing.table.DefaultTableColumnModel.getColumn(Unknown Source)
	at jmri.jmrit.beantable.BeanTableDataModel.setColumnToHoldButton(BeanTableDataModel.java:493)
	at jmri.jmrit.beantable.BeanTableDataModel.configDeleteColumn(BeanTableDataModel.java:473)
	at jmri.jmrit.beantable.BeanTableDataModel.configureTable(BeanTableDataModel.java:450)
	at jmri.jmrit.beantable.AbstractTableTabAction$TabbedTableItem.createDataModel(AbstractTableTabAction.java:201)
	at jmri.jmrit.beantable.AbstractTableTabAction$TabbedTableItem.<init>(AbstractTableTabAction.java:181)
	at jmri.jmrit.beantable.AbstractTableTabAction.createModel(AbstractTableTabAction.java:45)
	at jmri.jmrit.beantable.AbstractTableTabAction.getPanel(AbstractTableTabAction.java:80)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.addPanelModel(ListedTableFrame.java:418)
	at jmri.jmrit.beantable.ListedTableFrame$TabbedTableItem.<init>(ListedTableFrame.java:364)
	at jmri.jmrit.beantable.ListedTableFrame.initComponents(ListedTableFrame.java:112)
	at jmri.jmrit.beantable.ListedTableAction$1.run(ListedTableAction.java:68)
	at java.lang.Thread.run(Unknown Source)

```

devel-bobm, 2020-11-05:
The problem seems to relate to the names used for "system-specific" properties of beans.  I believe that the error message would occur for those tables where there are two or more connections which have the same system-specific property names.  If you had a connection type X connection and a connection type Y connection, and each connection type defined a system-specific property called "Z" for a given bean type, then you would get this error-message type.

The current `jmri.swing.JmriJTablePersistenceManager` coding seems to support a single connection having a bean type "T" with system-specific property by the name of "W", but has problems when there are two or more connections which provide system-specific property name "W" for bean type "T".

KenC57, 2020-11-05:
I just checked with a dual NCE system. It doesn't have system specific turnout elements. No errors in any of the tab displays: All, NCE, NCE2, Internal. Turning on or off any of the option displays made no errors since there were no duplicated column names.

The key seems to be the column names.

devel-bobm, 2020-11-05:
In the "problematic" scenario, the difficulty appears to come from the "All" tab.  That tab shows two copies of the system-specific property names as column headers when two connections are available and have system-specific properties using the same name.  If you select the connection-specific tabs, only one copy shows up (in each tab).

The console log message only occurs at table "creation" time, when the first table is opened.  Apparantly all beantables are created when one is opened, and any table which has the problematic situation will generate its errors when that first table is opened.

Perhaps the check in `jmri.swing.JmriJTablePersistenceManager" needs to be something other than a simple string comparison of the property name(s), and instead something more like a comparison of the object class and its property name?

KenC57, 2020-11-05:
With the current data model, is it likely to be able to 'collapse' the multiple column names in the 'All' tab? That would seem the right solution as only one of the column sets would be correct for any one systemname. Currently it lets you change values in any of those columns. If you change the wrong column, the wrong value results when you save the table. Some of the combinations seem to get the fields confused.

devel-bobm, 2020-11-06:
"Collapsing" the columns with identical names would only be appropriate if all connections use the same "system-specific property name" string for the same purpose.

The idea of collapsing the duplicated columns to a single column would fall apart if one active connection uses the system-specific property name "X" for checkbox information while another active connection used system-specific property name "X" to hold, for example, a combo-box selection.  So far as I know, the JTableDataModel does not allow row-specific data types, so this collapsing would fail in such a case.

icklesteve, 2021-02-04:
Issue also with CbusReporterManager columns

PR #9385 (2021-02-07): Fix for Duplicate Columns for Custom Bean Properties when multiple connections of same type
In AbstractProxyManager.java getKnownBeanProperties() prevents duplicate NamedBeanPropertyDescriptors when there are multiple system connections of the same type..

Adds equals() and hashcode() to NamedBeanPropertyDescriptor.java based on Identifier and default value class.

Fixes #7164 

Improved generic in AbstractTableTabAction.java
closed 2021-02-07 by danielb987