JMRI/JMRI #7178: Add button for "Select Different Profile?" to "Error initializing" dialog

mstevetodd, 2019-07-08:
When an error occurs opening JMRI, a dialog is shown to allow the user to Quit, Continue or Edit connections.
It would seem helpful to me if there were an additional option "Select Different Profile?" which took the user to the "Set Active Profile" dialog, so they could choose another profile to use.

mstevetodd, 2019-07-08:
[image]

rhwood, 2019-07-08:
We had that in 4.14, but I see it's not in 4.16. You might want to figure out why it was removed between those versions.

danielb987, 2019-07-08:
This dialog is from #5771, #5855 and #6340.

There has been a long discussion about this dialog so I will leave it as it is. But if you want to change it, feel free to do so.

mstevetodd, 2019-07-08:
Thanks for the links, @danielb987 . I've no interest in reopening that discussion, so I'l close this issue.

closed 2019-07-08 by mstevetodd