JMRI/JMRI #7179: 4.16 Throttle behavior with direct change is different from 4.14

KenC57, 2019-07-08:
From a user...
Since the loading of release 4.16 using the jmri throttle if a loco is traverling at any speed you press the forward or reverse button it has no effect. If the loco is stopped then the forward / reverse button is pressed then the speed increased it operates as normal. This is only happening since release 4.16, I have reloaded 4.14 and it works perfectly again, revert to 4.16 the problem reappears.
You used to be able to press forward or reverse whilst moving and the loco would stop then travel in the other direction.

This fault is tied up with my last report re- headcode orientation switching.

Regards
Michael 
=========================================================
JMRI Version: 4.16+R6f9aced   
JMRI configuration file name: null   (from org.jmri.apps.Apps.configFilename system property)
JMRI main window name: JMRI System console   
JMRI Application: PanelPro   
Connection 0: MERG connected via CAN via MERG CAN-RS or CAN-USB on COM12 Disabled false   
Connection 1: DCC4PC connected via DCC4PC on COM10 Disabled true   
Available Communication Ports:
 Found 1 serial ports
 Port: COM12 - in use by: JMRI app   
Active profile: My JMRI Railroad

rhwood, 2019-07-08:
Please include a link to original messages, so we can get in touch with the original reporter, or encourage the original reporter to submit issues directly to us.

KenC57, 2019-07-08:
This was via the Upload. User is ardmay@btinternet.com
-ken c

bobjacobsen, 2019-07-08:
From mailing list:

https://sourceforge.net/p/jmri/mailman/message/36711698/

https://sourceforge.net/p/jmri/mailman/message/36711238/

pabender, 2019-07-08:
I haven’t looked into the full context of this, but it is possible this is System specific behavior and not the general throttle behavior.

In particular,@icklesteve has been doing quite a bit of work on the MERG CBUS connections, and it could be related to some of those changes.

pabender, 2019-07-08:
PR #7181 adds a test that verify the JMRI throttle UI sends the direction changes to the throttle itself when the speed is not zero (there are already checks to verify the direction changes were sent when stopped).

I believe this confirms the issue is system specific and not a regression in the general throttle behavior.

icklesteve, 2019-07-08:
Thanks for tagging me in,
Can reproduce with CBUS Simulator Throttle, will look into this.

icklesteve, 2019-07-08:
Hi @KenC57 unfortunately I'm not subscribed to the reports list.

If you could send a message on my behalf advising that the issue has been located, a fix submitted, and unit test submitted to prevent this happening again?

Many thanks,
Steve.

KenC57, 2019-07-09:
Will do. And did.
-ken c

icklesteve, 2019-07-10:
Thanks @KenC57 , much appreciated.

After I've got 3rd party confirmation that this is fixed, I'll open a PR on the JMRI website to update release notes advising of issue.
I've also made MERG forum aware.

Quick question, how do I enable "shunter slider throttle option in a JMRI throttle" ?
I've heard this mentioned before, however I can't find how to enable?

Thanks,
Steve

KenC57, 2019-07-11:
Steve,

I recall it as a right click in the speed portion of the JMRI throttle. Same
if changing to speed step mode.
-ken c

icklesteve, 2019-07-12:
Many thanks, somehow had never spotted the right click properties menu on the slider before !!

I've had confirmation from a MERG member that direction changes are now being sent to layout as expected in head following #7182 being merged,

Steve.

KenC57, 2019-08-15:
Sounds like Steve had confirmation that #7182 fixed the issue so this can be closed.

closed 2019-08-15 by KenC57