JMRI/JMRI #7183: ESU Misc Decoder-Interior Lighting problem

JimLittle47, 2019-07-08:
When the "Advanced" or "Analog Controls" panes are written to the decoder, it changes the value of CV 29 to a "1". Previous value was 34 (Long address)  This happened numerous times. Using JMRI 4.16 / NCE Power Pro.       Jim Little, MMR

dheap, 2019-07-08:
I can't reproduce this problem.

Do you "Read Full Sheet" on the CVs pane and check CV 29 contents in JMRI before making any changes? there "Advanced" pane doesn't even reference CV29.

dheap, 2019-07-08:
@JimLittle47 Note that any attempt to write directly to CV285 on a Power Pro will corrupt CV29.

dheap, 2019-07-08:
@JimLittle47 Suggest you also set up "Open NCE Monitor" as a Startup Action in JMRI. The first time it opens, select "Show Raw Data" and "Show timestamps" and close the window (to save the settings).

Then when something goes wrong you can Select All and Copy the contents into a text file to upload here. Also check JMRI System Console).

JimLittle47, 2019-07-08:
Dave;

I have 3 of the decoders.  Each time I programmed them, the basic pane went
OK doing a write full sheet.  Couldn't figure out why after programming a
long address (#1000) they wouldn't respond on the main track (Programming
was done on the Programming track).  When I did a "Write full sheet" on
either "Advanced" or "Analog" panes, then went to CV pane and did a Compare
the pane goes red and shows the decoder with a value of 1 making the short
address active.  I saw this happen several times until I realized what was
happening.
Re: Your comment about CV 285 - I take it you mean the indexed (0.255.285)
in the decoder?
Will make those changes in my JMRI for having a monitor.

Thanks
Jim Little

dheap, 2019-07-08:
About CV285: attempt to directly write any CV>256 (indexed or not) in program track mode on a Power Pro will corrupt the decoder.

dheap, 2019-07-08:
Once set up with NCE Monitor:

- Clear NCE Monitor.
- Read CV29 on CVs pane.
- Perform operation you suspect corrupts.
- Read CV29 on CVs pane.
- Select All (Ctrl-A) on log and copy (Ctrl-C) and paste into text document.
- Upload text document.

dheap, 2019-07-08:
Alternatively, you can use the "Choose Log File...", "Start Logging" and "Stop Logging" buttons in the NCE Monitor pane.

JimLittle47, 2019-07-08:
Dave;
As you requested, I ran a NCE monitor and attached is the log.
What I did was:
      -read CV29
      -for Advanced pane - "Write Full Sheet"

dheap, 2019-07-09:
@JimLittle47 You can't attach documents via email. You need to go to https://github.com/JMRI/JMRI/issues/7183 and attach it there.

JimLittle47, 2019-07-09:
[7-8-19 ESU Lighting Decoder.txt](https://github.com/JMRI/JMRI/files/3370637/7-8-19.ESU.Lighting.Decoder.txt)

JimLittle47, 2019-07-09:
Sorry Dave - It's my first time using this system.

dheap, 2019-07-09:
@JimLittle47 JMRI is sending all the correct commands, according to the command monitor, but the decoder contents seem to change between reads. Seems like a decoder problem?

```
16:31:21.203:  Read CV 29 in direct mode
16:31:22.000:  Reply: 22 21
(reply from decoder value=0x22=34 21=! signifying read OK.
16:31:22.203:  Exit Programming track mode
16:31:22.500:  Reply: 21
(OK, done)
16:31:24.078:  Enter Programming track mode
16:31:24.093:  Reply: 21
(OK, done)
16:31:25.093:  Read CV 29 in direct mode
16:31:25.921:  Reply: 01 21
(reply from decoder value=0x1=1 21=! signifying read OK) - that's unexpected.
16:31:26.140:  Exit Programming track mode
16:31:26.437:  Reply: 21
(OK, done)
16:31:30.359:  Enter Programming track mode
16:31:30.375:  Reply: 21
(OK, done)
16:31:31.375:  Read CV 29 in direct mode
16:31:32.203:  Reply: 01 21
(reply from decoder value=0x1=1 21=! signifying read OK) - that's unexpected.
16:31:32.406:  Exit Programming track mode
16:31:32.718:  Reply: 21
(OK, done)

```

Seems the decoder corrupts CV29 after multiple reads and entry/exit from program track mode. Are you using a Program Track booster?

dheap, 2019-07-09:
`reply from decoder` is actually 'reply from Power Pro' but your other evidence indicates the decoder has the wrong value in CV29 ands not just a misread.

JimLittle47, 2019-07-09:
OK maybe I got a bad batch of decoders.  As I had the same problem with all
3 of them, I figured it was the  software somehow causing CV 29 to change
back to a value of 1 Clearly if you don't get the same results I do, maybe
I did get a bad batch of decoders.  It happened numerous times and  took me
awhile to figure out what was happening. I even took it over to a friend
who has a Lokprogrammer and he pointed out CV 29 had a value of 1 instead
of 34.
There is a Soundtraxx booster between the programming track and the Power
Pro.
.
Thanks for your patience and assistance.

Jim Little

dheap, 2019-07-09:
Odd. I don't have an ESU Misc Decoder-Interior Lighting to test with. I have lots of other ESU test decoders.

Have you tried using the Pro Cab in single-CV programming track mode.

- Go into Program Track Mode, write 34 to CV 29 and exit Program Track Mode.
- Go into Program Track Mode, read CV 29 and exit Program Track Mode.
- Go into Program Track Mode, read CV 29 and exit Program Track Mode.
- Go into Program Track Mode, read CV 29 and exit Program Track Mode.
- ...

I use a Soundtraxx PTB-100 with my Power Pro.

dheap, 2019-07-09:
@JimLittle47 Could you please read me the firmware version from the Information Pane?

JimLittle47, 2019-07-09:
Dave;

I only have the Power Pro and a PTB-100.   I 'upgraded from a System One
about 3 years ago.  Got tired of not being able to read in Direct Mode!
 Reading the firmware version from the information pane (all 3 decoders are
the same) comes back with a 0 in all 3 boxes.

Jim

dheap, 2019-07-09:
Looks like the ESU digital interior light doesn't support reading the RailCom page CVs.

But since you are only trying to read CV285, that shouldn't corrupt CV29. Apart from that, I didn't see any sign of you reading/writing CV285 anywhere.

I'll update the definition later.

dheap, 2019-07-09:
Your friend with a Lokprogrammer could read the firmware version and update the firmware. Latest is 1.0.1461.

dheap, 2019-07-09:
Did you try the [Read using your NCE throttle](https://github.com/JMRI/JMRI/issues/7183#issuecomment-509451256) procedure? I'd very much like to see the results of that.

JimLittle47, 2019-07-09:
I have used my throttle before ( I've had DCC for about 25 years now ) I
got into using JMRI.  I started with a Ramtraxx unit before the System One.
I'll try programming it with my throttle tomorrow and get back to you.

JimLittle47, 2019-07-10:
Dave;
I was able to program the light decoder today with my Procab.  I have 150
ohm resistors connected to the AUX1 in order to read CV's as per the
instruction sheet..
Let me know if you would like the instruction sheet for the decoder. I have
it as a pdf.

Jim Little

dheap, 2019-07-10:
Jim,

> On 10 Jul 2019, at 1:22 PM, JimLittle47 <notifications@github.com> wrote:
> 
> I was able to program the light decoder today with my Procab.

Good. Did you try multiple reads of CV29 with the Pro Cab to see if the value changed?

> I have 150
> ohm resistors connected to the AUX1 in order to read CV's as per the
> instruction sheet..

Hopefully you had them when using JMRI.

> Let me know if you would like the instruction sheet for the decoder. I have
> it as a pdf.

I've downloaded it, thanks. I'm trying to get hold of an Interior-Lighting decoder to see if I can reproduce your CV29 changing with repeated reads issue.

Dave in Australia

bobjacobsen, 2019-07-10:
Sorry to come to this late.

Some decoders will reset CV29 to 1 when they think that the short address (CV1) has been rewritten.  There used to be (not sure if there still is) NMRA dicta that decoders must do that so that an address-only programmer (i.e. can only change CV1) would set the decoder into short-address mode.

Is it possible that something in the programming sequence is kicking that off?

JimLittle47, 2019-07-10:
You might be on to something Bob.  It happened to all 3 of the decoders I
bought and not just once. The problem is repeatable.
It's way above my pay grade though.  I was a programmer back in the early
1970's using IBM assembler, PL1 and Cobol!

Jim Little

dheap, 2019-07-10:
@bobjacobsen I had memories of an ancient NMRA directive about programming order of the address CVs, but setting CV29 to 1 (reverse direction, 14 speed step, short address) seemed a bizarre value. 

Furthermore I couldn't see any signs of trying to program any of the other address CVs (1,17,18), only multiple writes and reads to CV29. The odd thing was that the writes to CV29 initially succeeded, but then seemed to revert after a period of time/number of reads. See the log file posted earlier.

JimLittle47, 2019-08-19:
Dave;

I received another ESU light strip over the weekend.  When I wrote to it
(JMRI  V4.16 / NCE Power Pro / Win XP) the basic page initially didn't
write the long address using Write Full Sheet. When I did Write Changes on
sheet, it took.  After each write, I kept checking CV 29. Later when I was
changing the Function Output pane, CV 29 was reset to 1.  This happened a
couple of times, so after I was finished setting it up the way I wanted, I
checked that CV 29 was 34.
I do think there's something in the program causing CV 29 to revert to a
value of '1'.

Jim Little

Jim

dheap, 2019-08-19:
@JimLittle47 I'm starting to suspect an ESU firmware bug. I've tried to get an ESU lighting strip for testing, but with no success as yet.

The [NCE command log](https://github.com/JMRI/JMRI/issues/7183#issuecomment-509440983) you provided shows nothing amiss. A JMRI error of this sort would manifest elsewhere. The only way to progress this further is for me to get hold of a test decoder and use my DCC Packet Analyser to observe what is actually happening on the programming track.

I've just discovered an issue with one of my ESU V4.0 test decoders which intermittently returns incorrect read values for CVs 66 & 95 (readonly in the V4.0) to  both my NCE system and my LokProgrammer! So i'm not ruling out ESU firmware issues either.

JimLittle47, 2019-08-19:
Thanks for the info Dave.
Hopefully you can get hold of one soon.  I'll be interested to see what you
find out about the units.

Jim little

dheap, 2019-08-20:
@JimLittle47 I've looked again at the [NCE Monitor](https://github.com/JMRI/JMRI/files/3370637/7-8-19.ESU.Lighting.Decoder.txt) log you posted.

To the best of our knowledge, the NCE Monitor records all the instructions we send to the NCE interface and records replies to these instructions from the NCE interface.

From that log there is absolutely no evidence that JMRI has attempted to change CV 29 values between successive reads, but the value has changed. The evidence suggests the ESU decoder has reverted the value after a period of time or number of reads.

@bobjacobsen [suggested](https://github.com/JMRI/JMRI/issues/7183#issuecomment-510145564) that some decoders may reset CV 29 to 1 after a write to CV 1, but I can't see anything in the log about CV 1 being written. Was that the entire log for the programming session with that decoder?

Another useful thing to provide may be a matching session.log file for the programming session with that decoder. (Help->Locations->Open Log Files Location). We could see if there were any errors logged.

JimLittle47, 2019-08-20:
I'll check my computer I use for JMRI regarding the log.   I've always
leave CV 1 at the value of 3  when I use the long address.  I was thinking
the same thing about what Bob Jacobsen said and CV 29 reverting to a value
of 1.
I'll get back to you on the file logs.

Jim

JimLittle47, 2019-08-22:
Dave;

Here's my log file.  I deleted all the messages prior to when I was loading
the decoders ( 2 TCS and the ESU lighting) from previous setups.  You're
probably right about a 'bug' in the decoder.  The only warnings and error
messages were when the NCE command station hung and I restarted it.
Should I contact Matt Herman or would you like to?

Sorry for the delay in getting this to you.  I'm recovering from back
surgery 3 weeks ago.

Jim Little

JimLittle47, 2019-08-22:
Sorry Dave;

Here's the file.

Jim

dheap, 2019-08-23:
> @JimLittle47 Should I contact Matt Herman or would you like to?

I'll contact him again. I did email him some time ago when we first encountered this problem, but haven't heard back. He's always very busy.

dheap, 2019-08-23:
> @JimLittle47 Here's the file.

Flies attached to an email reply are stripped. You'll have to upload it to a comment on the web page: https://github.com/JMRI/JMRI/issues/7183

JimLittle47, 2019-08-23:
[8-22-19 messages shortnd.log](https://github.com/JMRI/JMRI/files/3532529/8-22-19.messages.shortnd.log)

Dave;
Here's my log.  I stripped everything off prior to loading the latest ESU light strip as it was quite long and went back to 2015.
JIm

dheap, 2019-08-23:
> @JimLittle47 I stripped everything off prior to loading the latest ESU light strip as it was quite long and went back to 2015.

That's why session.log is usually better. Only shows the last jmri session, so less confusing.

dheap, 2019-08-23:
@JimLittle47 Apart from a failed session on August 18 where you apparently had a lockup of the Power Pro, nothing of concern in the log file you posted.

closed 2021-04-14 by stale[bot]