JMRI/JMRI #7184: java.lang.NullPointerException at sun.awt.windows.WWindowPeer.getActiveWindowHandles

pabender, 2019-07-08:
This issue is really more of an FYI than anything else.

We frequeltly see error messages in the Appveyor build of the form:

java.lang.NullPointerException at sun.awt.windows.WWindowPeer.getActiveWindowHandles(WWindowPeer.java:463)

Apparently this is a known issue in the version of the JDK 8 installed on Appveyor ( 1.8 update 162 ) see:

https://bugs.openjdk.java.net/browse/JDK-8198257

The resolution is to move to JDK 1.8 update 172 or later, but that isn’t currently an option.

bobjacobsen, 2019-07-08:
> The resolution is to move to JDK 1.8 update 172 or later, but that isn’t currently an option.

Would this be a good time to switch to a 1.9 Java for test running on AppVeyor?  We see more and more Windows users using Java 1.9 or later.

pabender, 2019-07-08:
> The resolution is to move to JDK 1.8 update 172 or later, but that isn’t currently an option.
> Would this be a good time to switch to a 1.9 Java for test running on AppVeyor? We see more and more Windows users using Java 1.9 or later.

I actually tried that, but there are other failures:
https://ci.appveyor.com/project/pabender/jmri-vis06/builds/25830605

Paul

rhwood, 2019-07-08:
Has anyone filed a bug with AppVeyor to get them to update their version of Java 1.8?

AFAIK Windows is the only environment we are really hanging onto Java 8 for, since newer versions of Java do not work on Windows XP.

Randall Wood

pabender, 2019-07-08:
@rhwood not yet.

The error I am getting on appveyor now may be an incompatibility with a library used for some embedded JavaScript.  Still investigating....

rhwood, 2019-07-08:
Not a warning about Nashorn being deprecated?

pabender, 2019-07-08:
Sent from my iPad

> On Jul 8, 2019, at 7:03 PM, Randall Wood <notifications@github.com> wrote:
> 
> Not a warning about Nashorn being deprecated?
> 

No, it just fails to start the interpreter.

pabender, 2019-07-08:
This is all Appveyor says about the JavaScript interpreter:

[ERROR] Failed to execute goal org.apache.maven.plugins:maven-antrun-plugin:1.8:run (test) on project jmri: An Ant BuildException has occured: The following error occurred while executing this line:
1865[ERROR] C:\projects\jmri-vis06\build.xml:393: The following error occurred while executing this line:
1866[ERROR] C:\projects\jmri-vis06\build.xml:311: Unable to create javax script engine for javascript
1867[ERROR] around Ant part ...<ant antfile="C:\projects\jmri-vis06/build.xml">... @ 4:51 in C:\projects\jmri-vis06\target\antrun\build-main.xml

rhwood, 2019-07-11:
In Java 9+, the Nashorn Javascript engine was moved into its own module. Perhaps that module is not on the ant command line. If we can't get that module on the ant command line, we would need to either accept the logging we currently use Javascript to silence in ant, or find an alternate mechanism to silence that (I do see some suggestions that due to differences between Java 8 and 9, in 8 you use Javascript and in 9 you use jshell).

pabender, 2019-07-11:
I did a test run where I just flat disabled that bit of JavaScript.

When I did that, there was a failure later on because of a replace we have ant doing ( it is removing Javadoc from generated sources ).

Disabling the replace, I did get a full build and test run, BUT, the output still contains the NPE I was trying to eliminate.... so, at this point, I am going to abandon the Java 9 path as a means to eliminate that from the logs.

rhwood, 2019-07-11:
Have you tried Java 11? Java 9 is no longer supported anywhere, while Java 8,11, and 12 are the releases with current support.

pabender, 2019-07-11:
@rhwood I have a java 11 build running now on appveyor, but I'm still seeing the NPEs.  I also am seeing out of memory exceptions. (and  I still have to have both the javascript and the replace operation disabled....)

rhwood, 2019-07-11:
Then I don't know. You may want to engage AppVeyor about this, or see if removing the part of the appveyor.yml where we attempt to set a display size makes any difference.

pabender, 2019-07-12:
I'm actually beginning to think the solution I found above is a red hearing, and the issue described may not have the same cause....

rhwood, 2019-07-12:
@pabender Do you think https://github.com/appveyor/ci/blob/master/scripts/set-screenresolution.ps1 might provide a clue?

pabender, 2019-09-21:
I found an old thread on a forum ( https://coderanch.com/t/538905/java/Closing-open-windows-opening ) that suggests using java.awt.Window.getWindows() can be have different behaviors on different JVM implementations.  In particular, the thread suggests that the returned value includes references to windows that are about to be garbage collected and are no longer in a state where you can execute dispose on them.

We use that method when closing windows:
https://github.com/JMRI/JMRI/blob/156c190607af1e49ea094d71745bb81113edfdc8/java/test/jmri/util/JUnitUtil.java#L1171

The thread suggests that calling the garbage collector before the getWindows() method will eliminate the disposed windows, making the call much less prone to errors.

I hate to put a direct call in to the garbage collector, especially for something that happens frequently during testing, but I am experimenting with that on one of my branches now.  If that behaves, I will create a PR.

bobjacobsen, 2019-11-18:
#7638 updates AppVeyor configuration to use a more recent JDK. We'll see if that fixes this.

pabender, 2019-12-07:
I recently received a new laptop with windows 10 installed.  Before changing the OS to Linux I decided to check and see if I could replicate this issue locally.  Running ‘mvn clean install’ does actually replicate the problem.  

This is with the latest JDK available for Windows.

I’m going to spend some time debugging before I wipe the machine.  Hopefully I can resolve this.

closed 2021-04-14 by stale[bot]