JMRI/JMRI #7185: Dispatcher AutoEngineer safe sections cannot be shorter than train?

mstevetodd, 2019-07-08:
Just got my layout back in running shape and now experimenting with the new "safe sections" logic, so I may just have something set up wrong.
But it appears that the logic to extend the allocation is not occurring until the previous block becomes unallocated. 
It's noticeable if the next block to be allocated has "work", such as changing turnouts, that occurs on allocation. If the train is shorter than the safe section, it works, but causes the train to stop until the work is complete. If the train is longer than the safe section, the allocation does not occur, so the work is never done, and the train stops.
Is this a problem to be addressed in the code, or is there a method for making this work?

Pugwash1, 2019-07-09:
Ah right. It's not by design. Its the way the we use releasing sections to trigger rescans of the  allocation request table, I have somewhere the code to move that process onto a dedicated thread which works way better than listener based triggers. It also removes a skid load of sync issues. I will dig that out so you can try it.

mstevetodd, 2019-07-28:
@pugwash1
Any update on this?

Pugwash1, 2019-07-28:
I have been looking at it, and tried a couple of things. Running the auto allocate on a timer is a big job, so put that off till the winter (october). Im trying a much simplier approach, triggering the autoallocate scan on a block going active, currently it only scans on one of a number of specific blocks going inactive. Should have something next week if get a rain or windless day.

On July 28, 2019 3:58:09 PM EDT, mstevetodd <notifications@github.com> wrote:
>@pugwash1
>Any update on this?

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

mstevetodd, 2019-08-06:
@Pugwash1 Quick test on my layout looks good, no sign of the earlier stopping behavior, so I merged your PR. Thanks!

mstevetodd, 2019-08-07:
@Pugwash1 Did some longer tests tonight:
Definite improvement, both looking forward and when waiting for another train to clear.
However, there are still some points where the train allocates much later than it "should", resulting in train slowing or stopping.
So hopefully you haven't abandoned your "separate thread for allocation" plan?

Pugwash1, 2019-08-07:
@mstevetodd 
Will be working on it come the fall.

mstevetodd, 2019-11-25:
@Pugwash1 Any progress on this?

Pugwash1, 2019-11-30:
@mstevetodd 
My work so far has been to show that the concurrency issues are way worse than I thought, the mAllocatedSections (in ActiveTrain) and allocatedSections (in DispatcherFrame) are particularlly irksome with items being deleted while being updated, and sorted while being deleted. When I move those off into single threaded classes (using try locks which makes debugging easy), I still get deadly embraces, and corruption as references to entries are being held in all sorts of odd places. (you must have noticed these messages "scanAllocationRequestList - maybe the allocationrequest was removed due to a terminating train??")
Most of it works currently due to the slowness of the process and sheer dumb luck, it was designed for an operator manually allocateing and releaseing each section, and added to without building a robust infrastructure.
Please feel free to have a go. The first changes will appear to work, then test with multiple trains with multiple actions per section, stopping by speed profile,  and contention for the same sections, thats when the wheels come off the cart.

mstevetodd, 2021-03-30:
My layout is not currently in run condition, so I can't test the current state of this issue. I'm going to close it and create new or reopen if future testing shows an issue.

closed 2021-03-30 by mstevetodd