JMRI/JMRI #7191: Cannot maintain route, get new one with IR prepended

mstevetodd, 2019-07-11:
Saw this with my existing panel, but here are steps to duplicate:

Up to date with Master. New empty configuration with only Digitrax Simulator defined.
Tools, Tables, Routes, Add....
Press Create. System name "IR:AUTO:0001" is created.
Press Cancel to close the Add/Edit Route window.
Press Edit in table for that new entry.
Press Update in Edit Route window.
A new route is added to the table as "IRIR:AUTO:0001".
Repeat, results in "IRIRIR:AUTO:0001", etc.

Same problem if I set the username, add elements to the route, etc.

mstevetodd, 2019-07-11:
I verified that this problem does not exist in 4.16.

rhwood, 2019-07-11:
In master, we have removed all normalization of system names. It’s clear that this table relied on normalization to work. If you can create a branch with a test that catches this, I can that add to that branch a fix if you don’t figure it out before then.

mstevetodd, 2019-07-12:
I will investigate changing/adding tests this afternoon, but I've found a recent addition, that if removed, makes the problem go away. 
https://github.com/JMRI/JMRI/commit/7b45537aa68cf4b67953f182762c6d1a01343120, specifically the addition of line 985 to RouteTableAction.java. If I comment out that line, I no longer get the incorrect route added.
@rhwood please advise your thoughts.

rhwood, 2019-07-12:
Manager.makeSystemName() and its implementations should not be blindly adding the system name prefix, so the real bug is in some makeSystemName() implementation. We need to find that and fix it instead of leaving users without the ability to rely on that method to pretend the system connection details to a system name.

mstevetodd, 2019-07-12:
Ok, see #7193

mstevetodd, 2019-07-14:
Closed by #7193.