JMRI/JMRI #7194: Problem with ver 4.16 and Locobuffer USB

frankfezzie, 2019-07-13:


frankfezzie, 2019-07-14:
On July 10, 2019 Egbert Broerse indicated in the JMRI User Group that he knew what the problem was and that a fix would be in the next release of JMRI.

silverailscolo, 2019-07-14:
Fix is in PR #7187. Looking at the code history the LocoBuffer USB has had this issue with a fresh config for years.

rhwood, 2019-07-15:
@silverailscolo Except that, as reported in this issue, the LocoBuffer USB "just worked" in 4.14, but does not in 4.16.

dsand47, 2019-07-15:
> @silverailscolo Except that, as reported in this issue, the LocoBuffer USB "just worked" in 4.14, but does not in 4.16.

In 4.14, the connection speed is `speed="57,600 baud"`.  In 4.16 it is `speed=""`.  This appears to have been introduced by #7107 in 4.15.9.

#7187 changes the default for USB devices from 19200 to 57600 which covers for the empty speed setting.

silverailscolo, 2019-07-17:
Sorry this did not show up in our tests, occurs if you _do not_ touch the port Additional details (and sometimes you can't as it is hidden). Filling in the correct speed in the profile.xml fixed it until you save your prefs again. Expect to have a fix in PR #7210.

closed 2020-07-25 by bobjacobsen