JMRI/JMRI #7197: PreferencesPanel crashes when running from source

toddthetrainnut, 2019-07-14:
I'm trying to add the ability to open the Bachrus Speedometer on startup. Unfortunately, without any changes on my end, I'm unable to open the preferences panel. This means I can't test that my changes to support launching the speedo on startup.

Here's the stacktrace:

`ant -f C:\\Users\\toddt\\Documents\\NetBeansProjects\\JMRI\\nbproject\\nbjdk.xml panelpro
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using C:\Program Files\Java\jdk1.8.0_111
panelpro:
Execute failed: java.io.IOException: Cannot run program "git" (in directory "C:\Users\toddt\Documents\NetBeansProjects\JMRI"): CreateProcess error=2, The system cannot find the file specified
JMRI.init:
JMRI.copyfiles:
Copying 4 files to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes
JMRI.jjtree:
JMRI.javacc:
JMRI.update-template-code:
Copying 1 file to C:\Users\toddt\Documents\NetBeansProjects\JMRI\java\tmp
JMRI.compile-generated-source:
Copying 1 file to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes\jmri
JMRI.compile:
Compiling 45 source files to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8'
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8'
2 warnings
JMRI.debug:
JMRI.panelpro:
Execute failed: java.io.IOException: Cannot run program "git" (in directory "C:\Users\toddt\Documents\NetBeansProjects\JMRI"): CreateProcess error=2, The system cannot find the file specified
JMRI.runtime-library-selection:
arch.lib.path C:\Users\toddt\Documents\NetBeansProjects\JMRI/lib/windows/x64:C:\Users\toddt\Documents\NetBeansProjects\JMRI/lib/windows
Launch normally (no debugger support)
2019-07-13 19:05:42,686 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-13 19:05:42,720 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\toddt\JMRI\log\messages.log [main]
2019-07-13 19:05:42,721 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\toddt\JMRI\log\session.log [main]
2019-07-13 19:05:42,726 apps.Apps                             INFO  - PanelPro version 4.17.2ish+toddt+20190714T0005Z starts under Java 1.8.0_111 on Windows 10 amd64 v10.0 at Sat Jul 13 19:05:42 CDT 2019 [main]
2019-07-13 19:05:50,009 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3eceb6c4 [main]
2019-07-13 19:05:50,338 node.NodeIdentity                     INFO  - Using jmri-aKA_LdphbhNiaaMlAq8JUt-3eceb6c4 as the JMRI storage identity for profile id 3eceb6c4 [AWT-EventQueue-0]
2019-07-13 19:05:50,556 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-13 19:05:50,556 jmrix.AbstractSerialPortController    WARN  - old profile format port speed value converted [main]
2019-07-13 19:05:50,918 usbdriver.UsbDriverAdapter            INFO  - NCE USB COM3 port opened at 19200 baud [main]
2019-07-13 19:05:50,984 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "Speedo" [main]
2019-07-13 19:05:50,984 jmrix.AbstractSerialPortController    WARN  - old profile format port speed value converted [main]
2019-07-13 19:05:51,004 serialdriver.SerialDriverAdapter      INFO  - COM4 port opened at 9600 baud, sees  DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-13 19:05:51,034 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 7.3.3 [AWT-EventQueue-0]
2019-07-13 19:05:51,297 util.FileUtilSupport                  INFO  - File path program: is C:\Users\toddt\Documents\NetBeansProjects\JMRI\ [main]
2019-07-13 19:05:51,297 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\toddt\JMRI\Todd's PowerCab\ [main]
2019-07-13 19:05:51,298 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\toddt\JMRI\Todd's PowerCab\ [main]
2019-07-13 19:05:51,298 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\toddt\JMRI\ [main]
2019-07-13 19:05:51,298 util.FileUtilSupport                  INFO  - File path home: is C:\Users\toddt\ [main]
2019-07-13 19:05:51,298 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\toddt\Documents\NetBeansProjects\JMRI\jython\ [main]
2019-07-13 19:05:51,839 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2019-07-13 19:05:55,761 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.util.ServiceConfigurationError: jmri.swing.PreferencesPanel: Provider jmri.profile.ProfilePreferencesPanel could not be instantiated
	at java.util.ServiceLoader.fail(ServiceLoader.java:232)
	at java.util.ServiceLoader.access$100(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:384)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at apps.gui3.tabbedpreferences.TabbedPreferences.init(TabbedPreferences.java:167)
	at apps.gui3.tabbedpreferences.TabbedPreferences.<init>(TabbedPreferences.java:118)
	at apps.gui3.tabbedpreferences.TabbedPreferencesFrame.lambda$getTabbedPreferences$0(TabbedPreferencesFrame.java:93)
	at java.util.Optional.orElseGet(Optional.java:267)
	at apps.gui3.tabbedpreferences.TabbedPreferencesFrame.getTabbedPreferences(TabbedPreferencesFrame.java:92)
	at apps.gui3.tabbedpreferences.TabbedPreferencesFrame.getTitle(TabbedPreferencesFrame.java:24)
	at jmri.util.JmriJFrame.generateWindowRef(JmriJFrame.java:216)
	at jmri.util.JmriJFrame.<init>(JmriJFrame.java:114)
	at jmri.util.JmriJFrame.<init>(JmriJFrame.java:124)
	at apps.gui3.tabbedpreferences.TabbedPreferencesFrame.<init>(TabbedPreferencesFrame.java:32)
	at apps.gui3.tabbedpreferences.TabbedPreferencesAction.lambda$actionPerformed$0(TabbedPreferencesAction.java:77)
	at java.util.Optional.orElseGet(Optional.java:267)
	at apps.gui3.tabbedpreferences.TabbedPreferencesAction.actionPerformed(TabbedPreferencesAction.java:76)
	at apps.gui3.tabbedpreferences.TabbedPreferencesAction.actionPerformed(TabbedPreferencesAction.java:99)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
[... 21 lines cut ...]
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
Caused by: java.lang.NoSuchMethodError: jmri.profile.ProfileManager.getAllProfiles()Ljava/util/ArrayList;
	at jmri.profile.ProfilePreferencesPanel.<init>(ProfilePreferencesPanel.java:76)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:380)
	... 54 more`

rhwood, 2019-07-14:
Try running `ant realclean` and re-compiling.

In NetBeans, right-click on `build.xml`, select "Run Target..." and then select "realclean", then try running again.

We just changed the library that generates the data the ServiceLoader needs, and it likely needs a complete recompilation to work.

bobjacobsen, 2019-07-14:
The email version of this Issue had some extra stuff at the top of the trace that's interesting (line breaks manually added and might not be right):
```
ant -f C:\\Users\\toddt\\Documents\\NetBeansProjects\\JMRI\\nbproject\\nbjdk.xml panelpro 
Warning: nbjdk.active=JDK_1.8 or nbjdk.home=${platforms.JDK_1.8.home} is an invalid Java platform; ignoring and using C:\Program Files\Java\jdk1.8.0_111 
panelpro: 
Execute failed: java.io.IOException: Cannot run program "git" (in directory "C:\Users\toddt\Documents\NetBeansProjects\JMRI"): CreateProcess error=2, The system cannot find the file specified 
JMRI.init: 
JMRI.copyfiles: 
Copying 4 files to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes 
JMRI.jjtree: 
JMRI.javacc: 
JMRI.update-template-code: 
Copying 1 file to C:\Users\toddt\Documents\NetBeansProjects\JMRI\java\tmp 
JMRI.compile-generated-source: 
Copying 1 file to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes\jmri 
JMRI.compile: 
Compiling 45 source files to C:\Users\toddt\Documents\NetBeansProjects\JMRI\target\classes warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.ServiceProviderProcessor' less than -source '1.8' 
warning: Supported source version 'RELEASE_7' from annotation processor 'org.netbeans.modules.openide.util.NamedServiceProcessor' less than -source '1.8' 
2 warnings 
JMRI.debug: 
JMRI.panelpro: 
Execute failed: java.io.IOException: Cannot run program "git" (in directory "C:\Users\toddt\Documents\NetBeansProjects\JMRI"): CreateProcess error=2, The system cannot find the file specified JMRI.runtime-library-selection: arch.lib.path C:\Users\toddt\Documents\NetBeansProjects\JMRI/lib/windows/x64:C:\Users\toddt\Documents\NetBeansProjects\JMRI/lib/windows 
Launch normally (no debugger support)
```

Something is going wrong in your build step, and there are several hints there.  The later exception is caused by the ServiceLoader code not being processed right.

toddthetrainnut, 2019-07-14:
Running antclean fixed it! Thanks Randall!

Bob, are you looking at the warnings about "invalid Java platform" or being unable to run "git" or something else?

bobjacobsen, 2019-07-14:
> Bob, are you looking at the warnings about "invalid Java platform" or being unable to run "git" or something else?

The git warnings. That part of the ant sequence is more complicated than I understand, and I thought those errors could be ending steps too soon. @rhwood’s solution is the correct one, though you might still be getting I.e. a wrong version number.

toddthetrainnut, 2019-07-14:
Ok, so I've got my startup option working. According to StartupActionFactory, my implementation of the interface, SpeedoStartupActionFactory, needs to be registered with a ServiceLoader. I manually added the class to the META-INF/services/apps.startup.StartupActionFactory as suggested by StartupActionFactory, which seems to be working. However, NetBeans won't let me commit the change to that META-INF file. Is there some other way I'm technically supposed to register my Factory?

danielb987, 2019-07-14:
Look at `jmri.jmrit.vsdecoder.VSDecoderCreationStartupActionFactory`

I don't think you need to add anything to META-INF since your code is included in the jmri.jar file.

toddthetrainnut, 2019-07-14:
I had missed the @ServiceProvider decorator on my class. Thanks for the tip, Daniel!

closed 2019-07-14 by toddthetrainnut