JMRI/JMRI #7207: CMRI baud rates mismatching in 4.16

JLang1951, 2019-07-17:
I updated JMRI from 4.15.5ish to 4.16 and I’m getting the same baud rate mismatch that was reported in previous threads. Used the release note trick to change the baud rate in preferences to a different number then back to 57,600 which the SIMNI boards are set to then did a save and re started and still no joy. This is my setup:

Dell Latitude Laptop running Mint 18.3
Java 1.8.0_201
2 TripLite Keyspan 19 USAHS USB adapters one to NCE Power Pro, 1 to CMRI RS232=RS485 converter to SMNIs

I’ve been following previous threads on this and tried to work my way to a solution but am not finding an answer. My setup has had no issues through various updates over time until I tried 4.16. Here is the system console with 4.15.5ish which works:

2019-07-15 09:00:45,289 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-15 09:00:46,134 util.Log4JUtil                        INFO  - This log is appended to file: /home/jessie/.jmri/log/messages.log [main]
2019-07-15 09:00:46,137 util.Log4JUtil                        INFO  - This log is stored in file: 
/home/jessie/.jmri/log/session.log [main]
2019-07-15 09:00:46,305 apps.Apps                             INFO  - PanelPro version 4.15.5ish+jenkins+20190321T2139Z+Rf386d8e starts under Java 1.8.0_201 on Linux amd64 v4.10.0-38-generic at Mon Jul 15 09:00:46 EDT 2019 [main]
2019-07-15 09:00:52,785 apps.Apps                             INFO  - Starting with profileLinux_Hardware_Connected.3ede90ae [main]
2019-07-15 09:00:53,441 node.NodeIdentity                     INFO  - Using ff8b17b0-48bb-11e8-8000-40490f74dbcd as the JMRI storage identity for profile id 3ede90ae [AWT-EventQueue-0]
2019-07-15 09:00:54,319 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-15 09:00:54,986 serialdriver.SerialDriverAdapter      INFO  - NCE ttyUSB0 port opened at 9600 baud [main]
2019-07-15 09:00:55,152 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2019-07-15 09:00:55,194 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2019-07-15 09:00:55,363 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-15 09:00:57,716 util.FileUtilSupport                  INFO  - File path program: is/home/jessie/JMRI/ [main]
2019-07-15 09:00:57,717 util.FileUtilSupport                  INFO  - File path preference: is/home/jessie/Dropbox/JMRI/ [main]
2019-07-15 09:00:57,718 util.FileUtilSupport                  INFO  - File path profile: is /home/jessie/.jmri/Linux_Hardware_Connected.jmri/ [main]
2019-07-15 09:00:57,720 util.FileUtilSupport                  INFO  - File path settings: is/home/jessie/.jmri/ [main]
2019-07-15 09:00:57,720 util.FileUtilSupport                  INFO  - File path home: is /home/jessie/ 
[main]
2019-07-15 09:00:57,721 util.FileUtilSupport                  INFO  - File path scripts: is /home/jessie/JMRI/jython/ [main]
2019-07-15 09:01:00,851 PanelPro.PanelPro                     INFO  - Main initialization done [main]

 
Here is the system console from 4.17.2ish where it opens at 9,600:

 
2019-07-15 09:11:03,099 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-15 09:11:03,213 util.Log4JUtil                        INFO  - This log is appended to file: 
/home/jessie/.jmri/log/messages.log [main]
2019-07-15 09:11:03,218 util.Log4JUtil                        INFO  - This log is stored in file:/home/jessie/.jmri/log/session.log [main]
2019-07-15 09:11:03,244 apps.Apps                             INFO  - PanelPro versio4.17.2ish+jenkins
+20190714T0751Z+R43088b3 starts under Java 1.8.0_201 on Linux amd64 v4.10.0-38-generic at Mon Jul 15 09:11:03 EDT 2019 [main]
2019-07-15 09:11:07,044 apps.Apps                             INFO  - Starting with profileLinux_Hardware_Connected.3ede90ae [main]
2019-07-15 09:11:07,470 node.NodeIdentity                     INFO  - Using ff8b17b0-48bb-11e8-8000-40490f74dbcd as the JMRI storage identity for profile id 3ede90ae [AWT-EventQueue-0]
2019-07-15 09:11:08,230 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-15 09:11:08,680 serialdriver.SerialDriverAdapter      INFO  - NCE ttyUSB0 port opened at 9600 baud [main]
2019-07-15 09:11:08,828 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2019-07-15 09:11:08,899 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2019-07-15 09:11:08,901 jmrix.AbstractSerialPortController    WARN  - old profile format port speed value converted [main]
2019-07-15 09:11:09,072 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 9600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-15 09:11:10,623 util.FileUtilSupport                  INFO  - File path program: is/home/jessie/JMRI/ [main]
2019-07-15 09:11:10,624 util.FileUtilSupport                  INFO  - File path preference: is/home/jessie/Dropbox/JMRI/ [main]
2019-07-15 09:11:10,626 util.FileUtilSupport                  INFO  - File path profile: is /home/jessie/.jmri/Linux_Hardware_Connected.jmri/ [main]
2019-07-15 09:11:10,627 util.FileUtilSupport                  INFO  - File path settings: is/home/jessie/.jmri/ [main]
2019-07-15 09:11:10,629 util.FileUtilSupport                  INFO  - File path home: is /home/jessie[main]
2019-07-15 09:11:10,630 util.FileUtilSupport                  INFO  - File path scripts: is /home/jessie/JMRI/jython/ [main]
2019-07-15 09:11:12,687 PanelPro.PanelPro                     INFO  - Main initialization done [main]

Notice the warn message: WARN  - old profile format port speed value converted [main]
2019-07-15 09:11:09,072 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 9600 baud  

I used the 4.17.2ish from Jenkens because it was said that had the fix but I still have the same problem as with 4.16.  The baud rate gets reset to 9,600. I also checked the profile xml’s in ./JMRI for the profile that I am using and both show the correct 57,600 baud rate for CMRI. Here is a <snip> of the xml for the regular profile and the machine profile:

<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">

     <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">

        <connection xmlns="" class="jmri.jmrix.nce.serialdriver.configurexml.ConnectionConfigXml" 

disabled="no" manufacturer="NCE" port="ttyUSB0" speed="" systemPrefix="N" userName="NCE">

            <options>

                <option>

                    <name>Eprom</name>

                    <value>2006 or later</value>

                </option>

            </options>

        </connection>

        <connection xmlns="" class="jmri.jmrix.cmri.serial.serialdriver.configurexml.ConnectionConfigXml" 

disabled="no" manufacturer="C/MRI" port="ttyUSB1" speed="57,600 baud" systemPrefix="C" userName="C/MRI">

<snip>

Thought that I would report it here as something has changed in 4.16 and it seems to effect other hardware setups as been reported. Thanks in advance.
John Lang

bobjacobsen, 2019-07-17:
Could you please follow the procedure in the release not (change speed, save, restart, change back, save, restart) and then post the profile file again?  Thanks.

JLang1951, 2019-07-17:
Ok, Started JMRI went to preferences, Changed CMRI from 57,600 to 9,600, SAVED, restarted JMRI, went to preferences, changed CMRI from 9,600 to 57,600, SAVED, restarted and still no dice.

2019-07-17 08:42:42,565 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-17 08:42:42,682 util.Log4JUtil                        INFO  - This log is appended to file: /home/jessie/.jmri/log/messages.log [main]
2019-07-17 08:42:42,686 util.Log4JUtil                        INFO  - This log is stored in file: /home/jessie/.jmri/log/session.log [main]
2019-07-17 08:42:42,717 apps.Apps                             INFO  - PanelPro version 4.17.2ish+jenkins+20190714T0751Z+R43088b3 starts under Java 1.8.0_201 on Linux amd64 v4.10.0-38-generic at Wed Jul 17 08:42:42 EDT 2019 [main]
2019-07-17 08:42:46,809 apps.Apps                             INFO  - Starting with profile Linux_Hardware_Connected.3ede90ae [main]
2019-07-17 08:42:47,020 node.NodeIdentity                     INFO  - Using ff8b17b0-48bb-11e8-8000-40490f74dbcd as the JMRI storage identity for profile id 3ede90ae [AWT-EventQueue-0]
2019-07-17 08:42:47,459 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-17 08:42:47,864 serialdriver.SerialDriverAdapter      INFO  - NCE ttyUSB0 port opened at 9600 baud [main]
2019-07-17 08:42:48,000 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2019-07-17 08:42:48,020 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2019-07-17 08:42:48,190 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 9600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-17 08:42:49,452 util.FileUtilSupport                  INFO  - File path program: is /home/jessie/JMRI/ [main]
2019-07-17 08:42:49,454 util.FileUtilSupport                  INFO  - File path preference: is /home/jessie/Dropbox/JMRI/ [main]
2019-07-17 08:42:49,455 util.FileUtilSupport                  INFO  - File path profile: is /home/jessie/.jmri/Linux_Hardware_Connected.jmri/ [main]
2019-07-17 08:42:49,456 util.FileUtilSupport                  INFO  - File path settings: is /home/jessie/.jmri/ [main]
2019-07-17 08:42:49,457 util.FileUtilSupport                  INFO  - File path home: is /home/jessie/ [main]
2019-07-17 08:42:49,458 util.FileUtilSupport                  INFO  - File path scripts: is /home/jessie/JMRI/jython/ [main]
2019-07-17 08:42:50,731 PanelPro.PanelPro                     INFO  - Main initialization done [main

KenC57, 2019-07-17:
After those changes, did you check to see if the profile.xml was showing
9600, or what for the speed?

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

JLang1951, 2019-07-17:
Yes Ken the profile.xml is exactly the same as the one I posted above showing "57.600 baud" in the speed statement. Just a note that in Linux the profile folder in ./JMRI that I'm using has a sub folder that has an alpha numeric name (looks like the MAC address of the adapter) that also has a profile.xml file in it. That matches exactly with the other XML which matches to what I posted above. All show the 57,600 baud.

pabender, 2019-07-17:
> 2019-07-17 08:42:48,020 xml.AbstractSerialConnectionConfigXml INFO - Starting to connect for "C/MRI" [main]
> 2019-07-17 08:42:48,190 serialdriver.SerialDriverAdapter INFO - ttyUSB1 port opened at 9600 baud with DTR: true RTS: true DSR: false CTS: false CD: false [main]

I am confused as to what the issue is.  Regardless of what the profile is showing, the log posted shows the port opened at 9600bps.

Are we sure the communication issue is due to a baud rate mismatch?

bobjacobsen, 2019-07-17:
I believe the hardware in this case is set for 57,600

bobjacobsen, 2019-07-17:
@JLang1951 Thanks for your help with this. Are you by any chance using a non-US language for JMRI? On the main window, does it show the Java version as ending in `(en_US)` or does it show something else?

[image]

JLang1951, 2019-07-17:
Bob, it was set to just en so I changed it (en_us) restarted and it still falls back to 9,600. And yes all 5 SIMNI's are hardware set (dip switches) to 57,600. The status light (green Led) flashes rapidly when it initializes. At initialization the RX/TX lights blink a couple of times showing that there is some kind of handshake going on. If I drop back to 4.15.5ish and start up JMRI the handshake RX/TX flashes and the rapidly flashing status light immediately drops to its one second on one second off heartbeat meaning those SMINI's are ready to do business.

Bob after setting the language to (us_en) should I make a new profile and start again. Anyway I have some errands that I have to attend to so won,t be back till afternoon. Thanks all for looking into this.
John L.

silverailscolo, 2019-07-17:
Really sorry this did not show up in our tests, only happens if you _do not_ touch the port Additional details (and sometime you can't as it is hidden). Selecting a speed from the Additional details or filling in the correct speed in the ``profile.xml`` fixed it until you save your prefs again.
Expect to have a fix ready in PR #7210.

JLang1951, 2019-07-27:
I just tried test version 4.17.2 and still have basically the same problem. I then tried making a new profile to see if that would make a difference but still have the same problem. Latest sys. console:

19-07-27 14:28:42,611 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-27 14:28:42,724 util.Log4JUtil                        INFO  - This log is appended to file: /home/jessie/.jmri/log/messages.log [main]
2019-07-27 14:28:42,729 util.Log4JUtil                        INFO  - This log is stored in file: /home/jessie/.jmri/log/session.log [main]
2019-07-27 14:28:42,749 apps.Apps                             INFO  - PanelPro version 4.17.2+Re9dde08 starts under Java 1.8.0_201 on Linux amd64 v4.10.0-38-generic at Sat Jul 27 14:28:42 EDT 2019 [main]
2019-07-27 14:28:46,344 apps.Apps                             INFO  - Starting with profile Linux_Hardware_Connected.3ede90ae [main]
2019-07-27 14:28:46,565 node.NodeIdentity                     INFO  - Using ff8b17b0-48bb-11e8-8000-40490f74dbcd as the JMRI storage identity for profile id 3ede90ae [AWT-EventQueue-0]
2019-07-27 14:28:46,996 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-27 14:28:47,418 serialdriver.SerialDriverAdapter      INFO  - NCE ttyUSB0 port opened at 9600 baud [main]
2019-07-27 14:28:47,554 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2019-07-27 14:28:47,569 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2019-07-27 14:28:47,570 jmrix.AbstractSerialPortController    WARN  - old profile format port speed value converted [main]
2019-07-27 14:28:47,736 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 19200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-27 14:28:49,110 util.FileUtilSupport                  INFO  - File path program: is /home/jessie/JMRI/ [main]
2019-07-27 14:28:49,112 util.FileUtilSupport                  INFO  - File path preference: is /home/jessie/Dropbox/JMRI/ [main]
2019-07-27 14:28:49,113 util.FileUtilSupport                  INFO  - File path profile: is /home/jessie/.jmri/Linux_Hardware_Connected.jmri/ [main]
2019-07-27 14:28:49,114 util.FileUtilSupport                  INFO  - File path settings: is /home/jessie/.jmri/ [main]
2019-07-27 14:28:49,116 util.FileUtilSupport                  INFO  - File path home: is /home/jessie/ [main]
2019-07-27 14:28:49,117 util.FileUtilSupport                  INFO  - File path scripts: is /home/jessie/JMRI/jython/ [main]
2019-07-27 14:28:50,779 PanelPro.PanelPro                     INFO  - Main initialization done [main]

The only difference between this console and the 4.16 console is the baud rate for CMRI defaults to 19,200 where as in 4.16 it defaulted to 9,600. I want to point out again that my profile xml's show that the speed settings for the CMRI show 57,600 (see above). This is not an issue of a blank speed setting in the profile xml's. 

Also I want to point out that there has never been a timeout failure in the system console nor a red connection failure in the main JMRI splash screen for as long as I've been using CMRI (about five years). No mater how you configure the serial port it seems to blindly except that it is connected when in fact the connection could be wrong. It's never been a bother to me because if I didn't set the connection right the flashing green status lights on the SIMI's would let me know.

John Lang

KenC57, 2019-07-27:
I think this is a key line for what's happening:

2019-07-27 14:28:47,570 jmrix.AbstractSerialPortController WARN - old
profile format port speed value converted [main]

So it is seeing the old flavor (or thinks it is) of baud rate and converts
it. I'm getting the feeling this is what fails and you end up with a blank
baud rate which then is the default, and the problem. Conjecting here but
the pattern fits.

Now if you do a save of the profile, what value does is put into the new
save of the profile.xml?
-ken c

JLang1951, 2019-07-28:
Ken
Here are the steps that I tried this morning. I looked at the profile xml's before starting JMRI. Speed setting at 57,600. I then opened JMRI and the connection reverted to 19,600. Their was no warning in the sys. console about old profile conversion this time. I then went to preferences and did a save as you asked. I then went to the profile xml and the speed setting was still at 57,600 after the save as it was in the prferences GUI. I then opened JMRI went to preferences changed the baud to 9,600 then SAVED, looked at the profile xml and the speed was changed to 9,600. So setting the baud rate in preferences does write the change to the profile xml. I then opened JMRI went to the preferences changed the baud back to 57,600 SAVED looked at the profile and it was back to 57,600 as it should be, restarted JMRI and it opened the port at 19,600 again. No dice. Sys. console:

19-07-27 14:28:42,611 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2019-07-27 14:28:42,724 util.Log4JUtil                        INFO  - This log is appended to file: /home/jessie/.jmri/log/messages.log [main]
2019-07-27 14:28:42,729 util.Log4JUtil                        INFO  - This log is stored in file: /home/jessie/.jmri/log/session.log [main]
2019-07-27 14:28:42,749 apps.Apps                             INFO  - PanelPro version 4.17.2+Re9dde08 starts under Java 1.8.0_201 on Linux amd64 v4.10.0-38-generic at Sat Jul 27 14:28:42 EDT 2019 [main]
2019-07-27 14:28:46,344 apps.Apps                             INFO  - Starting with profile Linux_Hardware_Connected.3ede90ae [main]
2019-07-27 14:28:46,565 node.NodeIdentity                     INFO  - Using ff8b17b0-48bb-11e8-8000-40490f74dbcd as the JMRI storage identity for profile id 3ede90ae [AWT-EventQueue-0]
2019-07-27 14:28:46,996 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "NCE" [main]
2019-07-27 14:28:47,418 serialdriver.SerialDriverAdapter      INFO  - NCE ttyUSB0 port opened at 9600 baud [main]
2019-07-27 14:28:47,554 nce.NceConnectionStatus               INFO  - NCE EPROM revision = 6.2.1 [AWT-EventQueue-0]
2019-07-27 14:28:47,569 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "C/MRI" [main]
2019-07-27 14:28:47,736 serialdriver.SerialDriverAdapter      INFO  - ttyUSB1 port opened at 19200 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2019-07-27 14:28:49,110 util.FileUtilSupport                  INFO  - File path program: is /home/jessie/JMRI/ [main]
2019-07-27 14:28:49,112 util.FileUtilSupport                  INFO  - File path preference: is /home/jessie/Dropbox/JMRI/ [main]
2019-07-27 14:28:49,113 util.FileUtilSupport                  INFO  - File path profile: is /home/jessie/.jmri/Linux_Hardware_Connected.jmri/ [main]
2019-07-27 14:28:49,114 util.FileUtilSupport                  INFO  - File path settings: is /home/jessie/.jmri/ [main]
2019-07-27 14:28:49,116 util.FileUtilSupport                  INFO  - File path home: is /home/jessie/ [main]
2019-07-27 14:28:49,117 util.FileUtilSupport                  INFO  - File path scripts: is /home/jessie/JMRI/jython/ [main]
2019-07-27 14:28:50,779 PanelPro.PanelPro                     INFO  - Main initialization done [main]

Obviously I'm just chasing my tail now with this. For now I'll just revert to 4.15.5ish so I can run things. If there is anything else that you want me to try I'm more than willing to help. I really do appreciate all that you folks do to make this a great program.

Thanks,
John Lang

KenC57, 2019-07-28:
John,

It seems you had the right things done. You followed what I understood to be
the right fix: save different baud rate, restart, set wanted baud rate,
restart, then it works. But it doesn't seem to have done that for you. My
reference system, currently on 4.16, is running at 9600 baud and that gets
me 20mS cycle time, so I'm happy. It is an older computer (Vista yuc!) but
does the job. We've got a op session on Tuesday. After that, I'll try update
to 4.17.2 and see what happens. But I just saw a code post that may be the
solution to this problem. So a bit more time and we may have figured out the
real issue in the code.
-ken c

bobjacobsen, 2019-07-28:
Dave Sand has provided a possible fix.  Installers including it will be built shortly (about an hour) as 3393 at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/   When they're done, you should be able to get them at http://jmri.tagadab.com/jenkins/job/Development/job/Packages/3393

JLang1951, 2019-07-29:
It works! Thanks so much Dave for the fix. I'll run the system later today with my panels loaded and five or six trains running at once to generate a lot of traffic on the network. I'm sure all be fine but want to be sure before I close this ticket. Again many thanks.

John Lang

closed 2019-07-30 by JLang1951