JMRI/JMRI #7214: DCC4PC, anyone able to help this user?

KenC57, 2019-07-19:
From the JMRI reports  help request: ardmay@btinternet.com

The hardware support for dcc4pc railcom reader boards is not working!

I have tested  the facility back through the releases with different effects.

jmri consistantly shows the following message in all testing:-
dcc4pc.Dcc4PcSensorManager            WARN  - timeout awaiting poll response for board 1 [DCC4PC Sensor Poll]

Rel
4.8       Works but slow. dcc4pc monitor shows hex on reply data.
4.10      Doesn't work. dcc4pc monitor shows hex on reply data.
4.11.9   Works but slow. dcc4pc monitor shows hex on reply data.
4.12      Works with errors. dcc4pc monitor shows hex on reply data.
4.13.7   Works but slow. dcc4pc monitor hieroglyphics on reply data.
4.14      Works but very slow. dcc4pc monitor hieroglyphics on reply data.
4.15.1   Works but slow. dcc4pc monitor hieroglyphics on reply data.
4.15.6   Works but slow. No address id. shown. dcc4pc monitor hieroglyphics on reply data.
4.15.9   Works but slow. No address id. shown. dcc4pc monitor hieroglyphics on reply data.
4.16      Doesn't work. dcc4pc monitor hieroglyphics on reply data.
4.17.1   Doesn't work. dcc4pc monitor hieroglyphics on reply data.
4.17.2   Doesn't work. dcc4pc monitor hieroglyphics on reply data.

The hardware has been tested by DCC4PC and is working ok!

I have tried to contact Kevin Dickerson the author of the code but have drawn a blank.

I would be most grateful if someone could have a look at this outstanding fault please.

Regards
Michael Smith 
=========================================================
JMRI Version: 4.17.2ish+jenkins+20190711T1352Z+Rfbf0f51   
JMRI configuration file name: null   (from org.jmri.apps.Apps.configFilename system property)
JMRI main window name: JMRI System Console   
JMRI Application: PanelPro   
Connection 0: MERG connected via CAN via MERG CAN-RS or CAN-USB on COM12 Disabled false   
Connection 1: DCC4PC connected via DCC4PC on COM22 Disabled false   
Available Communication Ports:
 Found 2 serial ports
 Port: COM12 - in use by: JMRI app   
[DCC4PC_Issues.zip](https://github.com/JMRI/JMRI/files/3410639/DCC4PC_Issues.zip)

 Port: COM22 - in use by: JMRI app   
Active profile: My JMRI Railroad   
Profile location: C:\Users\Owner\JMRI\My_JMRI_Railroad   
Profile ID: My_JMRI_Railroad.3e0692e0   
JMRI Network ID: jmri-BC773774D687-3e0692e0
JMRI Storage ID: jmri-mD03XcPaGhOiaaUhB1Cn_h-3e0692e0
Preferences directory: C:\Users\Owner\JMRI\My_JMRI_Railroad\   
Program directory: C:\Program Files\JMRI   
Roster index location: C:\Users\Owner\JMRI\My_JMRI_Railroad\roster.xml   
Current panel file: [none]   
Audio factory type: [not initialised]   
java.version: 1.8.0_211   
java.vendor: Oracle Corporation   
java.home: C:\Program Files\Java\jre1.8.0_211   
java.vm.version: 25.211-b12   
java.vm.vendor: Oracle Corporation   
java.vm.name: Java HotSpot(TM) Client VM   
java.specification.version: 1.8   
java.specification.vendor: Oracle Corporation   
java.specification.name: Java Platform API Specification   
java.class.version: 52.0   
java.class.path: .;classes;jmri.jar;lib\beansbinding-1.2.1.jar;lib\bluecove-2.1.1-SNAPSHOT.jar;lib\bluecove-bluez-2.1.1-SNAPSHOT.jar;lib\bluecove-gpl-2.1.1-SNAPSHOT.jar;lib\commons-compress-1.18.jar;lib\commons-io-1.3.1.jar;lib\commons-io-2.6.jar;lib\commons-lang3-3.2.1.jar;lib\commons-lang3-3.7.jar;lib\commons-text-1.2.jar;lib\gluegen-rt.jar;lib\hid4java-0.5.0.jar;lib\jackson-annotations-2.9.8.jar;lib\jackson-core-2.9.8.jar;lib\jackson-databind-2.9.8.jar;lib\jakarta-regexp-1.5.jar;lib\javacsv.jar;lib\javax.servlet-api-3.1.0.jar;lib\jdom-2.0.5.jar;lib\jdom.jar;lib\jetty-http-9.4.7.v20170914.jar;lib\jetty-io-9.4.7.v20170914.jar;lib\jetty-security-9.4.7.v20170914.jar;lib\jetty-server-9.4.7.v20170914.jar;lib\jetty-servlet-9.4.7.v20170914.jar;lib\jetty-util-9.4.7.v20170914.jar;lib\jhall.jar;lib\jhidrawplugin.jar;lib\jinput.jar;lib\jinputvalidator-0.4.0.jar;lib\jlfgr-1_0.jar;lib\jmdns-3.5.1.jar;lib\jmdns-3.5.3.jar;lib\jmdns-3.5.5.jar;lib\jmdns.jar;lib\jna-4.4.0.jar;lib\jna-platform-4.4.0.
 jar;lib\joal.jar;lib\json-schema-validator-1.0.10.jar;lib\json-schema-validator-1.0.4.jar;lib\jsplitbutton-1.3.1.jar;lib\jul-to-slf4j-1.7.25.jar;lib\jython-standalone-2.7.0.jar;lib\libusb4java-1.2.0-linux-arm.jar;lib\libusb4java-1.2.0-linux-x86.jar;lib\libusb4java-1.2.0-linux-x86_64.jar;lib\libusb4java-1.2.0-osx-x86.jar;lib\libusb4java-1.2.0-osx-x86_64.jar;lib\libusb4java-1.2.0-windows-x86.jar;lib\libusb4java-1.2.0-windows-x86_64.jar;lib\log4j-1.2.17.jar;lib\mailapi.jar;lib\mqtt-client-0.4.0.jar;lib\openlcb.jar;lib\pi4j-core.jar;lib\pi4j-device.jar;lib\pi4j-gpio-extension.jar;lib\pi4j-service.jar;lib\purejavacomm-1.0.1.jar;lib\slf4j-api-1.7.25.jar;lib\slf4j-log4j12-1.7.25.jar;lib\smtp.jar;lib\thumbnailator-0.4.8.jar;lib\usb-api-1.0.2.jar;lib\usb4java-1.2.0.jar;lib\usb4java-javax-1.2.0.jar;lib\vecmath-1.5.2.jar;lib\websocket-api-9.4.7.v20170914.jar;lib\websocket-client-9.4.7.v20170914.jar;lib\websocket-common-9.4.7.v20170914.jar;lib\websocket-server-9.4.7.v20170914.jar;lib\websocket-
 servlet-9.4.7.v20170914.jar;lib\xAPlib.jar;lib\xbee-java-library-1.2.0.jar;lib\xbee-java-library-1.2.1.jar;lib\xercesImpl.jar   
java.library.path: .;lib\x86;lib   
java.compiler: null   
java.ext.dirs: C:\Program Files\Java\jre1.8.0_211\lib\ext;C:\Windows\Sun\Java\lib\ext   
file.encoding: UTF-8   
os.name: Windows 7   
os.arch: x86   
os.version: 6.1   
python.home: C:\Users\Owner\JMRI\systemfiles   
python.path: null   
python.cachedir: null   
python.cachedir.skip: null   
python.startup: null   
user.name: Owner   
user.home: C:\Users\Owner   
user.dir: C:\Program Files\JMRI   
user.country: GB   
user.language: en   
user.timezone: Europe/London   
jmri.log.path: C:\Users\Owner\JMRI\log\   
FileSystemView#getDefaultDirectory(): C:\Users\Owner\Documents
FileSystemView#getHomeDirectory(): C:\Users\Owner\Desktop
Default JFileChooser(): C:\Users\Owner\Documents
Screen size h:768, w:1366 Inset t:0, b:40; l:0, r:0
Screen resolution: 96
Environment max bounds: java.awt.Rectangle[x=0,y=0,width=1366,height=728]
bounds[0] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
bounds[1] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
bounds[2] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
bounds[3] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
bounds[4] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
bounds[5] = java.awt.Rectangle[x=0,y=0,width=1366,height=768]
Device: \Display0 bounds = java.awt.Rectangle[x=0,y=0,width=1366,height=768] sun.awt.Win32GraphicsConfig@18da814[dev=Win32GraphicsDevice[screen=0],pixfmt=2]
JmriInsets t:0, b:40; l:0, r:0
Network Interface: wlan1
 Long Name: Intel(R) Centrino(R) Wireless-N 1030
 Host Name: LaptopPC.home
 IP address: 192.168.1.71
ZeroConfService host: jmri-bc773774d687-3e0692e0 running 0 service(s)
ZeroConfService host: jmri-bc773774d687-3e0692e0 running 0 service(s)
ZeroConfService host: jmri-bc773774d687-3e0692e0-1 running 0 service(s)
ZeroConfService host: jmri-bc773774d687-3e0692e0 running 0 service(s)
ZeroConfService host: jmri-bc773774d687-3e0692e0 running 0 service(s)
ZeroConfService host: jmri-bc773774d687-3e0692e0 running 0 service(s)

KenC57, 2019-07-19:
I added a zip file of the profile and logs, but the link is about 1/3 of the way down in the other text.
-ken c

bobjacobsen, 2019-07-27:
Has this been brought up on JMRIusers or some DCC4PC forum?  Just doing an "Upload debugging information" operation doesn't bring it to the attention of the user communities. As [it says:](http://jmri.org/help/en/package/jmri/jmrit/mailreport/Report.shtml)

```
Note: Uploading this information will not necessarily result in somebody leaping up to fix your problem. This is not a support request, it's just a convenient way to get your program information where people can see it.

If you want quick help with a problem, you should talk with some JMRI users in your area (it's always good to talk to friends), or ask on the JMRI users group.
``

closed 2021-04-14 by stale[bot]