JMRI/JMRI #7216: NamedBeanComboBox -- Editable Issue

dsand47, 2019-07-20:
Creating a new bean with editable enabled works fine.
[image]

Displaying any selection shows the system name instead of the display name.
[image]

[image]

rhwood, 2019-07-24:
The problem is that the selected NamedBean's `toString()` method is getting invoked to populate the ComboBox editor. I haven't figured out the fix yet, but have identified the problem.

dsand47, 2019-07-30:
Based on my understanding of #7238, even if the display name can be shown instead of the system name, it will not be possible to create new blocks by entering a new user name.  

There is currently a bug in BlockManager#provideBlock(String) that creates only system names contrary to the Javadoc.

rhwood, 2019-07-30:
Since the `provide()` method is documented as only attempting to use the system name to generate a new NamedBean, #7238 does change the validation of user input to only be concerned with the system name. That said, if an internal manager has no restrictions on the system name, it should validate the same as a user name would, so that if the BlockManager were fixed, the NamedBeanComboBox should still allow the user's input to be used.

That said, `git blame` shows that `provideBlock()` is behaving as it did when it was [originally written](https://github.com/JMRI/JMRI/commit/2abfbf9c800b0711b50d8686376b9e261dc4e3b9), so I think the JavaDocs (added about [five years later](https://github.com/JMRI/JMRI/commit/3cec6e167187f141f12d8799a75236b99ab5f2fc)), are just wrong, and have always been wrong.

dsand47, 2019-07-31:
The BlockManager provideBlock(String) method is only used by Panel Editor **Add Block Contents Label** when an entry has been entered into the **Add item to table** name text field.  Treating the input as a user name when entered without the IB prefix seems like a low risk change.

Prior to 4.17.2, block assignments to Layout Editor track components used the JmriBeanComboBox.  When an entry is selected, **provideLayoutBlock** in **LayoutEditor** is called to return the layout block and if necessary create the Layout Block and possibly the corresponding Block if a new user name was entered.

With the change to NamedBeanComboBox, new blocks using the editable option are created using the **ProvidingManager** structure before the Layout Editor processes gets control.  This can simplify provideLayoutBlock since the related block will always exist.  

With the necessary change provideBlock will work.  Once the editable combo box name issue (toString) has been resolved, I will need to decide whether I want to re-enable block creation during block assignment.

PR #7263 (2019-08-04): fix: render NamedBeans properly in editor
Use getDisplayName() instead of toString() to render the NamedBeans in the ComboBoxEditor.
Also fixes how tooltip validation messages are overridden.

Fix #7216
closed 2019-08-04 by bobjacobsen