JMRI/JMRI #7229: Improve Interaction between JMRI throttle and advanced consists

pabender, 2019-07-24:
This issue is intended to gather my thoughts around a user request in https://groups.io/g/jmriusers/message/161984 for future implementation.

To summarize the request, users would like the JMRI throttle to utilize the value of CV19 if the value is non-zero.

The current behavior of the throttle is (if a roster entry is selected) the primary address (long or short) is the only address registered with the throttle.

The consisting tool includes a throttle button.  What that button does for an advanced consist is to create a throttle window as follows:
1) The Speed Controls are associated with the consist address
2) The Function Controls are associated with the lead locomotive address from the consist.
3) The address selection section of the throttle shows the lead locomotive address.

Proposed changes to the throttle:

- [ ] When a roster entry is selected, the throttle UI will:

1) Read CV19 from the roster entry
2) If the CV19 entry indicates a consist is associated with the entry, the throttle will be configured as the Consist Tool configures the throttle.
3) If the CV19 entry indicates a consist is not associated with the entry, the throttle will be configured to only use the primary address.

Proposed change to the consist tool:

- [ ] When a locomotive is added to an advanced consist using a roster entry, the consisting tool will update the roster entry with the new CV19 value.

- [ ] Similarly, when a locomotive that was selected using a roster entry is removed from an advanced consist, the CV19 value in the roster entry will be set to 0.

McLae, 2019-07-24:
Add a configuration option to ignore or use CV19 as throttle address. 
Or better yet, if CV19 is not zero, popup a question "Use CV19 for throttle address?" The configuration setting would enable/suppress the popup.

When testing a Lok just set up with CV19 set, here is the sequence:
1. Select roster entry.
2. Create throttle for that entry. (With long or short address)
3. Look up consist address for that entry.
4. Set address to CV19 value.
5. Test Lok.

Steps 3 and 4 are a hassle. Especially when the value is right there in the roster entry.

And remember, not all users create consists with the tool.

pabender, 2019-07-24:
> Add a configuration option to ignore or use CV19 as throttle address.
> Or better yet, if CV19 is not zero, popup a question "Use CV19 for throttle address?" The configuration setting would enable/suppress the popup.
> 

I am not a fan of adding pop ups.

The option to load the throttle without the possibility of checking CV19 already exists.  You just enter the address.

> When testing a Lok just set up with CV19 set, here is the sequence:
>
> 1. Select roster entry.
> 2. Create throttle for that entry. (With long or short address)
> 3. Look up consist address for that [...]

Just for clarification, the value of CV19 is NOT stored in the roster index, which is the values read to populate the Drop down box.

In order to get CV19, you need to read an additional file, which contains the CVs for the roster entry.

> And remember, not all users create consists with the tool.

True, but the tool needs to be updated so that it’s interactions with advanced consists can work with a throttle configured by the user.

MikeHeintzman, 2019-07-25:
In reading the above and the ideas in the mail posting my only concern would be adding processing  overhead to the program where it’s not necessary. Would it be easier to add some kind of Auto-Build feature to the existing Consist tool that would build the consist based on the contents of CV19 in the roster entry? That way it keeps consist related stuff in the same program area.  It would also be nice to have a column in the Roster listing the consist address, too.

McLae, 2019-07-25:
A roster column (optional) that can be sorted would be great.
Lets you sort which Lok is in which consist. 
Especially for clubs, where multiple users have set up their own Loks.
Adding CV19 to roster entry may not be trivial, but might make processing quicker. As one of the three addresses for a Lok, makes sense to handle on same level as CV1 and CV17/18.

And if this all work integrated, I may learn how to use the consist tool.

MikeHeintzman, 2019-07-25:
As with many things, I had no idea the consist tool existed before this discussion started. I can see some real advantages to having a consist assignment show in the Roster panel. I’m wondering how well an auto consist build would work to show you what Lok is (and what shouldn’t be) assigned to a consist, especially on a club layout with people moving in and out.

McLae, 2019-07-25:
For the home user the auto consist build tool would let you quickly reset consists on the fly. Lots of good in that. For clubs, if the roster is updated with changes, and cv19 is on the roster page, you can quickly tell which consist address are in use with a sort. Big win there. (any way to do that now?)Lots of up side here.
And, you can ignore if you do not care? Best part of JMRI, you can pick and choose what parts to use without breaking *this* if you do not use *that*.
Thomas Stephens DeSoto, TX

pabender, 2019-07-25:
> In reading the above and the ideas in the mail posting my only concern would be adding processing overhead to the program where it’s not necessary. Would it be easier to add some kind of Auto-Build [...]

The Consist Tool was really designed to interact with the command station’s tools for building consists, so it intentionally stays away from the contents of the roster file.  It does store consist information for later retrieval, but it is outside of the roster.

Restricting my thoughts to advanced consists, several manufacturers ( Lenz, Digitrax, NCE ) have special features in the command station if you build the Consist using the instructions they provide e.g. Lenz systems allow you to select any locomotive in the Consist and control the speed and direction of the Consist while controlling that locomotives functions ).  The consisting tool tends to defer to the command station’s view of the world because it has the most up to date information.

My biggest issue with using the contents of the roster files to build the Consist is that for large rosters this will take some time.

Perhaps that is something that can be added to the Consist tool in a menu.  I’ll look into that.

> It would also be nice to have a column in the Roster listing the consist address, too.

That is where this discussion started on the users list.  I’m not a huge fan of this idea because I see consists as being more dynamic than anything we show in the roster listing.

pabender, 2019-07-25:
> A roster column (optional) that can be sorted would be great.
> Lets you sort which Lok is in which consist.
> Especially for clubs, where multiple users have set up their own Loks.
> Adding CV19 to roster entry may not be trivial, but might make processing quicker. 

I am not sure I follow your thinking on the speed here.

> As one of the three addresses for a Lok, makes sense to handle on same level as CV1 and CV17/18.

The roster screen does not actually show both the short address and the long address.  It only shows the active primary address.

Unlike the primary address, it is easier to change the Consist address outside of JMRI on most systems.

McLae, 2019-07-25:
Look at it this way:
If you set CV19, this changes the data in the decoder. Should not the roster file have the latest data on what is in the CVs?
When I do a write all sheets, I just broke the consist! And no clue what happened!

pabender, 2019-07-25:
> Look at it this way:
> If you set CV19, this changes the data in the decoder. Should not the roster file have the latest data on what is in the CVs?
> When I do a write all sheets, I just broke the consist! And no clue what happened!

This is true, and one of the reasons I am working on changes, but we can’t prevent a user from changing CVs without JMRI’s knowledge.  

Exactly the same situation occurs if you create an advanced Consist with a handheld controller.

MikeHeintzman, 2019-07-25:
On the program  speed issue, probably my misunderstanding of the original proposal, but it seemed like the program would have been constantly rereading the roster files to check the contents of CV19.

pabender, 2019-07-25:
> On the program speed issue, probably my misunderstanding of the original proposal, but it seemed like the program would have been constantly rereading the roster files to check the contents of CV19.

The proposal isn’t actually to read the values, but to update the values in the roster file when the Consist tool changes them.  So yes, the file gets read, but only when an actual change is made by the tool.

bobjacobsen, 2019-07-25:
> When I do a write all sheets, I just broke the consist! And no clue what happened!

Decoder definitions don’t (or at least, are not supposed to, though a few do) contain CV19. So this shouldn’t happen. 

CV19, though formally a configuration variable, is really more of a runtime item that can be changed by special instructions, etc, in ways different from other CVs . It’s unfortunate that NMRA politics made it impossible to really separate these and get clean implementations. But the reality is that the CV19 value really isn’t a roster-like thing.

bobjacobsen, 2019-07-25:
> A roster column (optional) that can be sorted would be great.
> Lets you sort which Lok is in which consist.
> Especially for clubs, where multiple users have set up their own Loks.

Could you say a few more words about _why_ that would be useful?  After you’ve sorted that, what are you going to do with that information?

McLae, 2019-07-25:
Looking at the roster grid, if I can sort by CV19, I get a list of which Loks are using consist 14, and 22, 87 and 99.So if a member asks who has Loks on consist #22, the answer is Thomas (me) and they pick a different number. IN my modular club, the engine master enters all consists into JMRI. He can look and see who has what consist active, and when the next member shows up to run, and asks if consist 14 is in use, sort, and yes/no pops right up.
I admit, a single user on the home layout may not care, unless he has an operating session with BYOP.
CV19 as run time variable, got lost in translation somewhere.Most of the folks I run with set a consist, with CV21 & 22 set for f/r functions, and do not change these CVs for years. (I have Loks set to consist #21 for about a decade)
It always seemed a common cross-system feature, that advanced consist would run on any system.I run on EasyDCC and Digitraxx, program with Digitraxx, Lokprogrammer, EasyDCC, and SPROG. (No NCE - yet)
Thomas Stephens DeSoto, TX 

 

Could you say a few more words about _why_ that would be useful? After you’ve sorted that, what are you going to do with that information?

pabender, 2019-07-25:
> Looking at the roster grid, if I can sort by CV19, I get a list of which Loks are using consist 14, and 22, 87 and 99.So if a member asks who has Loks on consist #22, the answer is Thomas (me) and [...]

Except for the name of the operator, you can already get this information from the Consist tool.

Is const 22 in use?  Use the drop down list of Consist numbers to check.
What locomotives are in Consist 5? Select the Consist from the list and it appears.

MikeHeintzman, 2019-07-25:
‘Is const 22 in use? Use the drop down list of Consist numbers to check.

> 
> What locomotives are in Consist 5? Select the Consist from the list and it appears.’
Will a locomotive that has CV19 set to 22 but has not been manually entered into the consist builder appear in the list? Or if you’ve changed a loco’s CV19 and been distracted by a bright shiny before entering it into the consist builder appear in the list? I don’t use consists often (my layout is mainly switching) so I don’t have a horse in this, but I can see the use of a Roster column for Consists, or the ability to build a consist automatically in the builder.

pabender, 2019-07-25:
> Will a locomotive that has CV19 set to 22 but has not been manually entered into the consist builder appear in the list? Or if you’ve changed a loco’s CV19 and been distracted by a bright shiny before entering it into the consist builder appear in the list? 

The answer to this question currently is maybe.

If you set CV19 using a programmers, the consist manager does not know anything about that change.  This is true whether we are talking about JMRI or any other device.

On a Lenz or Digitrax system, If you set CV19 using the manufacturers procedure, the the Consist tool will read the memory from the command station, and show that.

On an NCE system, it is possible to get at least partial Consist information from the command station ( The command station itself has a limit as to the number of locomotives it stores ).

On Lenz, Digitrax, NCE, and CVP, command station assisted consists are also supported.  ( CVP doesn’t have native support for advanced consists, but it allows us to send arbitrary DCC packets, so we manage them ourselves ).

On all other systems, we manage the consists ourselves and only support advanced consists.

dheap, 2019-07-26:
@pabender That is where this discussion started on the users list. I’m not a huge fan of this idea because I see consists as being more dynamic than anything we show in the roster listing.

I'll add that I'm not a huge fan of this idea either, particularly because of possible issues with systems (such as NCE) where the command station does a lot of user-driven consist management  and this isn't automatically fed back to the roster.

But I'm travelling (by train) at present so limited access to interns and no hardware to play with.

pabender, 2019-07-27:
So I am moving forward with some of the changes noted above.

I have added the ability to scan the roster for consists.  Once I get some additional tests in place, I think this is ready to go.

I have also made the changes required in the Consist managers to update CV19 in the roster file for the locomotive when the Consist Manager makes a change to CV19.  This one works, but I have a concern or two about this.  The primary concern is that this operation requires reading the entire roster file, and then re-writing it.  The read/write considerably slows down the add and remove process.  This is a particular concern when the files have a large number of CVs ( ESU Loksound decoders, for example ).  Because of the potential addition time, I am going to make this an optional operation that defaults to off ( I.e. the current behavior ).

MikeHeintzman, 2019-07-27:
That’s kind of what I was envisioning when I mentioned the software processing overhead. You might consider putting the responsibility of changing CV19 on the operator, as for any other CV, then you’d just have to read the roster files when you create or delete from the consist. I think that’s the way most people would use it. Having it edit CV19 automatically would be a really nice feature, but the additional wait time might be a feature killing feature. Just my suggestion.

pabender, 2019-07-28:
> That’s kind of what I was envisioning when I mentioned the software processing overhead. You might consider putting the responsibility of changing CV19 on the operator, as for any other CV, then you’d just have to read the roster files when you create or delete from the consist. 

The user currently has to manually update the file for the roster, so what you suggest above is status quo.

>I think that’s the way most people would use it. Having it edit CV19 automatically would be a really nice feature, but the additional wait time might be a feature killing feature. Just my suggestion.

This is why we will make it optional.  Users who want it can use it.  Users who don’t are not stuck with it.

pabender, 2019-07-28:
The update to the roster tool to allow scanning roster entry files is in PR #7244.  I would still like to add some more tests.

pabender, 2019-07-30:
Support for scanning the roster for. Insists is now in place via #7244.  I updated the help while I was at it.

Up next will be having the Consist tool update the decoder files, with the option to disable that feature.

MikeHeintzman, 2019-07-30:
If you need any help with testing or anything let me know. Happy to help, although my programming skills are atrophied.

pabender, 2019-08-01:
PR #7270 adds the modifications to the consist tool noted previously in this issue.

pabender, 2019-08-02:
I am back to thinking about the throttle interaction, and the more I think about it, the more I think that having the throttle read the CVs is the wrong approach.

I'm going to go way back to when the consist manager was new (December,203, and sourceforge feature request number 7 https://sourceforge.net/p/jmri/feature-requests/7/ ).

In the feature request,  last paragraph is releavant:
> Additionally, it would be nice to be able to save these 
> as 'consists' or 'trains' and show these in the selection 
> dropdown list that is shown in the throttle.

The desire back then, and I think the right approach still, is to have the consists appear in the selection list in the throttles.

The approach to that is different however.  What this means is we need a viable mechanism to insert a consist into the throttle roster selection widget.  We started down that road before, but the effort was never completed

Years ago, @DanielBoudreau added a widget that can display the NCE Consist Roster in the throttle, but that is tied to the NCE specific consist support, not the more generic Consist Managers.  I'm assuming that does the job.  The biggest issue however is that it is a second selection widget.

I think it's time we revisit this approach and do it right this time.

closed 2021-04-14 by stale[bot]