JMRI/JMRI #7236: Using sonarqube as a standard for Java code practices #cleanup

mickster04, 2019-07-27:
https://www.sonarqube.org/

I use this at work and it is a useful tool to ensure some standards across our builds and developers. We use the plugin with our jenkins server to block builds that would add bugs to the code base, or lessen code quality (code smells) and make sure little things are done with a minimum standard using an automatic tool. I am happy to do some work on this or provide insight into how it is used in our setup. In addition to this, there are plugins available for most common IDEs meaning all developers could ensure they use the standards decided for the JMRI builds as these plugins link into the sonar server if desired, using the same rules and quality gates.

In my view the basic settings are acceptable as a starting point, but I'm sure there are some rules which could be added or removed as is applicable fro JMRI.

I am raising this as an issue first in order to start a dialogue rather than just barging ahead and making what would be a raft of changes which might be a bit rude as an intro into JMRI development.

pabender, 2019-07-27:
I tend to be in favor of these types of tools.  We’ve used, or at least played with, similar tools in the past.

I see there is integration with Jenkins.  We use Jenkins, but only after we have merged code into master.   For pre merge code CI checks, we use Appveyor and Travis.  Does sonarqube integrate into those systems?
 
Basically, I am interested in how we get feedback from sonarqube on a pull request?

It might be beneficial for you to go ahead with a PR so we can get a feel for how this works.

mickster04, 2019-07-27:
Ok, I have done some more research. Because my work isn't FOSS there are restrictions that don't apply here.
Sonarcloud.io is free for FOSS and also hooks in like travis and appveyor, provide repo permissions, and given the right file, will automatically review master branch changes and PRs. I have some more research to do for this and have a 'play' around, but it's looking promising in terms of ease of access/installation.

rhwood, 2019-07-27:
If sonarcube cannot be run locally from the command line, I’d suggest codeclimate.com, since it can be run locally as well as integrated into PRs.

mickster04, 2019-07-27:
sonarlint provides IDE integration, and maven can be used to trigger scans. not sure how that last bit works with sonar cloud though

bobjacobsen, 2019-07-27:
Also in favor of this, thanks for considering it.

Our Jenkins setup is a bit old-style, in that we’re using the task-based config still.  Happy to work with you on that when you get to that point.

bobjacobsen, 2019-07-27:
When adding other tools in the past, i.e. SpotBugs (nee FindBugs), the number of warnings was originally overwhelming.  Rather than add that in CI, we set them up to just run in Jenkins and on developer desktops until the number of warnings could be pushed down, we could figure out which types to ignore, warn on and fail on, etc.  That might be a good approach here as a I expect that sonarqube will point out lots of places our code can be improved!

pabender, 2019-07-27:
> When adding other tools in the past, i.e. SpotBugs (nee FindBugs), the number of warnings was originally overwhelming. Rather than add that in CI, we set them up to just run in Jenkins and on [...]

I agree with Bob.

As an example of that, I recently added the ability to use JQAssistant from Maven. In this tool, you write your own rules to check for various things.  One of the things I was interested in was tests that don’t either check something with an Assert or say an exception is expected.  That one check is currently flagging 1566 problematic tests. ( it was over 2000 when I started looking at this.). This is way too many issues to address immediately, but over time, we will reduce that number.

mickster04, 2019-07-28:
It is possible to set the gates to only apply to changes, so the build will only fail if the number of issues grows, rather than always failing because it is > 1 (ask me how i know) so it could be used to ensure at least that the code doesn't get worse ;) and over time, we could adjust the gates as the code improves so the baseline also improves... I don't have experience with alternatives, so my view is biased.

mickster04, 2019-07-29:
So far I am having difficulty getting sonar to find the source code, however it has found the tests! It has found over 1k tests without assertions. Aside from that I have it integrated with travis CI and doing...something? Assuming I get the src folder scanned, the next question is where to stick it into the travis file? It uses the mvn sonar:sonar job with some -Dsonar params. Currently I have it at line 15 of the travis.sh file above the if STATIC. but I imagine those who actually understand this bit will know of a more appropriate place...

rhwood, 2019-07-29:
I’m assuming you are testing this out of the master branch in your fork.

I would get it working as a plugin in the maven pom.xml first. That will import all the settings you need, so you don’t need to manually specify the source locations (there are multiple).

Once that is complete, do not attempt to run it as part of the static analysis Travis CI runs—sonarqube needs the output from jacoco to function, so run it after the coverage report is uploaded to coveralls, but without rerunning all the tests (use the already created jacoco output).

mickster04, 2019-07-29:
sonar scanner has good tie in into travis, therefore i don't need to use sonarqube, it's sonarcloud instead. Current issue is analysis of 4500 files takes too long, the job times out... it looks to me as if I'm making progress though.

rhwood, 2019-07-29:
Looking into your sources, I can’t find any configuration of sonar in the pom.xml; if developers are to be expected to use this tool, they must be able to do so without relying on online access after initial installation.

pabender, 2019-07-29:
I agree with @rhwood.

I would suggest the following

- [ ] add the required configuration to maven
- [ ] document instructions for how to run the code on maven
- [ ] we configure a build on our jenkins server to view results.
- [ ] If we decide it is worthwhile, we will configure it into our static analysis.

See, for example PR #7213 as how I recently added jQAssistant to the maven build configuration (the first two steps above actually).

There is a jenkins job that runs the jQAssistant (the 3rd step above) checks here:
http://jmri.tagadab.com/jenkins/job/Development/job/jQAssistant%20Reports/

We haven't gotten to the 4th step with jQAssistant yet. (and I don't like the way the results are presented on Jenkins.  The local results are better).

mickster04, 2019-07-30:
ok, so just to clarify, we intend on enabling every developer to use the same sonarcloud instance to run analysis against?

I have managed to get travis ci triggering analysis, but the time it take to complete the full code base exceeds 50 minutes causing travis to fail the build. doing analysis on just the src/apps folder works completely. It seems therefore that I can't use travis ci to run sonarcloud. so I will have to do as you say anyway. However, for anyone who forks the repo, they will need online access to use sonar anyway right? and to use travis? So can I confirm what I am trying to achieve now? The doco i read included setting up travis and appveyor? 

the bits that make travis do sonar stuff are all in the yml. the add ons section and then

sonar-scanner -Dsonar.projectKey=mickster04_JMRI -Dsonar.projectVersion=1.3 -Dsonar.sources=java/src/apps -Dsonar.java.binaries=target/classes/apps

I can't resolve the long job problem though I think so a static job in your jenkins set up is the only way to make this work. and if this is in your jenkins server this won't be too helpful for forks?

pabender, 2019-07-30:
> ok, so just to clarify, we intend on enabling every developer to use the same sonarcloud instance to run analysis against?
> 

Perhaps.  If we look at our Travis and Appveyor setup, anyone who wants to can configure their own fork to use those services.

> I have managed to get travis ci triggering analysis, but the time it take to complete the full code base exceeds 50 minutes causing travis to fail the build. doing analysis on just the src/apps folder works completely. It seems therefore that I can't use travis ci to run sonarcloud.

The src/apps folder is only a small portion of the source.  Without performing analysis on src/jmri, the tool does us no good at all.

> so I will have to do as you say anyway. However, for anyone who forks the repo, they will need online access to use sonar anyway right? and to use travis? So can I confirm what I am trying to [...]

Ok, let’s step back a bit from Travis/Appveyor/Jenkins

Can you find a way to add this into the Maven pom file or the ant build.xml file ( not both necessarily... ).  

Maven configuration is preferred, because it reduces the setup load ( Maven will automatically download dependencies ) also, as @rhwood mentioned, Maven already knows how to find the source and other files related to the project.

pabender, 2019-07-30:
https://docs.sonarqube.org/latest/analysis/scan/sonarscanner-for-maven/

bobjacobsen, 2019-07-30:
> I think so a static job in your jenkins set up is the only way to make this work. and if this is in your jenkins server this won't be too helpful for forks?

It’s possible to set up Jenkins as a CI step in GitHub PRs.  

Our existingJenkins server can run only one job at a time due to memory limitations; lots of long jobs might backlog it a lot.  But we might be able to create other workers for it.

rhwood, 2019-08-29:
@mickster04 What rules were you thinking of using?

mickster04, 2019-08-30:
> @mickster04 What rules were you thinking of using?

to begin with, the standard sonar stuff. it would take a bit of trial and error to set up our own custom rules. 
If we are thinking of putting it in maven and running it from jenkins, it doesn't need to break the build (but it can) however, trying to set up as a new dev would be a bit trickier because the setup is not in code like the travis builds for example. But that's neither here nor there.

I've had a ramp up in work recently so my progress has been nill... apologies!

rhwood, 2020-07-20:
Now that we are using GitHub Actions, and they have a runtime limit of 6 hours, I think this is doable.

closed 2021-04-14 by stale[bot]