JMRI/JMRI #7249: Doehler & Haass - need clean-up for ProductID

AlanUS, 2019-07-28:
I'm addressing to the main contributors to the Doehler & Haass decoder family;
@PierreMc  @Klb4ever 

I've just made a number of changes essentially to add the new CVs for versions 3.10 and 3.11, but doing so, I was confronted to the naming rule used for ProductID, where the version name (or year/month) is appended.
First of all, this was not working as using a dot (.) in the name (like in PD12A_2016.03) was not recognized by the parser (though not generating any error). I have replaced the dot by an underscore in a number of the ProductIDs.

Given that the general design pattern for these decoders is to create as many files as there are versions, with appropriate xinclude files, appending the version to the productID should not be necessary. On the contrary, it creates a lot of burden when new releases are added. I have already removed the version from the ID for PD05A and PD12A.

I've not done it for the other decoders yet; I just wanted to check if there is some special care to be taken before extending the concept to all decoders. Hence this message.

Also I have the impression that in a number of cases, the exclusion of certain decoders does not need to be specified as a proper minOut qualifier will do the job anyway (and it is more robust).

Also given that the decoder type is now available in CV261, it wont take long to improve automatic decoder identification by using the appropriate productID code from CV261 instead of named ID. This would require some change in the Java code, but this is not complicated as the general mechanism is there already. This would however require the ProductID to be independent from the product version.

Feedback welcome

bobjacobsen, 2019-07-28:
> Also given that the decoder type is now available in CV261, it wont take long to improve automatic decoder identification by using the appropriate productID code from CV261 instead of named ID. This [...]

I’m not sure what you mean by “independent” here.

The Java to read CVs and create a productID value is in java/src/jmri//jmrit/decoderdefn/IdentifyDecoder.java

It checks the manufacturer and model ID, and if it recognizes those it goes on to read CVs to create a productID.

But if it doesn’t, it returns default values.

AlanUS, 2019-07-28:
@bobjacobsen 
What I meant is that the original designer created an adhoc naming convention for the productID for these decoders that mixed the name of the product and the version of the product. What I mean by independent is to use the ProductID qualifier only for the product and not mix it with the product version that can be found in CV7.
And thanks for reminding me the path the code for doing this automatic identification.

PierreMc, 2021-03-30:
Hi @AlanUS, the bot just informed me about this issue soon getting closed. Sorry I completely missed it 2y ago.
This was actually recently discussed with @bobjacobsen in https://github.com/JMRI/JMRI/pull/9510 .

TLDR: I have included the change in the Java code for decoder identitication, and the correct product ID as per CV 261 for current decoders. For older product IDs, the consensus was to **not** clean the past decoders defs (many of them not supporting CV 261 anyway), due to the complexity of updating all include/exclude terms.

Let me know if you're fine with this, after which I think this issue could be closed. Thanks Alain!
Pierre

closed 2021-05-29 by stale[bot]