JMRI/JMRI #7253: LocoNet turnout message retry problem

devel-bobm, 2019-07-30:
When a LocoNet turnout control message is "rejected", the JMRI resend can transmit the wrong LocoNet message.  This effect has been seen in a JMRI testcase attempting to exercise the "rejected turnout message" resend mechanism.

In `LnTurnoutManager`, when a turnout control message is received, the code attempts to save the received turnout control message (for possible "re-send" upon receipt of an OPC_LONG_ACK switch request rejection) using the following code:
```
    public void message(LocoNetMessage l) {
        log.debug("LnTurnoutManager message {}", l);
        String prefix = getSystemPrefix();
        // parse message type
        int addr;
        switch (l.getOpCode()) {
            case LnConstants.OPC_SW_REQ: {               /* page 9 of LocoNet PE */

                int sw1 = l.getElement(1);
                int sw2 = l.getElement(2);
                addr = address(sw1, sw2);

                // store message in case resend is needed
                lastSWREQ = l;
```

The last line above is problematic, in that `lastSWREQ` references `l` instead of copying `l`.  This means that if another LocoNet message is received before `lastSWREQ` is referenced, `lastSWREQ` will reflect that more-recent message, not the (desired) turnout control message.  

The obvious fix is to swap in the following line in place of the last line shown above:

```
                lastSWREQ = new LocoNetMessage(l);
```

bobjacobsen, 2019-08-11:
Was this fixed by #7258, in which case we can close it?

devel-bobm, 2019-08-12:
This issue is corrected by #7258.

closed 2019-08-12 by devel-bobm