JMRI/JMRI #7259: Crossover with 2 physical devices both with feedback  causes inconsistant status

Fire-Power, 2019-07-31:
I attach a worked example of a crossover which is configured as two physical devices.
Each physical device (turnout) has two sensors one for closed and one for thrown.
When operated separately the turnouts  work logically correctly.
But when the tick box "two physical devices"  is ticked then the turnout status of the first turnout becomes "inconsistent". The four feedback status being correct at each stage of the operation. Thus it seems that the status of the turnouts is not handled correctly when two devices in a crossover  both have feedback enabled.

[crossover.xml.txt](https://github.com/JMRI/JMRI/files/3453650/crossover.xml.txt)

icklesteve, 2019-07-31:
Hi @Fire-Power and thanks for uploading the MERG CBUS panel.

Do you get the same issue if you turn off Turnout Feedback Automation?
300ms might be OK for solenoids but servos may take longer to complete the action.

Note that you'll get inconsistent turnout feedback for this type in CAN Simulation mode due to the simulated responses not knowing that they need to match.

It would be good to see your CBUS Console log, ( I am not near a layout at present ), along with anything relevant from your main JMRI Console,

Steve.

icklesteve, 2019-08-01:
Am able to reproduce issue via CBUS console log sent via PM on MERG forum.
CBUS messages as expected and all ok resulting in correct feedback state.
Issue seems to be with layoutEditor.LayoutTurnout , will do some more digging tomorrow.

bobjacobsen, 2019-08-01:
I'm not sure that "Sensor" automation makes sense with TwoSensor feedback.  It's using the same sensor to do two different things.

Fire-Power, 2019-08-01:
The sensor automation was something I tried and got left on.
With the sensor automation off the issue is still as stated.

icklesteve, 2019-08-01:
Hi @Fire-Power , the fix should be in package build 3400 onwards from http://builds.jmri.org/jenkins/job/Development/job/Packages/

I've tested with the CAN frame sequence you provided and appears to be sorted.
If you do test, you may want to backup your profile first so you can move back to 4.16 if there's an issue !!

I've raised a separate issue for the save error you spotted, see #7261 

Steve

Fire-Power, 2019-08-01:
Steve
Many thanks for that great work in finding the problem and fixing it so quickly.
I have loaded it on the main layout and the crossovers now work properly.
Bob
Thanks for the guidance  in the forum and incorporating the upgrade as well. A great team response.

icklesteve, 2019-08-01:
Good to hear @Fire-Power , the panel upload here and confirming the CBUS messages helped.
After you've run your main panel for a while and confirmed sorted if you could close this issue down that helps to keep the total number of open issues down.
Ta, Steve.

closed 2019-08-09 by Fire-Power

icklesteve, 2019-08-09:
Thanks for re-confirming.
Note that https://github.com/JMRI/JMRI/pull/7271 adds a few further minor tweaks which should improve 2 sensor feedback under certain conditions.