JMRI/JMRI #7262: LE - Unable to select current Turnout

icklesteve, 2019-08-01:
Select a turnout in LE, edit menu, select the dropdown and the selected Turnout disappears and is not in the dropdown list.
Same behaviour with supporting Turnout.

eg. This is Layout Editor turnout 1 with turnout 2 as supporting, after selecting the dropdowns.

[image]

dsand47, 2019-08-01:
Based on the description, you are probably running 4.17.2 or later.

The turnout list shows the **available** turnouts.  Once a turnout name is assigned, it cannot be assigned to another turnout on the same panel.  

Prior to 4.17.2 this was shown by making the turnout name red.

icklesteve, 2019-08-01:
Hi @dsand47 , issue in current head.
I would expect the currently selected Turnout ( in example above "turnout 1" ) to remain selected at top of the list.
As it disappears, there's no way of knowing what turnout is currently selected.

dsand47, 2019-08-01:
I use the context menu to see what turnouts are assigned, but I can see how not showing the selected name in the combo list can be confusing.  

I have a pretty good idea what needs to be changed.

rhwood, 2019-08-01:
Not using colors in the lists in NamedBeanComboBox was due to two things:
- visual accessibility means that color differences may not be discernible to the user (due to color blindness or the use of high contrast monitors)
- the native Swing UI on both macOS and Linux do not allow colors in combo boxes

dsand47, 2019-08-01:
I want to confirm that you do see the current turnout names in the Edit dialog:
[image]

The assigned turnouts are removed from the turnout list when the drop down button is selected.  This results in an **available** list.  The current turnout is technically not available, but selecting a different makes it available once Done has been selected.

I have a probable solution available if it makes sense.

geowar1, 2019-08-02:
> the native Swing UI on both macOS and Linux do not allow colors in combo boxes
Actually both my macOS and Linux machines allow colors in combo boxes (in both the TextFields and the popup(/dropdown) menus. The previous (to this PR) code worked exactly as expected.
The colorblind issue however is valid.
The only advantage of removing turnouts from the list would be for layouts with more turnouts than would fit on a screen… perhaps for layouts with fewer that a screen heights worth of turnouts just disabling them would be a better solution.

dsand47, 2019-08-02:
Prior to 4.17.2, LE turnouts that were in use for the current panel used red text.  This did not prevent selecting the turnout with the resulting error dialog.

With the move from JmriBeanComboBox to NamedBeanComboBox at 4.17.2, the red text concept was replaced by excluding **_in use_** turnouts.  This prevents the error dialog since those turnouts cannot be selected.

Except for some final testing, I have a change ready that will include the current turnout in the edit dialog.  This actually makes sense since the edited layout turnout already owns the physical turnout.  Selecting a different turnout will release it for use by another layout turnout.

rhwood, 2019-08-03:
> Actually both the native macOS and Linux UI’s allow colors in combo boxes (in both the TextFields and the popup(/dropdown) menus. The previous (to this PR) code worked exactly as expected on all my machines.

That's because we overrode any possibility of the native Swing UIs functioning in that menu, including the native (Aqua and GTK) ones, by using a non-standard renderer.

geowar1, 2019-08-03:
The point is that it worked. So this in itself wasn’t a valid reason not to use colors. The colorblind issue is so it really doesn’t matter.

rhwood, 2019-08-03:
Now that I recall, what didn't work is that the lists could not show the current selection as well as which item in the list the mouse was hovering over because that logic was overridden; so yes, maybe your custom color worked for one purpose, but they broke other expected interactions with the list (i.e. different combo boxes in same window within JMRI display different selection and mouse showing behavior). On one native Swing UI, I forget which, combo boxes use a check mark to indicate the selected item throughout JMRI and other Java programs, except for the old NamedBean selector, so no, it did not work consistently--is was these problems that led to making the changes in the first place.

icklesteve, 2019-08-03:
Every select box I can recall seeing has the currently selected item at the top when opened and displays the currently selected item when closed.
Without this, it doesn't feel right imho.

For info, I was attempting to swap the primary Turnout and the Secondary turnout over to see what effect it had.
While trying to do this, with the currently selected box showing nothing selected, and neither the primary or secondary turnout in the list, it's a UI cul-de-sac.

If colour blindness is the issue for not displaying them, perhaps use the same colour but with less opacity?

rhwood, 2019-08-03:
The combo box takes in a list of beans it should not list; it’s up to whatever is displaying it to decide the contents of that list. I think Dave Sand is working on making the current turnout not be in that list.

Two shades of grey have same problem as color blindness; they are indistinguishable for some visual impairments other than blindness, or when high contrast settings are turned on.

KenC57, 2019-08-03:
If we think that showing the used vs non-used entries is important (I find
it comforting to see them) how about different fonts? Like normal vs
italics? I suspect this isn't something useful for international fonts, but
just listing more ideas in case something works.

What about the way we show valid vs invalid menu items, greyed out is the
term I think? If that is acceptable for menu items, it should be ok in
select lists.
-ken c

bobjacobsen, 2019-08-03:
There's an entire industry that works to provide good Java Look&Feels.  Each one takes an immense amount of effort to get it to work consistently, intuitively, effectively, etc.  There are _lots_ of them in use in lots of different places.

JMRI needs to work regardless of which is selected by the user.

Put together, that means that anybody proposing a change to the way that the L&F code works (i.e. here, changing the rendering of a JComboBox) has a huge burden:  It has to work with those different Look&Feels, it has to stay consistent, intuitive, effective, etc, and the proponent needs to work through all that.  

Most proposals in this area are "we'll just do this here", focusing on that one particular area, and don't consider all the _other_ uses, all the _other_ places where the change will cause issues with consistency, intuition and effectiveness.  

An example:  Greyness is used to indicate enabled/disabled in many L&Fs, but italics are used in others, and bold/roman in others.  If one naively uses italics to indicate something else, when using JMRI with the Liquid L&F (popular in the UK) some controls will be confusing.

Code wins, but if the result is confusing behavior, the code to pull out that source of confusion will also win.

dsand47, 2019-08-03:
**Background Information**

A turnout name can only be used **once** within a Layout Editor panel.  This is a requirement for creating the block topology which supports block routing, signal mast logic, entry/exit, etc.

An attempt to use the same turnout name a second time results in an error dialog.

A year or two ago, color was added to the turnout names in the combo list.  Either green or red is used to indicate the current status.

At 4.17.2, many of the combo boxes were changed from JmriBeanComboBox to NamedBeanComboBox.  The JmriBeanComboBox is **String** based while NamedBeanComboBox is **Bean** based.  This resulted in many changes to the Layout Editor package, including replacing the red/green approach for **"in use"** turnouts with an excluded list approach.

**Current Activity**

I have completed the changes to include the current turnout name in the available list for the Edit dialogs.  Based on the latest post from Steve, I now need to include both turnouts when there is a pair listed.

I have recently discovered a significant oversight in the available list logic:  Turntables using turnouts for ray selection.  These need to be included in the available list determination.

**Direction**

I see two choices for turnout selection in the Edit dialogs and the Turnout lists in the toolbar.
- Display **available** turnouts by excluding, with specific
 exceptions, used turnouts.
- Display all turnouts and use the error dialog.

My current plan is to continue with the exclude approach.  The tooltip will display appropriate text indicating that the list is partial and there will be changes required for the Layout Editor documentation.

icklesteve, 2019-08-06:
Hi @dsand47 , looking great.

When swapping over you have to go via null selected then Done to be able to re-select the Turnout, 
but now it's clear that you're actually selecting a null Turnout :-)

Thanks again,
Steve.

closed 2019-08-06 by icklesteve