JMRI/JMRI #7298: PR4 and DCS52 (and possibly DCS210 & DCS240) should default to Direct Mode

Breezlys, 2019-08-08:
Based on [this thread for the DCS52](https://groups.io/g/jmriusers/topic/dcs52_loksound_select_jmri/32717659?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,32717659), and on [this thread](https://groups.io/g/jmriusers/topic/new_digitrax_pr4/30734958?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,540,30734958) (and my own experience) for the PR4, they should default to Direct instead of Paged. I don't have access to either a DCS210 or DCS240, but I suspect that they should also default to Direct instead of Paged.

bobjacobsen, 2019-08-08:
The last time we looked at this, there was concern that certain old command stations don't properly handle Direct mode.  IIRC, they were early DCS100 and DCS050 models; we don't know how to identify "early and not OK" vs "later and OK" boxes of the same type.

Given that, there are a couple different things we can do:
- Change the default for more modern types, i.e. DC052, DCS210, etc
- In addition to the existing types, add a new "DCS050 with known good Paged mode"  (This is better than having all default to Direct except a new i.e. "DC050 that can't do Direct" because it won't break any existing setups) and similar for DCS100
- (Harder) Create a user-settable default for the order, either overall (I prefer _not_ to do this because it'll drive confusion) or for LocoNet or for just specific LocoNet command station configs
- Just make the change and have people with problematic command stations manually change mode (though that'll confusing for a few, it'll be just a few)

I'll post this list on JMRIusers and see what I get for feedback.  Comments in Issues tend to be just the True Believers in specific changes.

dheap, 2019-08-08:
See my comments on [jmriusers](https://groups.io/g/jmriusers/message/162644).

Pugwash1, 2019-08-08:
The Digitrax decoder manual v2 01 2014, copyright 2017, clearly states on page 28, the Digitrax recommends direct mode for the programming track. No reference to command station type. I think we should just bring our selves in line with the official script.

bobjacobsen, 2019-08-18:
After some pretty extensive testing via the jmriusers list, there are certainly cases where both paged and direct is unreliable, and cases where paged is bad and direct fine, but nobody found a case with direct bad and paged OK.  So I'll create a PR to change the order.

closed 2019-08-21 by Breezlys