JMRI/JMRI #7302: NamedBeanComboBox sort issue

dsand47, 2019-08-11:
When a bean has objects with and without user names, the sort sequence is unpredictable which makes finding an item difficult.

This is a simple example, 5 Internal sensors, one with a user name and five LocoNet sensors, two with user names.
[image]

rhwood, 2019-08-11:
In that example, what sort order was specified for the combo box?

rhwood, 2019-08-11:
(It looks like sort by system name (the default for sorting NamedBeans) was set.)

dsand47, 2019-08-12:
The combo box specifies DisplayName.

rhwood, 2019-08-12:
The problem code is in `jmri.util.NamedBeanUserNameComparator`. It is clearly only sorting user names when two non-null user names sort together and otherwise is using system names.

PR #7304 (2019-08-14): fix: correctly compare NamedBeans when not all beans have user names
@dsand47 Can you verify this works?

Fixes #7302 by only comparing the system names of two NamedBeans when neither NamedBean has a user name; otherwise this compares the user name of a NamedBean against the system name of a NamedBean if one of the NamedBeans does not have a user name. This does not use the display name of a NamedBean to avoid possibly attempting to sort two NamedBeans without user names incorrectly.
commit 005d3dd9 (2019-08-14): fix: correctly compare NamedBeans when not all beans have user names

Fixes #7302 by only comparing the system names of two NamedBeans when neither NamedBean has a user name; otherwise this compares the user name of a NamedBean against the system name of a NamedBean if one of the NamedBeans does not have a user name. This does not use the display name of a NamedBean to avoid possibly attempting to sort two NamedBeans without user names incorrectly.
closed 2019-08-14 by bobjacobsen